victronenergy/venus #2: bbb: simple-upnp is not starting

mpvader, 2015-12-09:
@scottellis 
The /etc/simple-upnpd.xml file was missing. Looks like line 36 and 37 of the recipe are not being executed. The problem was resolved after manually running those two lines on the BBB.

Recipe is here:
https://github.com/victronenergy/meta-victronenergy/blob/WIP_bbb/recipes-networking/simple-upnpd/simple-upnpd_v0.04.bb#L36

scottellis, 2015-12-14:
I modified the `simple-upnpd` *pkg_postinst* to return an error if not running on the host, otherwise it gets marked as already run during the *build rootfs* step on the build workstation and never runs on the BBB.

After that an `/etc/simple-upnpd.xml` file is created, but **:::MAC:::** gets replaced by **\<nothing\>** because *eth0* is not up.

The default *run-postinsts* script runs as `rcS.d/S99run-postinsts`, but the network scripts don't run until runlevel 5.

If I move the *run-postinsts* to `rc5.d/S10run-postinsts`.

This puts it after `rc5.d/S01networking`, `rc5.d/S02dbus-1` and `rc5.d/S05connman`.

With this change after first boot I get this

    root@venus:~# cat /etc/simple-upnpd.xml
    <?xml version="1.0" encoding="utf-8"?>
    <root xmlns="urn:schemas-upnp-org:device-1-0">
      <specVersion>
        <major>1</major>
        <minor>0</minor>
      </specVersion>

      <device>
        <deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
          <friendlyName>BBB Venus (84:EB:18:E2:31:2F)</friendlyName>
          <manufacturer>Victron Energy</manufacturer>
          <manufacturerURL>http://www.victronenergy.com</manufacturerURL>
          <modelName>CCGX</modelName>
          <modelNumber>1.0</modelNumber>
          <modelURL>http://www.victronenergy.com/live/ccgx:start</modelURL>
              <presentationURL>http://www.victronenergy.com/live/ccgx:start</presentationURL>
          <UDN>uuid:com.victronenergy.ccgx.84:EB:18:E2:31:2F</UDN>
        </device>
    </root>

Anyone see a problem with this?

There are some other recipes in the victronenergy repos that might need this same fix to the pkg_postinst function to return an error if not running on the host. Otherwise they probable are not doing what was intended.

mpvader, 2015-12-15:
@scottellis regarding the fix in pkg_postinst: is this caused by a change in bitbake between Danny and Jethro? Or has this always been like this? Please explain, so we can check if it is also necessary for ccgx/danny.

And fix all other packages that have this issue too.

scottellis, 2015-12-15:
I didn't know this about post install scripts behavior until yesterday. 

Post-install scripts can run
1) At the finish of the build of the rootfs
2) On first boot
3) After a package upgrade

If the post-install script succeeds in (1) then it won't be run again in (2). 

The "x$D" = "x" check tests whether you are running in (1) or (2), but you have to explicitly fail the script with an *exit 1* in (1) if you want the script to run again in (2).

At least back through Danny it looks like they have worked this way.

`Section 5.3.6` of the Danny manual

http://www.yoctoproject.org/docs/1.3/mega-manual/mega-manual.html

I'll look at the usage in other victronenergy recipes.

mpvader, 2016-05-31:
closed. simple-upnp works fine on the bbb now

closed 2016-05-31 by mpvader