victronenergy/venus #3: bbb: serial-starter is not running because of down file

mpvader, 2015-12-09:
There is a down file in /service/serial-starter. That down file is normally removed at some point. I don't know when or what process normally removes that file though. 

@jhofstee, do you have a hint for @scottellis  ?

mpvader, 2015-12-13:
The down file is created here:
https://git.victronenergy.com/ccgx/meta-ccgx/blob/danny_venus/recipes-bpp3/serial-starter/serial-starter_0.0.bb#L19

Reason to put it down initially is that ccgx mixes console uart with functional ve.direct port 1. And in the factory our test equipment needs to access the ccgx on the console port.

Ccgx is the exception: the bbb and also hopefully all other future boards we'll keep console separate. And downing serial starter initially is then not necessary. 

So @scottellis, you can make the recipe so that the down file is only there for Ccgx.

And when this is fixed you'll run into the next problem: serial-starter expects all serial related executables in /opt/color-control/. But now they are in /opt/victronenergy/. Also Gui doesn't show version and build date, probably caused by same paths issue.

And after that the serial console will need slightly different config: it's default serial ports (in serial-starter.sh) will need to be set to the tty paths that are mapped to ve.direct ports in the cape.

scottellis, 2015-12-13:
Fixed with these commits to meta-ccgx [WIP_bbb]

    commit d0fa08f8aceef961973642bcdd338c08accb176f
    Author: Scott Ellis <scott@jumpnowtek.com>
    Date:   Sun Dec 13 11:51:51 2015 -0500

        serial-starter: bbb: Add machine specific serial-starter script

    commit 5d3e5b9fd890f4c44a9184429ebf74d38ffb5b36
    Author: Scott Ellis <scott@jumpnowtek.com>
    Date:   Sun Dec 13 11:02:05 2015 -0500

        serial-starter: Down by default only on ccgx

And this commit to meta-victronenergy [WIP_bbb]

    commit d9b137a9960e51bdb003e59c40af9e58b1b04e2b
    Author: Scott Ellis <scott@jumpnowtek.com>
    Date:   Sun Dec 13 11:50:48 2015 -0500

        ve_package: bbb: use opt color-control for vedir

closed 2015-12-13 by scottellis