victronenergy/venus #4: bbb: inserting VE.Direct to USB interface doesn't result in /dev/ttyUSBx

mpvader, 2015-12-09:
log output
```
Dec  9 21:00:40 venus user.debug kernel: [ 6857.672042] musb-hdrc musb-hdrc.1.auto: VBUS_ERROR in a_wait_bcon (99, VALID), retry #1, port1 00000104
Dec  9 21:00:40 venus user.info kernel: [ 6858.124405] usb 1-1: new full-speed USB device number 2 using musb-hdrc
Dec  9 21:00:40 venus user.info kernel: [ 6858.275379] usb 1-1: New USB device found, idVendor=0403, idProduct=6015
Dec  9 21:00:40 venus user.info kernel: [ 6858.282432] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Dec  9 21:00:40 venus user.info kernel: [ 6858.289966] usb 1-1: Product: FT232EX
Dec  9 21:00:40 venus user.info kernel: [ 6858.293816] usb 1-1: Manufacturer: FTDI
Dec  9 21:00:40 venus user.debug kernel: [ 6858.324626] usb 1-1: no of_node; not parsing pinctrl DT
Dec  9 21:00:40 venus user.debug kernel: [ 6858.330750] usbtest 1-1:1.0: no of_node; not parsing pinctrl DT
```

And /dev:
```
root@venus:/dev# ls
autofs              kmsg                mmcblk0p1           ram14               tty                 tty21               tty35               tty49               tty62               usbmon0
block               log                 mmcblk0p2           ram15               tty0                tty22               tty36               tty5                tty63               usbmon1
bus                 loop-control        mtab                ram2                tty1                tty23               tty37               tty50               tty7                vcs
char                loop0               network_latency     ram3                tty10               tty24               tty38               tty51               tty8                vcs1
console             loop1               network_throughput  ram4                tty11               tty25               tty39               tty52               tty9                vcsa
cpu_dma_latency     loop2               null                ram5                tty12               tty26               tty4                tty53               ttyO0               vcsa1
disk                loop3               psaux               ram6                tty13               tty27               tty40               tty54               ttyO2               watchdog
fd                  loop4               ptmx                ram7                tty14               tty28               tty41               tty55               ttyO4               watchdog0
full                loop5               pts                 ram8                tty15               tty29               tty42               tty56               ttyO5               zero
hwrng               loop6               ram0                ram9                tty16               tty3                tty43               tty57               ttyS0
i2c-0               loop7               ram1                random              tty17               tty30               tty44               tty58               ttyS1
iio:device0         memory_bandwidth    ram10               shm                 tty18               tty31               tty45               tty59               ttyS2
initctl             mmcblk0             ram11               stderr              tty19               tty32               tty46               tty6                ttyS3
input               mmcblk0boot0        ram12               stdin               tty2                tty33               tty47               tty60               ubi_ctrl
kmem                mmcblk0boot1        ram13               stdout              tty20               tty34               tty48               tty61               urandom
```
@scottellis you have that hardware (VE.Direct to USB interface), right?

scottellis, 2015-12-10:
Added a number of USB serial drivers to kernel including USB_FTDI_SIO.

    [   49.150167] usb 1-1: new full-speed USB device number 2 using musb-hdrc
    [   49.301425] usb 1-1: New USB device found, idVendor=0403, idProduct=6015
    [   49.308478] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    [   49.316010] usb 1-1: Product: FT232EX
    [   49.319850] usb 1-1: Manufacturer: FTDI
    [   49.359490] usb 1-1: no of_node; not parsing pinctrl DT
    [   49.383823] usbtest 1-1:1.0: no of_node; not parsing pinctrl DT
    [   49.419572] usbcore: registered new interface driver usbserial
    [   49.425906] usbcore: registered new interface driver usbserial_generic
    [   49.432871] usbserial: USB Serial support registered for generic
    [   49.499135] usbcore: registered new interface driver ftdi_sio
    [   49.505408] usbserial: USB Serial support registered for FTDI USB Serial Device
    [   49.513228] ftdi_sio 1-1:1.0: no of_node; not parsing pinctrl DT
    [   49.519610] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
    [   49.526775] ftdi_sio ttyUSB0: no of_node; not parsing pinctrl DT
    [   49.533146] usb 1-1: Detected FT-X
    [   49.618387] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0

mpvader, 2016-05-19:
re-opening the issue, as it stopped working in latest image. See message on the mailing list [here](https://groups.google.com/d/msg/victron-dev-venus/4EK2q2NNPUM/cMGWc7qMEAAJ). 

Probably doesn't work because the necessary kernel module was not in the rootfs, and did not include ftdi_sio or any other usb serial modules:

```
# ls -R /lib/modules/
/lib/modules/:
4.1.15-venus

/lib/modules/4.1.15-venus:
kernel               modules.builtin.bin  modules.order
modules.alias        modules.dep          modules.softdep
modules.alias.bin    modules.dep.bin      modules.symbols
modules.builtin      modules.devname      modules.symbols.bin

/lib/modules/4.1.15-venus/kernel:
drivers

/lib/modules/4.1.15-venus/kernel/drivers:
scsi  usb

/lib/modules/4.1.15-venus/kernel/drivers/scsi:
scsi_mod.ko  sd_mod.ko

/lib/modules/4.1.15-venus/kernel/drivers/usb:
serial   storage

/lib/modules/4.1.15-venus/kernel/drivers/usb/serial:
usbserial.ko

/lib/modules/4.1.15-venus/kernel/drivers/usb/storage:
usb-storage.ko
```

Fixed [here in meta-victronenergy](https://github.com/victronenergy/meta-victronenergy/commit/aef1366ee02495116149f657fbc17ff0a505be96). 

Needs to be verified with a usb-vedirect cable before (re)closing this issue.

mpvader, 2016-05-20:
Indeed fixed by adding `MACHINE_EXTRA_RRECOMMENDS += "kernel-modules"` to beaglebone.conf in [this commit](https://github.com/victronenergy/meta-victronenergy/commit/aef1366ee02495116149f657fbc17ff0a505be96).

Belog log shows inserting a VE.Direct - USB interface, up to the end result: a VE.Direct Solar Charger visible on the D-Bus.

```
[  234.149957] usb 1-1: new full-speed USB device number 4 using musb-hdrc
[  234.291271] usb 1-1: New USB device found, idVendor=0403, idProduct=6015
[  234.298319] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  234.305841] usb 1-1: Product: FT232EX
[  234.309681] usb 1-1: Manufacturer: FTDI
[  234.339516] usbcore: registered new interface driver usbserial
[  234.345878] usbcore: registered new interface driver usbserial_generic
[  234.354234] usbserial: USB Serial support registered for generic
[  234.389523] usbcore: registered new interface driver ftdi_sio
[  234.395865] usbserial: USB Serial support registered for FTDI USB Serial Device
[  234.403831] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
[  234.411175] usb 1-1: Detected FT-X
[  234.423407] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0

root@beaglebone:~# ls /dev/tty*
/dev/tty      /dev/tty2     /dev/tty31    /dev/tty43    /dev/tty55    /dev/ttyO0
/dev/tty0     /dev/tty20    /dev/tty32    /dev/tty44    /dev/tty56    /dev/ttyO2
/dev/tty1     /dev/tty21    /dev/tty33    /dev/tty45    /dev/tty57    /dev/ttyO4
/dev/tty10    /dev/tty22    /dev/tty34    /dev/tty46    /dev/tty58    /dev/ttyO5
/dev/tty11    /dev/tty23    /dev/tty35    /dev/tty47    /dev/tty59    /dev/ttyS0
/dev/tty12    /dev/tty24    /dev/tty36    /dev/tty48    /dev/tty6     /dev/ttyS1
/dev/tty13    /dev/tty25    /dev/tty37    /dev/tty49    /dev/tty60    /dev/ttyS2
/dev/tty14    /dev/tty26    /dev/tty38    /dev/tty5     /dev/tty61    /dev/ttyS3
/dev/tty15    /dev/tty27    /dev/tty39    /dev/tty50    /dev/tty62    /dev/ttyUSB0
/dev/tty16    /dev/tty28    /dev/tty4     /dev/tty51    /dev/tty63
/dev/tty17    /dev/tty29    /dev/tty40    /dev/tty52    /dev/tty7
/dev/tty18    /dev/tty3     /dev/tty41    /dev/tty53    /dev/tty8
/dev/tty19    /dev/tty30    /dev/tty42    /dev/tty54    /dev/tty9

root@beaglebone:~# dbus -y
org.freedesktop.DBus
com.victronenergy.system
com.victronenergy.qwacs
com.victronenergy.logger
com.victronenergy.fronius
com.victronenergy.settings
com.victronenergy.solarcharger.ttyUSB0
net.connman
```

closed 2016-05-20 by mpvader