victronenergy/venus #5: bbb: vrmlogger not starting (python-requests dependencies)

mpvader, 2015-12-09:
As emailed to dev-venus mailing list, vrmlogger can't import the python cgi module. But after adding that there are more issues, and now I remember encountering the same issue while working at fido_venus.

The problem is that there is a new recipe of python-requests somewhere, which doesn't have the right dependencies. 

On BBB:
```
root@venus:/dev# opkg status python-requests
Package: python-requests
Version: 2.7.0-r0
Depends: python-core
Status: install ok installed
Architecture: cortexa8hf-vfp-neon
Installed-Time: 1449436429
```

On CCGX:
```
root@ccgx:/usr/lib/python2.7# opkg status python-requests
Package: python-requests
Version: 2.2.1-r0
Depends: python, python-netserver, python-core, python-email
Provides:
Status: install ok installed
Architecture: armv7a-vfp-neon
Installed-Time: 1449254695
```

The CCGX recipe is here:
https://github.com/victronenergy/meta-victronenergy-overlay/blob/danny_venus/meta/recipes-devtools/python/python-requests_2.2.1.bb

Previously fixed here I suppose:
https://github.com/victronenergy/meta-openembedded/commit/da52e8ad81a987912b9f2ddf03b7f04d6c359b3a

scottellis, 2015-12-09:
It's already fixed.

I just haven't posted a new image.

root@venus:~# opkg status python-requests
Package: python-requests
Version: 2.7.0-r0
Depends: python-email, python-json, python-netserver, python-zlib
Status: install ok installed
Architecture: cortexa8hf-vfp-neon
Installed-Time: 1449696313

I'll summarize what I have running now and post a new image for you.

scottellis, 2015-12-10:
Fixed in meta-victronenergy-overlay

commit cfb0a833f861bdeadf2d95afce3170a7add94938
Author: Scott Ellis <scott@jumpnowtek.com>
Date:   Thu Dec 10 03:59:32 2015 -0500

    python-requests: Add RDEPENDS

closed 2015-12-10 by scottellis