victronenergy/venus #6: bbb: wifi says 'no access points'

mpvader, 2015-12-09:
[image]

Some background: the gui uses connman. Connman is also accessible via dbus. And it does find the wifi dongle on the bbb:
```
root@venus:/var/log# dbus -y net.connman
/
/net
/net/connman
/net/connman/service
/net/connman/service/ethernet_6ceceb5c712b_cable
/net/connman/technology
/net/connman/technology/ethernet
/net/connman/technology/wifi
root@venus:/var/log# dbus -y net.connman /net/connman/technology/wifi GetProperties
properties = {'Connected': False,
 'Name': 'WiFi',
 'Powered': True,
 'Tethering': False,
 'Type': 'wifi'}
root@venus:/var/log# dbus -y net.connman /net/connman/technology/wifi Scan
Traceback (most recent call last):
  File "/usr/bin/dbus", line 364, in <module>
    ret = obj.object.get_dbus_method(method.name, iface.name)(*args)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: net.connman.Error.NotSupported: Not supported
```

I don't know why Scan doesn't work. It works better on a CCGX:
```
root@ccgx:~# dbus -y net.connman /net/connman/technology/wifi Scan
Traceback (most recent call last):
  File "/usr/bin/dbus", line 374, in <module>
    pprint.pprint(to_native_type(arg))
  File "/usr/bin/dbus", line 284, in to_native_type
    return int(data)
TypeError: int() argument must be a string or a number, not 'NoneType'
```
(the NoneType error is a bug in dbus-cli)

More info from a CCGX:
```
Dec  9 22:39:42 CCGX user.info kernel: [12413.893218] usb 1-1.1: new high-speed USB device number 4 using ehci-omap
Dec  9 22:39:42 CCGX user.info kernel: [12414.027404] usb 1-1.1: New USB device found, idVendor=0586, idProduct=341e
Dec  9 22:39:42 CCGX user.info kernel: [12414.027435] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec  9 22:39:42 CCGX user.info kernel: [12414.027465] usb 1-1.1: Product: 802.11 n WLAN
Dec  9 22:39:42 CCGX user.info kernel: [12414.027496] usb 1-1.1: Manufacturer: Ralink
Dec  9 22:39:42 CCGX user.info kernel: [12414.027496] usb 1-1.1: SerialNumber: 1.0
Dec  9 22:39:42 CCGX user.info kernel: [12414.093933] Loading modules backported from Linux version v3.16-0-g19583ca
Dec  9 22:39:42 CCGX user.info kernel: [12414.093963] Backport generated by backports.git v3.16-1-0-g87df966
Dec  9 22:39:42 CCGX user.info kernel: [12414.341430] cfg80211: Calling CRDA to update world regulatory domain
Dec  9 22:39:42 CCGX user.info kernel: [12414.424499] usb 1-1.1: reset high-speed USB device number 4 using ehci-omap
Dec  9 22:39:43 CCGX user.info kernel: [12414.554107] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
Dec  9 22:39:43 CCGX user.info kernel: [12414.609527] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0005 detected
Dec  9 22:39:43 CCGX user.debug kernel: [12414.616638] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Dec  9 22:39:43 CCGX user.debug kernel: [12414.622161] Registered led device: rt2800usb-phy0::radio
Dec  9 22:39:43 CCGX user.debug kernel: [12414.622497] Registered led device: rt2800usb-phy0::assoc
Dec  9 22:39:43 CCGX user.debug kernel: [12414.622863] Registered led device: rt2800usb-phy0::quality
Dec  9 22:39:43 CCGX user.info kernel: [12414.623199] usbcore: registered new interface driver rt2800usb
Dec  9 22:39:43 CCGX daemon.info connmand[1452]: wlan0 {create} index 5 type 1 <ETHER>
Dec  9 22:39:43 CCGX daemon.info connmand[1452]: wlan0 {update} flags 4098 <DOWN>
Dec  9 22:39:43 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 address 10:7B:EF:C7:79:A4 mtu 1500
Dec  9 22:39:43 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 operstate 2 <DOWN>
Dec  9 22:39:43 CCGX daemon.info connmand[1452]: Adding interface wlan0 [ wifi ]
Dec  9 22:39:43 CCGX user.info kernel: [12414.686614] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
Dec  9 22:39:43 CCGX user.info kernel: [12414.724975] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.29
Dec  9 22:39:44 CCGX user.info kernel: [12415.766082] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {update} flags 36931 <UP,RUNNING>
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 address 10:7B:EF:C7:79:A4 mtu 1500
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 operstate 0 <UNKNOWN>
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {update} flags 36867 <UP>
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 address 10:7B:EF:C7:79:A4 mtu 1500
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 operstate 2 <DOWN>
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {update} flags 36866 <DOWN>
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 address 10:7B:EF:C7:79:A4 mtu 1500
Dec  9 22:39:44 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 operstate 2 <DOWN>
Dec  9 22:39:44 CCGX user.warn wpa_supplicant: Libgcrypt warning: missing initialization - please fix the application
Dec  9 22:40:07 CCGX user.info kernel: [12439.236236] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Dec  9 22:40:07 CCGX daemon.info connmand[1452]: wlan0 {update} flags 36867 <UP>
Dec  9 22:40:07 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 address 10:7B:EF:C7:79:A4 mtu 1500
Dec  9 22:40:07 CCGX daemon.info connmand[1452]: wlan0 {newlink} index 5 operstate 2 <DOWN>
Dec  9 22:40:10 CCGX user.info kernel: [12441.656616] cfg80211: Calling CRDA to update world regulatory domain
root@ccgx:~# dbus -y net.connman
/
/net
/net/connman
/net/connman/service
/net/connman/service/ethernet_9059af69d820_cable
/net/connman/service/wifi_107befc779a4_4e45544745415234352d7a6f6c646572_managed_psk
/net/connman/service/wifi_107befc779a4_4e4554474541523435_managed_psk
/net/connman/technology
/net/connman/technology/ethernet
/net/connman/technology/wifi
root@ccgx:~# dbus -y net.connman /net/connman/technology/wifi GetProperties
properties = {'Connected': False,
 'Name': 'WiFi',
 'Powered': True,
 'Tethering': False,
 'Type': 'wifi'}
```

scottellis, 2015-12-10:
Another data point.

The kernel detects and instantiates the wan0 device whether the dongle is attached at boot or at runtime.

If the USB Wifi dongle is attached at boot, dbus sees it. 
    
    root@venus:~# uname -a
    Linux venus 4.1.14-venus #1 Thu Dec 10 04:51:20 EST 2015 armv7l GNU/Linux

    root@venus:~# dbus -y net.connman
    /
    /net
    /net/connman
    /net/connman/service
    /net/connman/service/ethernet_c8a030a69fbd_cable
    /net/connman/technology
    /net/connman/technology/ethernet
    /net/connman/technology/wifi

    root@venus:~# dbus -y net.connman /net/connman/technology/wifi GetProperties
    properties = {'Connected': False,
     'Name': 'WiFi',
     'Powered': False,
     'Tethering': False,
     'Type': 'wifi'}

    root@venus:~# dbus -y net.connman /net/connman/technology/wifi Scan
    Traceback (most recent call last):
      File "/usr/bin/dbus", line 364, in <module>
        ret = obj.object.get_dbus_method(method.name, iface.name)(*args)
      File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
        **keywords)
      File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
        message, timeout)
    dbus.exceptions.DBusException: net.connman.Error.NotSupported: Not supported

mpvader, 2015-12-11:
Error on the bbb in messages:

Activated service 'fi.w1.wpa_supplicant1' failed: Failed to execute program fi.w1.wpa_supplicant1: Success

From bbb, while no wifi dongle is inserted:
```
root@venus:~# cat /etc/version
20151206211337
root@venus:~# cat /var/log/messages
....
Dec 10 21:19:03 venus daemon.info connmand[938]: eth0 {create} index 2 type 1 <ETHER>
Dec 10 21:19:03 venus daemon.info connmand[938]: eth0 {update} flags 69699 <UP,RUNNING,LOWER_UP>
Dec 10 21:19:03 venus daemon.info connmand[938]: eth0 {newlink} index 2 address 6C:EC:EB:5C:71:2B mtu 1500
Dec 10 21:19:03 venus daemon.info connmand[938]: eth0 {newlink} index 2 operstate 6 <UP>
Dec 10 21:19:03 venus daemon.notice dbus[820]: [system] Activated service 'fi.w1.wpa_supplicant1' failed: Failed to execute program fi.w1.wpa_supplicant1: Success
Dec 10 21:19:03 venus daemon.info connmand[938]: Adding interface eth0 [ ethernet ]
Dec 10 21:19:03 venus daemon.info connmand[938]: can0 {newlink} index 3 address 00:00:00:00:00:00 mtu 16
Dec 10 21:19:03 venus daemon.info connmand[938]: can0 {newlink} index 3 operstate 6 <UP>
Dec 10 21:19:03 venus daemon.info connmand[938]: can1 {newlink} index 4 address 00:00:00:00:00:00 mtu 16
Dec 10 21:19:03 venus daemon.info connmand[938]: can1 {newlink} index 4 operstate 2 <DOWN>
....
```

from /data/log/messages on ccgx (while no wifi dongle is inserted):
```
Dec 11 12:58:15 CCGX daemon.info connmand[1082]: eth0 {update} flags 4099 <UP>
Dec 11 12:58:15 CCGX daemon.info connmand[1082]: eth0 {newlink} index 4 address 90:59:AF:69:D8:20 mtu 1500
Dec 11 12:58:15 CCGX daemon.info connmand[1082]: eth0 {newlink} index 4 operstate 2 <DOWN>
Dec 11 12:58:15 CCGX daemon.info connmand[1082]: Adding interface eth0 [ ethernet ]
Dec 11 12:58:15 CCGX daemon.notice dbus[1043]: [system] Successfully activated service 'fi.w1.wpa_supplicant1'
Dec 11 12:58:16 CCGX daemon.info connmand[1082]: eth0 {add} route ff00:: gw :: scope 0 <UNIVERSE>
Dec 11 12:58:16 CCGX daemon.info connmand[1082]: eth0 {add} route fe80:: gw :: scope 0 <UNIVERSE>
Dec 11 12:58:16 CCGX daemon.info connmand[1082]: eth0 {update} flags 69699 <UP,RUNNING,LOWER_UP>
```

mpvader, 2015-12-11:
Found it. The problem is in /etc/dbus-1/system.conf, and then the lines:
```
  <!-- This is a setuid helper that is used to launch system services -->
  <servicehelper>/usr/libexec/dbus-daemon-launch-helper</servicehelper>
```

That program is there on the ccgx. But on BBB it is elsewhere. Changing those lines into this fixes it:
```
  <!-- This is a setuid helper that is used to launch system services -->
  <servicehelper>/usr/lib/dbus/dbus-daemon-launch-helper</servicehelper>
```

Background: D-Bus can start services/executables. Service definitions are here:
root@venus:~# cat /usr/share/dbus-1/system-services/fi.w1.wpa_supplicant1.service 
[D-BUS Service]
Name=fi.w1.wpa_supplicant1
Exec=/usr/sbin/wpa_supplicant -u
User=root
SystemdService=wpa_supplicant.service
```

Command to start it from command line:
```
root@venus:~# dbus-send --system --print-reply --dest=org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus.StartServiceByName string:fi.w1.wpa_supplicant1 uint32:0
Error org.freedesktop.DBus.Error.Spawn.ExecFailed: Failed to execute program fi.w1.wpa_supplicant1: Success
```

mpvader, 2015-12-12:
changed system.conf file added here:
https://github.com/victronenergy/meta-victronenergy-overlay/commit/46a4f5b8e6a00c49540314e1769f4f0ba1ecc78d

closed 2015-12-12 by mpvader