victronenergy/venus #8: bbb: /data, /log and /conf

mpvader, 2015-12-11:
(moved it from the mailing to here to keep a better overview)

/data needs to be a separate partition, which it is now. good.

Which leaves /conf and /log, they need to be symlinks. Or, even better: we now decide to remove them entirely. All applications should use /data/conf/ directly. ccgx-gui run script already uses that. bbb now doesn't , please change that so it is the same. And the only other program would be localsettings, which is easy enough to change, I'll do so in a few minutes.

And lets remove /log too. Jeroen and myself are quite happy with logs in /var/log instead of /data/log. Less work any way, changing all kinds of standard scripts.. But then do symlink /var/log to /data/log/ instead of /var/volatile which it is now. Logs shouldn't get lost during an image upgrade, and neither during a reboot.

/db is not used by anythin, so just remove that symlink.

From CCGX:
```
root@ccgx:/# ls -al
drwxr-xr-x   20 root     root          1512 Dec  9 19:13 .
drwxr-xr-x   20 root     root          1512 Dec  9 19:13 ..
drwxr-xr-x    2 root     root           232 Dec  9 19:13 Settings
drwxr-xr-x    2 root     root          5968 Dec  4 18:45 bin
drwxr-xr-x    2 root     root           224 Dec  9 19:13 boot
lrwxrwxrwx    1 root     root            10 Dec  9 19:13 conf -> /data/conf
drwxr-xr-x    9 root     root           624 Oct 17 19:34 data
lrwxrwxrwx    1 root     root             8 Dec  9 19:13 db -> /data/db
drwxr-xr-x   11 root     root          3480 Dec 11 12:58 dev
drwxr-xr-x   36 root     root          5864 Dec  9 19:13 etc
drwxr-sr-x    4 root     root           288 Dec  1 13:44 home
drwxr-xr-x    7 root     root          4544 Dec  4 18:45 lib
lrwxrwxrwx    1 root     root             9 Dec  9 19:13 log -> /data/log
drwxr-xr-x   10 root     root           680 Dec  4 18:45 media
drwxr-xr-x    2 root     root           416 Dec  4 18:45 mnt
drwxr-xr-x    3 root     root           232 Dec  1 14:13 opt
dr-xr-xr-x  187 root     root             0 Jan  1  1970 proc
drwxr-xr-x    2 root     root          5552 Dec  4 18:45 sbin
drwxr-xr-x    2 root     root          2880 Dec 11 12:59 service
dr-xr-xr-x   12 root     root             0 Dec 11 12:58 sys
drwxrwxrwt    4 root     root          1240 Dec 11 12:58 tmp
drwxr-xr-x   11 root     root           736 Dec  1 14:17 usr
drwxr-xr-x    8 root     root           944 Dec  9 19:13 var
```

mpvader, 2015-12-11:
@scottellis localsettings change done, version v1.04 now.
https://github.com/victronenergy/localsettings/releases/tag/v1.04

mpvader, 2015-12-11:
probably best to leave /log (as symlink to /data/log) in place for now. Or spend a lot of time on something not urgent, as /log is used in many places:
```
root@venus:/etc# grep messages *
php-fpm.conf:; message. This lets syslogd specify that messages from different facilities
syslog-startup.conf:LOGFILE=/log/messages		# where to log (file)
syslog-startup.conf.busybox:LOGFILE=/log/messages		# where to log (file)
watchdog.conf:#file			= /var/log/messages
```

``` 
root@venus:/opt/color-control# cat */service/log/run
#!/bin/sh
exec 2>&1
exec multilog t s99999 n8 /var/log/dbus-fronius
#!/bin/sh
exec 2>&1
exec multilog s99999 n8 /log/dbus-qwacs
#!/bin/sh
exec 2>&1
exec multilog t s99999 n8 /log/gui
#!/bin/sh
exec 2>&1
exec multilog t s99999 n8 /var/log/serial-starter
#!/bin/sh
exec 2>&1
exec multilog t s99999 n8 /log/ssh-tunnel
```

mpvader, 2015-12-11:
and create /data/db, which will fix:
```
root@venus:/data/db# cat /var/log/vrmlogger/current | tai64nlocal
2015-12-11 19:28:00.132485500 MainThread-vrmlogger: /opt/victronenergy/vrmlogger/vrmlogger.py v2.37 is starting up
2015-12-11 19:28:00.133639500 MainThread-vrmlogger: Loglevel set to INFO
2015-12-11 19:28:00.172378500 MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
2015-12-11 19:28:00.189277500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
2015-12-11 19:28:00.193418500 MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
2015-12-11 19:28:00.199626500 MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
2015-12-11 19:28:00.456357500 MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2015-12-11 19:28:00.462519500 MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
2015-12-11 19:28:00.662350500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
2015-12-11 19:28:00.812002500 Traceback (most recent call last):
2015-12-11 19:28:00.812471500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 554, in <module>
2015-12-11 19:28:00.813198500     main()
2015-12-11 19:28:00.813471500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 537, in main
2015-12-11 19:28:00.814042500     logger.info("Backlog buffer initialised. Count == %d" % buffer.count)
2015-12-11 19:28:00.814239500   File "/opt/victronenergy/vrmlogger/nonvolatilequeue.py", line 109, in count
2015-12-11 19:28:00.814471500     return self._count
2015-12-11 19:28:00.814721500 AttributeError: 'NonVolatileQueue' object has no attribute '_count'
```

scottellis, 2015-12-14:
There was a recursive link to /data/log/log in the last build. Not sure where it came from.
Updated the datadir.sh init script to remove it and things look better.

    root@venus:~# ls -l /
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 Settings
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 bin
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 boot
    drwxr-xr-x    6 root     root          4096 Jan  1  1970 data
    drwxr-xr-x    9 root     root          3000 Dec 14 15:29 dev
    drwxr-xr-x   36 root     root          4096 Dec 14 15:29 etc
    drwxr-xr-x    3 root     root          4096 Dec  8 23:29 home
    drwxr-xr-x    7 root     root          4096 Dec 14 15:29 lib
    lrwxrwxrwx    1 root     root             9 Jan  1  1970 log -> /data/log
    drwx------    2 root     root         16384 Dec 14 15:31 lost+found
    drwxr-xr-x    2 root     root          4096 Dec  8 23:29 media
    drwxr-xr-x    2 root     root          4096 Dec  8 23:29 mnt
    drwxr-xr-x    3 root     root          4096 Dec  9 00:12 opt
    dr-xr-xr-x  176 root     root             0 Jan  1  1970 proc
    drwxr-xr-x    8 root     root           440 Dec 14 15:29 run
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 sbin
    drwxr-xr-x    2 root     root          4096 Dec 14 15:30 service
    dr-xr-xr-x   12 root     root             0 Jan  1  1970 sys
    lrwxrwxrwx    1 root     root             8 Jan  1  1970 tmp -> /var/tmp
    drwxr-xr-x   10 root     root          4096 Dec  9 15:25 usr
    drwxr-xr-x    9 root     root          4096 Jan  1  1970 var

    root@venus:~# ls -l /data
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 conf
    drwxr-xr-x    2 www-data root          4096 Dec 14 15:29 db
    drwxr-xr-x   29 root     root          4096 Dec 14 15:30 log
    drwx------    2 root     root         16384 Dec 14 15:31 lost+found

    root@venus:~# ls -l /data/conf
    -rw-r--r--    1 root     root          3382 Dec 14 15:29 settings.xml
    -rw-r--r--    1 root     root             0 Dec 14 15:29 vncpassword.txt

    root@venus:~# ls -l /data/db
    -rw-r--r--    1 root     root          6144 Dec 14 15:29 vrmlogger-backlog.sqlite3

    root@venus:~# ls -l /data/log
    -rw-r--r--    1 root     root             0 Jan  1  1970 boot
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-cgwacs.ttyO2
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-cgwacs.ttyO4
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-cgwacs.ttyO5
    drwx------    2 root     root          4096 Dec 14 15:29 dbus-fronius
    drwx------    2 root     root          4096 Dec 14 15:29 dbus-qwacs
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-redflow.ttyO2
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-redflow.ttyO4
    drwx------    2 root     root          4096 Dec 14 15:30 dbus-redflow.ttyO5
    drwx------    2 root     root          4096 Dec 14 15:29 dbus-systemcalc-py
    drwx------    2 root     root          4096 Dec 14 15:29 dbus-vebus-to-pvinverter
    -rw-r--r--    1 root     root         36923 Jan  1  1970 dmesg
    drwx------    2 root     root          4096 Dec 14 15:29 gps-dbus.ttyO2
    drwx------    2 root     root          4096 Dec 14 15:29 gps-dbus.ttyO4
    drwx------    2 root     root          4096 Dec 14 15:29 gps-dbus.ttyO5
    drwx------    2 root     root          4096 Dec 14 15:29 gui
    drwxr-xr-x    2 root     root          4096 Dec 14 15:29 hiawatha
    -rw-r--r--    1 root     root           292 Dec 14 15:30 lastlog
    drwx------    2 root     root          4096 Dec 14 15:29 localsettings
    -rw-r--r--    1 root     root         67701 Dec 14 15:30 messages
    drwx------    2 root     root          4096 Dec 14 15:29 opkg
    -rw-------    1 root     root           115 Dec 14 15:29 php-fpm.log
    -rw-r--r--    1 root     root           219 Dec 14 15:29 postinstall.log
    drwx------    2 root     root          4096 Dec 14 15:29 serial-starter
    drwx------    2 root     root          4096 Dec 14 15:29 ssh-tunnel
    drwx------    2 root     root          4096 Dec 14 15:29 sshd
    drwx------    2 root     root          4096 Dec 14 15:29 vecan-dbus
    drwx------    2 root     root          4096 Dec 14 15:29 vedirect.ttyO2
    drwx------    2 root     root          4096 Dec 14 15:29 vedirect.ttyO4
    drwx------    2 root     root          4096 Dec 14 15:29 vedirect.ttyO5
    drwx------    2 root     root          4096 Dec 14 15:29 vrmlogger
    drwxr-x---    2 root     root          4096 Dec 14 15:29 watchdog
    drwx------    2 root     root          4096 Dec 14 15:29 websockify-c

    root@venus:~# ls -l /var
    drwxr-xr-x    2 root     root          4096 Dec  8 23:29 backups
    drwxr-xr-x    4 root     root          4096 Dec 14 15:29 cache
    drwxr-xr-x    9 root     root          4096 Dec 14 15:29 lib
    drwxr-xr-x    2 root     root          4096 Dec  8 23:29 local
    lrwxrwxrwx    1 root     root            11 Dec  8 23:29 lock -> ../run/lock
    lrwxrwxrwx    1 root     root             9 Jan  1  1970 log -> /data/log
    lrwxrwxrwx    1 root     root             6 Dec  8 23:29 run -> ../run
    lrwxrwxrwx    1 root     root            22 Jan  1  1970 services -> /var/volatile/services
    drwxr-xr-x    4 root     root          4096 Dec 14 15:29 spool
    lrwxrwxrwx    1 root     root            12 Dec  8 23:29 tmp -> volatile/tmp
    drwxrwxrwt    4 root     root            80 Jan  1  1970 volatile
    drwxr-xr-x    4 root     root          4096 Dec 14 15:29 www

mpvader, 2016-05-23:
closed with latest bbb build, see [bbb change-log](https://github.com/victronenergy/venus/wiki/bbb-changelog).

closed 2016-05-23 by mpvader