victronenergy/venus #9: vrmlogger pulling cpu to 100% when no internet

mpvader, 2015-12-12:
vrmlogger is keeping the bbb really busy, because the http calls fail immediately, and there is no delay: it will retry immediately.

This is probably also happening on the CCGX, since I can't see any sleep or delay in the code (http_endpoint.py). 

I found this on a bbb with only the console connected. No ethernet/internet.

I'll have a look at it myself, since I am the one that wrote http_endpoint.py... Adding a one second delay in case of result==False already reduces vrmlogger cpu usage to 0%.

```
# cat /log/vrmlogger/current | tai64nlocal
2015-12-12 15:14:28.594017500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.612410500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.613819500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.632399500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.633813500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.651970500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.653262500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.671746500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.673033500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.691228500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.692518500 HTTPThread-http_endpoint: result: False, id: 11
2015-12-12 15:14:28.710999500 HTTPThread-vrmhttp: ('Connection aborted.', gaierror(-3, 'Temporary failure in name resolution'))
2015-12-12 15:14:28.712272500 HTTPThread-http_endpoint: result: False, id: 11
```

```
root@venus:~# top
Mem: 139508K used, 364852K free, 832K shrd, 11708K buff, 63072K cached
CPU:  73% usr  24% sys   0% nic   0% idle   0% io   0% irq   1% sirq
Load average: 1.47 1.18 0.67 2/150 1066
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1045   880 root     S    25740   5%  85% {vrmlogger.py} /usr/bin/python -u /opt
  827     1 root     S     6708   1%  11% /usr/sbin/connmand
  899   881 root     S     1720   0%   2% multilog t s99999 n8 /var/log/vrmlogge
  874   854 root     S     3192   1%   0% {vecan-dbus} /opt/victronenergy/vecan-
 1066   970 root     R     2964   1%   0% top
 ```

kpschaper, 2016-01-21:
I can confirm that the high cpu load also happens for the CCGX with fw v1.31 when not connected to the internet.
Cant find the same log entries in the log (/log/vrmlogger/current) though.

The vrmlogger is not publicly available right?
Otherwise it's maybe better to move this issue..

mpvader, 2016-02-02:
@kpschaper: yes, the repository that contains vrmlogger is not publicly available. But still I wanted to have the issue public. 

The issue is fixed in vrmlogger v2.40. And it will be included in CCGX v1.32. We are about to release the next test of that, v1.32~7. Jeroen's laptop is probably crunching through the build right now.

mpvader, 2016-02-02:
@kpschaper and of course thanks for double checking!

closed 2016-02-02 by mpvader