victronenergy/venus #10: bbb: boot type is always 0

mpvader, 2015-12-14:
For CCGX, we have a mechanism to read the boot type: this program runs before the watchdog is started:
https://git.victronenergy.com/ccgx/meta-ccgx/blob/danny_venus/recipes-extended/watchdog/watchdog-5.12/get_boot_type.c

Output (/tmp/last_boot_type) is then read by vrmlogger.py, and sent to vrmportal. So we can see watchdog reboots.

On BBB the contents of /tmp/last_boot_type is 0, both on normal power up and after a reboot.

Definition of possible values which I have is:
0=Old kernel, no boottype support;
1=Normal boot;
17=Watchdog reboot;
-1=Reading watchdog register failed (generated by get_boot_type);
-2=Reading tmp file failed (generated by vrmlogger.py)

I don't fully know all the details of this, but this probably means that reading boot type needs to be enabled in the kernel some how(?).

mpvader, 2015-12-15:
It is probably this patch:
https://github.com/victronenergy/linux/commit/ca732e7a00bdac30c0ba7e82aba31d8558d794fe

and could be that there is a more mainstream solution in sys or proc, up to you @scottellis

luctius, 2016-08-24:
Right now I get '-3' on a clean power-up and '0' after a soft reboot.

mpvader, 2016-08-24:
Yes, I see I missed that code when making the list above. I hadded -3 now. For background, see:
https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/vrmlogger.py#L96.

So to test this, disable the vrmlogger: 

    touch /service/vrmlogger/down

luctius, 2016-08-24:
Status at this moment is that the watchdog driver (omap_wdt.c) for this kernel does not implement an ioctl or sysfs interface to retrieve the boot status. Thus the generic wdt ioctl interface is called and that one will always reply '0' to an boot status (WDIOC_GETBOOTSTATUS) ioctl, hence our problem.

mpvader, 2016-09-14:
The CCGX (TI AM3517) boot statuses are detailed in chapter 4.12.2.11.3 of the [technical reference manual](http://www.ti.com/lit/ug/sprugr0c/sprugr0c.pdf).

Looking the VRM database, out of thousands of CCGX-es, there are only two reporting status 2. Which is Global software reset occurred.

mpvader, 2016-10-16:
@luctius: you posted an email to a mailing list somewhere for this, right? Please post a link to that email in the ML archive here in this issue.

luctius, 2016-11-08:
http://www.spinics.net/lists/linux-omap/msg132813.html

http://www.spinics.net/lists/linux-watchdog/msg10276.html

mpvader, 2016-12-06:
@luctius fixed it:
https://github.com/victronenergy/linux/commits/omap-wdt-bootstatus

waiting for test and inclusion in master.

mansr, 2017-01-11:
With those patches applied on the beaglebone, I get these values:
reset button: 0
cold boot: 1
reboot command: 2
watchdog reset: 16

mpvader, 2017-01-11:
@mansr thanks. So unfortunately those values do not entirely match up to CCGX list. Which has become our enum definition, also in the frontend (vrm portal & database).

So I'd like to keep, and extend our already existing list to keep all values unique. Which means shifting some of the Beaglebone boot-reasons around..

See table below for my proposal. Codes 3, 4 and 5 are the new ones.

code | description | generic | ccgx | beaglebone 
----- | ------------- | ------- | ----- | -------------------
-3 | tmp file already read, vrmlogger restarted!? | x | | |
-2 |Reading tmp file failed | x | | |
-1 | Reading watchdog register failed | x | | |
0 | Old kernel, no boottype support | | x | |
1 | Cold boot or reboot | | x | |
2 | Unreproducable reset on CCGX | | x | |
3 | Reset button | | | x |
4 | Cold boot | | | x |
5 | Reboot command | | | x |
17 | Watchdog reboot | | x | x |
29997 | Max load avg exceeded | x | | |
30012 | watchdog-ENOMEM | x | | |

Can you arrange this? For example by making a beaglebone specific implementation of get_boot_type.c; or another way you think is best.

In case you are wondering where 29997 and 30012 are coming from, see the [init script in that same recipe](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-extended/watchdog/files/init#L25).

Make sure to add a link to this issue in the commit.

And can you also add the necessary kernel changes to the linux branch we are currently at? While at that, pls see if you can review & add the commits made for #105 while at it.

jhofstee, 2017-01-20:
linux now reports the reboot in 2.01~10. We still need to translate the values though..

mansr, 2017-01-25:
Translation added to `get_boot_type`.

mpvader, 2017-01-29:
@jhofstee please add to master.
https://github.com/victronenergy/meta-victronenergy/commits/mans/boot-type

jhofstee, 2017-02-21:
this is included in v2.03. For completeness, I modified the program to not crash when the obtained value > 16 and added some comment how we are (ab)using the ioctl, see https://github.com/victronenergy/meta-victronenergy/commit/562f23bce6b12b3f6fe0138282abcb20d7038dc3.

closed 2017-02-21 by jhofstee