victronenergy/venus #11: gui: until remote support has been enabled once, status shows online

mpvader, 2015-12-15:
The paths in localsettings are created only after enabling remote support. So probably second-last line in this code needs a change:
```
		MbSwitch {
			name: qsTr("Remote support (SSH)")
			bind: "com.victronenergy.settings/Settings/System/RemoteSupport"
		}

		MbItemValue {
			description: qsTr("Remote support tunnel")
			item.value: remotePort.item.value !== 0 ? qsTr("Online") : qsTr("Offline")
		}
```

From: 
https://git.victronenergy.com/ccgx/gui/blob/master/qml/PageSettingsGeneral.qml#L60

And easily reproducible by loading a new image in a bbb and then browsing to that menu using Remote Console on LAN.

mpvader, 2015-12-15:
Log from localsettings:
```
root@venus:~# cat /log/localsettings/current | tai64nlocal
2015-12-14 20:07:50.856095500 localsettings v1.04 starting up
2015-12-14 20:07:51.068220500 2015-12-14 20:07:51,067 INFO Localsettings version is: 0x0104
2015-12-14 20:07:51.069649500 2015-12-14 20:07:51,068 INFO Localsettings PID is: 848
2015-12-14 20:07:51.079954500 2015-12-14 20:07:51,078 INFO Settings file /data/conf/settings.xml validated
2015-12-14 20:08:01.205268500 2015-12-14 20:08:01,204 INFO Setting /Settings/Fronius/KnownIPAddresses changed. Old: , New: 192.168.4.147
2015-12-14 20:08:01.215749500 2015-12-14 20:08:01,214 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/Phase. default:1, type:i, min:0, max: 0
2015-12-14 20:08:01.230620500 2015-12-14 20:08:01,229 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/Position. default:0, type:i, min:0, max: 0
2015-12-14 20:08:01.243803500 2015-12-14 20:08:01,242 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/CustomName. default:, type:s, min:0, max: 0
2015-12-14 20:08:01.257262500 2015-12-14 20:08:01,256 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/IsActive. default:1, type:i, min:0, max: 0
2015-12-14 20:08:01.285240500 2015-12-14 20:08:01,283 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/L1Energy. default:0.0, type:f, min:0.0, max: 1000000.0
2015-12-14 20:08:01.299079500 2015-12-14 20:08:01,297 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/L2Energy. default:0.0, type:f, min:0.0, max: 1000000.0
2015-12-14 20:08:01.313893500 2015-12-14 20:08:01,312 INFO Added new setting /Settings/Fronius/Inverters/I176_37823/L3Energy. default:0.0, type:f, min:0.0, max: 1000000.0
2015-12-14 20:08:01.333650500 2015-12-14 20:08:01,332 INFO Setting /Settings/Fronius/InverterIds changed. Old: , New: I176_37823
2015-12-14 20:09:14.918621500 2015-12-14 20:09:14,917 INFO Setting /Settings/System/RemoteSupport changed. Old: 0, New: 1
2015-12-14 20:09:15.163116500 2015-12-14 20:09:15,161 INFO Added new setting /Settings/System/RemoteSupportPort. default:0, type:i, min:0, max: 0
2015-12-14 20:09:15.231593500 2015-12-14 20:09:15,230 INFO Added new setting /Settings/System/RemoteVncPort. default:0, type:i, min:0, max: 0
2015-12-14 20:09:18.655328500 2015-12-14 20:09:18,653 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 0, New: 57025
```

mpvader, 2015-12-16:
I assume we have the same problem here:
https://git.victronenergy.com/ccgx/gui/blob/master/qml/PageSettingsRemoteConsole.qml#L48

jepefe, 2016-05-08:
Fixed in fix_remote_status branch

jhofstee, 2016-05-13:
this will fixed in gui v2.8.1, thanks

closed 2016-05-13 by jhofstee