victronenergy/venus #14: bbb: inserting usb-gps gives an error

mpvader, 2015-12-15:
What you see in below log is that I first disconnected the ve.direct usb interface, and then plugged in a usb gps. It shows an error, Jeroen hinted that it perhaps has to do with missing firmware?

```
Dec 14 21:22:27 venus user.info kernel: [ 4489.565541] usb 1-1: USB disconnect, device number 2
Dec 14 21:22:27 venus user.info kernel: [ 4489.671365] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
Dec 14 21:22:27 venus user.info kernel: [ 4489.681009] ftdi_sio 1-1:1.0: device disconnected
Dec 14 21:22:32 venus user.debug kernel: [ 4494.790360] musb-hdrc musb-hdrc.1.auto: VBUS_ERROR in a_wait_bcon (91, <VBusValid), retry #1, port1 00000104
Dec 14 21:22:32 venus user.info kernel: [ 4495.198405] usb 1-1: new full-speed USB device number 3 using musb-hdrc
Dec 14 21:22:33 venus user.info kernel: [ 4495.339480] usb 1-1: New USB device found, idVendor=067b, idProduct=2303
Dec 14 21:22:33 venus user.info kernel: [ 4495.346540] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Dec 14 21:22:33 venus user.info kernel: [ 4495.354053] usb 1-1: Product: USB-Serial Controller
Dec 14 21:22:33 venus user.info kernel: [ 4495.359182] usb 1-1: Manufacturer: Prolific Technology Inc.
Dec 14 21:22:33 venus user.debug kernel: [ 4495.367345] usb 1-1: no of_node; not parsing pinctrl DT
Dec 14 21:22:33 venus user.debug kernel: [ 4495.371972] usbtest 1-1:1.0: no of_node; not parsing pinctrl DT
Dec 14 21:22:33 venus user.info kernel: [ 4495.390450] usbcore: registered new interface driver pl2303
Dec 14 21:22:33 venus user.info kernel: [ 4495.396635] usbserial: USB Serial support registered for pl2303
Dec 14 21:22:33 venus user.debug kernel: [ 4495.402985] pl2303 1-1:1.0: no of_node; not parsing pinctrl DT
Dec 14 21:22:33 venus user.info kernel: [ 4495.403110] pl2303 1-1:1.0: pl2303 converter detected
Dec 14 21:22:33 venus user.debug kernel: [ 4495.422814] pl2303 ttyUSB0: no of_node; not parsing pinctrl DT
Dec 14 21:22:33 venus user.info kernel: [ 4495.424522] usb 1-1: pl2303 converter now attached to ttyUSB0
Dec 14 21:23:38 venus user.err kernel: [ 4560.398449] musb_ep_program 896: broken !rx_reinit, ep11 csr 0203
```

@scottellis up to you to decide if this is something that needs fixing or can be left as is.

The GPS works fine:
[image]

mpvader, 2016-07-13:
needs testing on a [recent image](https://github.com/victronenergy/venus/wiki/bbb-user-instructions), to see if the issue needs fixing or if it can be closed. Latest images might contain

luctius, 2016-08-16:
Current Log
Aug 16 11:31:28 beaglebone user.info kernel: [76074.735074] usb 1-1: new full-speed USB device number 2 using musb-hdrc
Aug 16 11:31:28 beaglebone user.info kernel: [76074.876747] usb 1-1: New USB device found, idVendor=0e8d, idProduct=3329
Aug 16 11:31:28 beaglebone user.info kernel: [76074.883818] usb 1-1: New USB device strings: Mfr=3, Product=4, SerialNumber=0
Aug 16 11:31:28 beaglebone user.info kernel: [76074.891352] usb 1-1: Product: GPS Receiver
Aug 16 11:31:28 beaglebone user.info kernel: [76074.895678] usb 1-1: Manufacturer: MTK
Aug 16 11:31:29 beaglebone user.debug kernel: [76074.926785] usb 1-1: no of_node; not parsing pinctrl DT
Aug 16 11:31:29 beaglebone user.debug kernel: [76074.935541] usbtest 1-1:1.0: no of_node; not parsing pinctrl DT
Aug 16 11:31:29 beaglebone user.debug kernel: [76074.942695] usbtest 1-1:1.1: no of_node; not parsing pinctrl DT
Aug 16 11:31:29 beaglebone user.debug kernel: [76074.970089] cdc_acm 1-1:1.0: no of_node; not parsing pinctrl DT
Aug 16 11:31:29 beaglebone user.debug kernel: [76074.970407] cdc_acm 1-1:1.1: no of_node; not parsing pinctrl DT
Aug 16 11:31:29 beaglebone user.info kernel: [76074.980456] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
Aug 16 11:31:29 beaglebone user.info kernel: [76074.991739] usbcore: registered new interface driver cdc_acm
Aug 16 11:31:29 beaglebone user.info kernel: [76074.997805] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

Looks fixes to me.

luctius, 2016-09-06:
ttyACM0 is added to serial_starter, and the original error does not occur any more. closed.

closed 2016-09-06 by luctius