victronenergy/venus #19: jethro: doing a fresh make fetch-all gives an error for bitbake and meta-yocto

mpvader, 2016-04-29:
```
builder@vebuild5:/media/ram/venus-jethro$ rm -rf ./sources
builder@vebuild5:/media/ram/venus-jethro$ ls -al
total 44
drwxrwxr-x 6 builder builder  360 Apr 29 16:34 .
drwxrwxrwt 7 builder builder  160 Apr 29 15:20 ..
drwxrwxr-x 8 builder builder  300 Apr 29 16:34 .git
-rw-rw-r-- 1 builder builder   45 Apr 29 15:20 .gitignore
-rw-rw-r-- 1 builder builder 2390 Apr 29 15:20 Makefile
-rw-rw-r-- 1 builder builder 4072 Apr 29 16:34 README
drwxrwxr-x 6 builder builder  160 Apr 29 15:22 build
lrwxrwxrwx 1 builder builder   14 Apr 29 15:20 conf -> configs/jethro
drwxrwxr-x 6 builder builder  120 Apr 29 15:20 configs
drwxrwxr-x 5 builder builder  100 Apr 29 15:33 deploy
-rwxrwxr-x 1 builder builder  593 Apr 29 15:20 git-fetch-remote.sh
-rwxrwxr-x 1 builder builder 1317 Apr 29 15:20 git-push-status.py
-rwxrwxr-x 1 builder builder  459 Apr 29 15:20 git-show-remote.sh
-rw-rw-r-- 1 builder builder 1143 Apr 29 15:22 metas.found
-rw-rw-r-- 1 builder builder  624 Apr 29 15:20 metas.whitelist
-rwxrwxr-x 1 builder builder   46 Apr 29 15:20 push-status.sh
-rwxrwxr-x 1 builder builder   47 Apr 29 15:20 repos
-rwxrwxr-x 1 builder builder  455 Apr 29 15:20 repos_cmd
builder@vebuild5:/media/ram/venus-jethro$ make fetch-all

--------------- sources/meta-swupdate - git@github.com:victronenergy/meta-swupdate.git -------------
Cloning into 'sources/meta-swupdate'...
remote: Counting objects: 256, done.
remote: Total 256 (delta 0), reused 0 (delta 0), pack-reused 256
Receiving objects: 100% (256/256), 36.99 KiB | 0 bytes/s, done.
Resolving deltas: 100% (73/73), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Branch master set up to track remote branch master from origin.

--------------- sources/meta-yocto - https://github.com/victronenergy/meta-yocto.git -------------
Cloning into 'sources/meta-yocto'...
remote: Counting objects: 7839, done.
remote: Total 7839 (delta 0), reused 0 (delta 0), pack-reused 7839
Receiving objects: 100% (7839/7839), 1.02 MiB | 870.00 KiB/s, done.
Resolving deltas: 100% (4833/4833), done.
Checking connectivity... done.
Branch jethro set up to track remote branch jethro from origin.
Switched to a new branch 'jethro'
error: the requested upstream branch 'upstream/jethro' does not exist
hint: 
hint: If you are planning on basing your work on an upstream
hint: branch that already exists at the remote, you may need to
hint: run "git fetch" to retrieve it.
hint: 
hint: If you are planning to push out a new local branch that
hint: will track its remote counterpart, you may want to use
hint: "git push -u" to set the upstream config as you push.

--------------- sources/meta-openembedded - https://github.com/victronenergy/meta-openembedded.git -------------
Cloning into 'sources/meta-openembedded'...
remote: Counting objects: 53881, done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 53881 (delta 66), reused 59 (delta 59), pack-reused 53793
Receiving objects: 100% (53881/53881), 24.78 MiB | 11.53 MiB/s, done.
Resolving deltas: 100% (30661/30661), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
Branch jethro_venus set up to track remote branch jethro from origin.

--------------- sources/meta-victronenergy-private - git@git.victronenergy.com:ccgx/meta-victronenergy-private.git -------------
Cloning into 'sources/meta-victronenergy-private'...
remote: Counting objects: 641, done.
remote: Compressing objects: 100% (622/622), done.
remote: Total 641 (delta 362), reused 0 (delta 0)
Receiving objects: 100% (641/641), 84.39 KiB | 0 bytes/s, done.
Resolving deltas: 100% (362/362), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Branch master set up to track remote branch master from origin.

--------------- sources/bitbake - https://github.com/victronenergy/bitbake.git -------------
Cloning into 'sources/bitbake'...
remote: Counting objects: 39010, done.
remote: Total 39010 (delta 0), reused 0 (delta 0), pack-reused 39010
Receiving objects: 100% (39010/39010), 19.09 MiB | 8.03 MiB/s, done.
Resolving deltas: 100% (25061/25061), done.
Checking connectivity... done.
Branch 1.28 set up to track remote branch 1.28 from origin.
Switched to a new branch '1.28'
error: the requested upstream branch 'upstream/1.28' does not exist
hint: 
hint: If you are planning on basing your work on an upstream
hint: branch that already exists at the remote, you may need to
hint: run "git fetch" to retrieve it.
hint: 
hint: If you are planning to push out a new local branch that
hint: will track its remote counterpart, you may want to use
hint: "git push -u" to set the upstream config as you push.

--------------- sources/openembedded-core - https://github.com/victronenergy/openembedded-core.git -------------
Cloning into 'sources/openembedded-core'...
remote: Counting objects: 266872, done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 266872 (delta 41), reused 23 (delta 23), pack-reused 266787
Receiving objects: 100% (266872/266872), 104.84 MiB | 23.74 MiB/s, done.
Resolving deltas: 100% (184585/184585), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
Branch jethro_venus set up to track remote branch jethro from origin.

--------------- sources/meta-victronenergy - https://github.com/victronenergy/meta-victronenergy.git -------------
Cloning into 'sources/meta-victronenergy'...
remote: Counting objects: 2585, done.
remote: Compressing objects: 100% (68/68), done.
remote: Total 2585 (delta 211), reused 181 (delta 181), pack-reused 2333
Receiving objects: 100% (2585/2585), 632.72 KiB | 501.00 KiB/s, done.
Resolving deltas: 100% (1155/1155), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Branch master set up to track remote branch master from origin.
```

mpvader, 2016-04-29:
this will fix it. Even without fully fetching the upstream remotes. Tested for danny and jethro.

```
matthijs@matthijs-VirtualBox:/media/venusbuilds/venus-mv/configs$ git show fix_fetch_all_error_for_upstream 
commit aa6ff084f892268555a5f40684e685110ab9082c
Author: mpvader <mvader@victronenergy.com>
Date:   Fri Apr 29 19:56:39 2016 +0000

    git-fetch-remote: fix error setting upstream branch
    
    fetching upstream first fixes this error:
    --------------- sources/bitbake - https://github.com/victronenergy/bitbake.git -------------
    Cloning into 'sources/bitbake'...
    remote: Counting objects: 39010, done.
    remote: Total 39010 (delta 0), reused 0 (delta 0), pack-reused 39010
    Receiving objects: 100% (39010/39010), 19.09 MiB | 12.70 MiB/s, done.
    Resolving deltas: 100% (25061/25061), done.
    Checking connectivity... done.
    Branch 1.28 set up to track remote branch 1.28 from origin.
    Switched to a new branch '1.28'
    fatal: Couldn't find remote ref upstream/1.28
    Unexpected end of command stream
    error: the requested upstream branch 'upstream/1.28' does not exist
    hint:
    hint: If you are planning on basing your work on an upstream
    hint: branch that already exists at the remote, you may need to
    hint: run "git fetch" to retrieve it.
    hint:
    hint: If you are planning to push out a new local branch that
    hint: will track its remote counterpart, you may want to use
    hint: "git push -u" to set the upstream config as you push.
    
    new output is:
    --------------- sources/bitbake - https://github.com/victronenergy/bitbake.git -------------
    Cloning into 'sources/bitbake'...
    remote: Counting objects: 39010, done.
    remote: Total 39010 (delta 0), reused 0 (delta 0), pack-reused 39010
    Receiving objects: 100% (39010/39010), 19.09 MiB | 2.25 MiB/s, done.
    Resolving deltas: 100% (25061/25061), done.
    Checking connectivity... done.
    Already on 'danny_venus'
    Your branch is up-to-date with 'origin/danny_venus'.
    From https://github.com/victronenergy/bitbake
     * branch            danny_venus -> FETCH_HEAD
    Branch danny_venus set up to track remote branch danny_venus from origin.

diff --git a/git-fetch-remote.sh b/git-fetch-remote.sh
index 2bc58d0..1771861 100755
--- a/git-fetch-remote.sh
+++ b/git-fetch-remote.sh
@@ -3,7 +3,7 @@
 printf "\n--------------- $1 - $2 -------------\n"
 
 if [ $# -lt 4 ]; then
-   echo "$0 directory fetch-url push-url (upstream|-)"
+   echo "$0 directory fetch-url push-url (upstream|-) (checkout-branch|-) (upstream-branch|-)"
    exit 1
 fi
 
@@ -17,11 +17,15 @@ if [ "$4" != "-" ]; then
    git --git-dir=$1/.git remote add upstream $4
 fi
 
-# optionally set an upstream repository
+# optionally checkout a branch
 if [ "$5" != "-" ]; then
    git --git-dir=$1/.git --work-tree=$1 checkout "$5"
 fi
 
+# optionally set the upstream branch
 if [ "$6" != "-" ]; then
+   if [ "$4" != "-" ]; then
+       git --git-dir=$1/.git --work-tree=$1 fetch $(echo $6 | tr // " ")
+   fi
    git --git-dir=$1/.git --work-tree=$1 branch -u "$6"
 fi
```

jhofstee, 2016-05-02:
fixed

closed 2016-05-02 by jhofstee