victronenergy/venus #22: don't generate locales

mpvader, 2016-05-03:
Should be possible, see below text from https://github.com/openembedded/openembedded-core/blob/master/meta/conf/local.conf.sample.extended
```
# ENABLE_BINARY_LOCALE_GENERATION controls the generation of binary locale
# packages at build time using qemu-native. Disabling it (by setting it to 0)
# will save some build time at the expense of breaking i18n on devices with
# less than 128MB RAM.
#ENABLE_BINARY_LOCALE_GENERATION = "1"

# Set GLIBC_GENERATE_LOCALES to the locales you wish to generate should you not
# wish to perform the time-consuming step of generating all LIBC locales.
# NOTE: If removing en_US.UTF-8 you will also need to uncomment, and set
# appropriate value for IMAGE_LINGUAS.
# WARNING: this may break localisation!
#GLIBC_GENERATE_LOCALES = "en_GB.UTF-8 en_US.UTF-8"
#IMAGE_LINGUAS ?= "en-gb"
```

And we already tried something, but that didn't work yet: https://github.com/victronenergy/venus/blob/master/build/conf/local.conf#L222 and
http://updates.victronenergy.com/feeds/ccgx/1.0/release/armv7a/

Besides causing build-time it also clutters.

mpvader, 2016-05-03:
At the CCGX we use:
```
root@ccgx:~# locale
LANG=en_GB
LC_CTYPE="en_GB"
LC_NUMERIC="en_GB"
LC_TIME="en_GB"
LC_COLLATE="en_GB"
LC_MONETARY="en_GB"
LC_MESSAGES="en_GB"
LC_PAPER="en_GB"
LC_NAME="en_GB"
LC_ADDRESS="en_GB"
LC_TELEPHONE="en_GB"
LC_MEASUREMENT="en_GB"
LC_IDENTIFICATION="en_GB"
LC_ALL=
```

jhofstee, 2016-12-30:
This should help: https://github.com/victronenergy/meta-victronenergy/commit/86639a435396fea4887fd493d6ad32dd707f2c63

closed 2016-12-30 by jhofstee