victronenergy/venus #30: bbb: mounting devpts on /dev/pts failed: No such file or directory

mpvader, 2016-05-23:
```
[    2.052226] usb 1-1: Product: 802.11 n WLAN
[    2.056612] usb 1-1: Manufacturer: Ralink
[    2.060829] usb 1-1: SerialNumber: 1.0
version 2.88 booting
mount: mounting devpts on /dev/pts failed: No such file or directory
[    2.921792] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
Starting udev
[    3.131456] udevd[659]: starting version 182
[    3.732287] cfg80211: Calling CRDA to update world regulatory domain
[    4.510193] usb 1-1: reset high-speed USB device number 2 using musb-hdrc
```

mpvader, 2016-05-23:
And later in the boot log there is this related error:
```
[    4.900064] usbcore: registered new interface driver rt2800usb
[    5.136170] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
bootlogd: cannot allocate pseudo tty: No such file or directory
[    5.272219] random: dd urandom read with 45 bits of entropy available
Populating dev cache
Mon May 23 19:18:18 UTC 2016
```

mpvader, 2016-07-13:
Though I can of course find out, I don't know what pts is, I don't know if the error hurts, and therefore I don't know if it needs fixing. Anyone who does, let me know so I can close this one.

luctius, 2016-08-16:
root@beaglebone:~# mount
/dev/root on / type ext4 (rw,relatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=243808k,nr_inodes=60952,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /var/volatile type tmpfs (rw,relatime)
/dev/mmcblk0p3 on /data type ext4 (rw,relatime,data=ordered)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620)
tracefs on /sys/kernel/debug/tracing type tracefs (rw,relatime)

'mount' says /dev/pts is mounted and I do not see an error about it in dmesg.

mpvader, 2016-08-17:
I checked, and you don't see them in dmesg because you are probably not building with the last commit in this branch here:
https://github.com/victronenergy/openembedded-core/tree/jethro_venus_mountall

I hadn't pushed that. But, since apparently it ends up being mounted anyway, perhaps later, we can just close this and leave it? Up to you.

mpvader, 2016-08-17:
ps. and then also we can delete that mountall branch and forget about it. I stumbled on to this trying to fix another partition not being mounted. And that has been resolved.

Also I had a short discussion on the openembedded mailinglist about this:
http://lists.openembedded.org/pipermail/openembedded-core/2016-May/122019.html

closed 2016-08-23 by luctius