victronenergy/venus #32: gui: keybaord input device error

mpvader, 2016-05-26:
```
root@beaglebone:~# cat /var/log/gui/current | tai64nlocal
2016-05-26 18:16:23.711451500 *** Starting gui ***
2016-05-26 18:16:23.711460500 *** headless device=1
2016-05-26 18:16:31.409428500 *** Localsettings is up, continuing...
2016-05-26 18:16:31.504323500 method return sender=:1.14 -> dest=:1.17 reply_serial=2
2016-05-26 18:16:31.504334500    int32 0
2016-05-26 18:16:31.610194500 *** Starting gui, with VNC enabled (VncLocal=1 VncInternet=0)
2016-05-26 18:16:32.355827500 QVNCServer created on port 5900
2016-05-26 18:16:32.418710500 Cannot open keyboard input device '/dev/input/by-path/platform-gpio-keys-event': No such file or directory
2016-05-26 18:16:32.475155500 GUI v2.8.4 started
2016-05-26 18:16:32.475166500 Built with Qt 4.8.7 running on 4.8.7
```

jhofstee, 2016-05-26:
root@ccgx:~# cat /service/gui/env/QWS_KEYBOARD 
linuxinput:/dev/input/by-path/platform-gpio-keys-event
root@ccgx:~#

jhofstee, 2016-05-26:
Does the bbb have a /dev/input/event0?

mpvader, 2016-05-31:
no it doesn't . Also doesn't need it: there are no buttons that users press and which then need to generate key events. I'll see how I can remove that QWS_KEYBOARD file from the list of installed files on the bbb, or keep it empty.

mpvader, 2016-05-31:
Fixed. See patch in [WIP-bbb4 in meta-ccgx](https://git.victronenergy.com/ccgx/meta-ccgx/commits/WIP-bbb4)

closed 2016-05-31 by mpvader