victronenergy/venus #33: gui: d-bus errors during startup

mpvader, 2016-05-26:
Errors on the bbb during startup. Not further looked into it yet.

```
root@beaglebone:~# cat /var/log/gui/current | tai64nlocal
2016-05-26 18:16:23.711451500 *** Starting gui ***
2016-05-26 18:16:23.711460500 *** headless device=1
2016-05-26 18:16:31.409428500 *** Localsettings is up, continuing...
2016-05-26 18:16:31.504323500 method return sender=:1.14 -> dest=:1.17 reply_serial=2
2016-05-26 18:16:31.504334500    int32 0
2016-05-26 18:16:31.610194500 *** Starting gui, with VNC enabled (VncLocal=1 VncInternet=0)
2016-05-26 18:16:32.355827500 QVNCServer created on port 5900
2016-05-26 18:16:32.418710500 Cannot open keyboard input device '/dev/input/by-path/platform-gpio-keys-event': No such file or directory
2016-05-26 18:16:32.475155500 GUI v2.8.4 started
2016-05-26 18:16:32.475166500 Built with Qt 4.8.7 running on 4.8.7
2016-05-26 18:16:32.685841500 LocalThemeDaemonClient: Looking for assets in "/opt/color-control/themes/ccgx"
2016-05-26 18:16:32.703881500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2016-05-26 18:16:32.743833500 TRACE [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs"
2016-05-26 18:16:33.465020500 INFO  CCGX version "v1.40" build on  "20160526173423"
2016-05-26 18:16:33.465032500 INFO  GUI built on "May 25 2016" at "13:22:23"
2016-05-26 18:16:33.489336500 TRACE [Service] marking "/service/dbus-motordrive" as a conflict of "/service/lg-resu-interface"
2016-05-26 18:16:33.489349500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/dbus-motordrive"
2016-05-26 18:16:33.489354500 TRACE [Service] marking "/service/dbus-valence" as a conflict of "/service/lg-resu-interface"
2016-05-26 18:16:33.489360500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/dbus-valence"
2016-05-26 18:16:33.489365500 TRACE [Service] marking "/service/vecan-dbus" as a conflict of "/service/lg-resu-interface"
2016-05-26 18:16:33.489503500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/vecan-dbus"
2016-05-26 18:16:33.574395500 INFO  [CmManager] Add technology "/net/connman/technology/ethernet"
2016-05-26 18:16:33.574407500 INFO  [CmManager] Add technology "/net/connman/technology/wifi"
2016-05-26 18:16:33.606140500 TRACE ColorControl::setDemoMode stop
2016-05-26 18:16:36.155872500 QDBusConnection: name 'com.victronenergy.system' had owner '' but we thought it was ':1.20'
2016-05-26 18:16:36.157883500 TRACE ColorControl::cmdlineFinished
2016-05-26 18:16:36.161452500 TRACE ColorControl::dbusDisplayOffTimeChanged:  600
2016-05-26 18:16:36.171944500 TRACE ColorControl::dbusReleaseTypeChanged:  0
2016-05-26 18:16:36.171955500 TRACE TimeZone::valueChanged:  "/UTC"
2016-05-26 18:16:36.171959500 TRACE ColorControl::dbusAccessLevelChanged:  1
2016-05-26 18:16:36.171963500 TRACE ColorControl::setAccessLevel:  1
2016-05-26 18:16:36.171966500 INFO  [Service] DBus service "/service/ssh-tunnel" changed: "com.victronenergy.settings/Settings/System/RemoteSupport" - "0"
2016-05-26 18:16:36.178899500 INFO  [Service] DBus service "/service/ssh-tunnel" changed: "com.victronenergy.settings/Settings/System/VncInternet" - ""
2016-05-26 18:16:36.178912500 INFO  [Service] DBus service "/service/sshd" changed: "com.victronenergy.settings/Settings/System/RemoteSupport" - "0"
2016-05-26 18:16:36.189358500 INFO  [Service] DBus service "/service/dbus-generator-starter" changed: "com.victronenergy.settings/Settings/Relay/Function" - "0"
2016-05-26 18:16:36.189371500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189375500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189378500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.189381500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189385500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189388500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.189391500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189527500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189531500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.189534500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189538500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189541500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.189544500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189548500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189550500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.189676500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.189680500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.189683500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.199356500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.199367500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.199370500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.199373500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.199377500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.199380500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.199383500 TRACE [BusItemCons::PropertiesChanged]
2016-05-26 18:16:36.199386500 TRACE Item:  "Text" : ""
2016-05-26 18:16:36.199389500 TRACE Item:  "Value" : ""
2016-05-26 18:16:36.199519500 INFO  [Service] DBus service "/service/dbus-modbustcp" changed: "com.victronenergy.settings/Settings/Services/Modbus" - "0"
2016-05-26 18:16:36.199528500 INFO  [Service] DBus service "/service/vegetty" changed: "com.victronenergy.settings/Settings/Services/Console" - "0"
2016-05-26 18:16:36.199534500 INFO  [Service] DBus service "/service/lg-resu-interface" changed: "com.victronenergy.settings/Settings/Services/LgResu" - "0"
2016-05-26 18:16:36.199540500 INFO  [Service] DBus service "/service/vrmpubnub" changed: "com.victronenergy.settings/Settings/Services/Vrmpubnub" - "0"
[... 33 lines cut ...]
2016-05-26 18:16:36.433085500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.441184500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.442849500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.444391500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.450039500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.451691500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.453538500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.455112500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.456598500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.461294500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.462935500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.464523500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
2016-05-26 18:16:36.466609500 Error value: QDBusError("org.freedesktop.DBus.Error.LimitsExceeded", "The maximum number of pending replies per connection has been reached") "com.victronenergy.system"
[... cut ...]
2016-05-26 18:16:38.801888500 TRACE serviceOwnerChanged "com.victronenergy.fronius" "" ":1.22"
2016-05-26 18:16:43.531894500 TRACE Updater::runCheckUpdateScript:
2016-05-26 18:16:50.339613500 TRACE ColorControl::opkgUpdated:  "/tmp/opkg-updates"
2016-05-26 18:16:50.339625500 TRACE ColorControl::opkgUpdated: numOfUpdates =  0
2016-05-26 18:16:53.738724500 TRACE ColorControl::opkgUpdated:  "/tmp/opkg-updates"
2016-05-26 18:16:53.738731500 TRACE ColorControl::opkgUpdated: numOfUpdates =  0
2016-05-26 18:16:53.738735500 TRACE Updater::opkgCheckingForUpdatesFinished:
root@beaglebone:~#
```

mpvader, 2016-07-13:
Needs testing on a [recent image](https://github.com/victronenergy/venus/wiki/bbb-user-instructions), to see if the issue is still there. And if it is, it needs to be further looked into.

mpvader, 2016-10-22:
this still happens in latest image (v1.80~16 20161021093731):. seems to do no harm though.

and i haven't looked any further into it.
```
2016-10-22 08:04:16.512598500 *** Starting gui, with VNC enabled (VncLocal=1 VncInternet=1)
2016-10-22 08:04:16.964510500 QVNCServer created on port 5900
2016-10-22 08:04:17.033567500 GUI v2.9.9 started
2016-10-22 08:04:17.033578500 Built with Qt 4.8.7 running on 4.8.7
2016-10-22 08:04:17.285668500 LocalThemeDaemonClient: Looking for assets in "/opt/color-control/themes/ccgx"
2016-10-22 08:04:17.298717500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2016-10-22 08:04:17.643753500 TRACE [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs"
2016-10-22 08:04:18.108130500 INFO  CCGX version "v1.80~16" build on  "20161021093731"
2016-10-22 08:04:18.108142500 INFO  GUI built on "Oct 21 2016" at "09:28:22"
2016-10-22 08:04:18.135495500 TRACE [Service] marking "/service/dbus-motordrive" as a conflict of "/service/lg-resu-interface"
2016-10-22 08:04:18.135508500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/dbus-motordrive"
2016-10-22 08:04:18.135514500 TRACE [Service] marking "/service/dbus-valence" as a conflict of "/service/lg-resu-interface"
2016-10-22 08:04:18.135519500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/dbus-valence"
2016-10-22 08:04:18.135524500 TRACE [Service] marking "/service/vecan-dbus" as a conflict of "/service/lg-resu-interface"
2016-10-22 08:04:18.135669500 TRACE [Service] marking "/service/lg-resu-interface" as a conflict of "/service/vecan-dbus"
2016-10-22 08:04:18.277847500 INFO  [CmManager] Add technology "/net/connman/technology/ethernet"
2016-10-22 08:04:18.290982500 DEBUG [CmManager] Add service "/net/connman/service/ethernet_b0d5ccfd7052_cable"
2016-10-22 08:04:20.975601500 unknown service com.victronenergy.system
2016-10-22 08:04:21.910471500 QDBusConnection: name 'com.victronenergy.temperature.builtin_adc3_di1' had owner ':1.34' but we thought it was ''
2016-10-22 08:04:21.912615500 QDBusConnection: name 'com.victronenergy.system' had owner '' but we thought it was ':1.17'
2016-10-22 08:04:21.914429500 QDBusConnection: name 'com.victronenergy.qwacs' had owner '' but we thought it was ':1.28'
2016-10-22 08:04:21.926489500 TRACE serviceOwnerChanged "com.victronenergy.temperature.builtin_adc3_di1" ":1.34" ""
2016-10-22 08:04:21.945683500 TRACE serviceOwnerChanged "com.victronenergy.system" "" ":1.17"
2016-10-22 08:04:22.027830500 TRACE [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs"
2016-10-22 08:04:22.145114500 TRACE serviceOwnerChanged "com.victronenergy.qwacs" "" ":1.28"
2016-10-22 08:04:22.184087500 TRACE serviceOwnerChanged "com.victronenergy.logger" "" ":1.36"
2016-10-22 08:04:22.184098500 TRACE ColorControl::dbusDisplayOffTimeChanged:  600
2016-10-22 08:04:22.193865500 TRACE ColorControl::dbusReleaseTypeChanged:  2
2016-10-22 08:04:22.193876500 INFO  [Service] DBus service "/service/vegetty" changed: "com.victronenergy.settings/Settings/Services/Console" - "0"
2016-10-22 08:04:22.193882500 INFO  [Service] DBus service "/service/dbus-modbustcp" changed: "com.victronenergy.settings/Settings/Services/Modbus" - "0"
2016-10-22 08:04:22.193889500 INFO  [Service] DBus service "/service/mosquitto" changed: "com.victronenergy.settings/Settings/Services/Mqtt" - "0"
2016-10-22 08:04:22.194071500 INFO  [Service] DBus service "/service/dbus-mqtt" changed: "com.victronenergy.settings/Settings/Services/Mqtt" - "0"
2016-10-22 08:04:22.194078500 INFO  [Service] DBus service "/service/dbus-motordrive" changed: "com.victronenergy.settings/Settings/Services/OceanvoltMotorDrive" - "0"
2016-10-22 08:04:22.194084500 INFO  [Service] DBus service "/service/dbus-valence" changed: "com.victronenergy.settings/Settings/Services/OceanvoltValence" - "0"
2016-10-22 08:04:22.194129500 INFO  [Service] DBus service "/service/lg-resu-interface" changed: "com.victronenergy.settings/Settings/Services/LgResu" - "0"
2016-10-22 08:04:22.194136500 INFO  [Service] DBus service "/service/vecan-dbus" changed: "com.victronenergy.settings/Settings/Services/VeCan" - "1"
2016-10-22 08:04:22.194263500 INFO  [Service] Start "/service/vecan-dbus"
2016-10-22 08:04:22.194267500 TRACE [Service] stopping conflicting service "/service/lg-resu-interface"
2016-10-22 08:04:22.231286500 INFO  [Service] DBus service "/service/sshd" changed: "com.victronenergy.settings/Settings/System/RemoteSupport" - "1"
2016-10-22 08:04:22.231300500 INFO  [Service] Start "/service/sshd"
2016-10-22 08:04:22.259467500 INFO  [Service] DBus service "/service/dbus-generator-starter" changed: "com.victronenergy.settings/Settings/Relay/Function" - "0"
2016-10-22 08:04:22.259481500 INFO  [Service] DBus service "/service/dbus-pump-starter" changed: "com.victronenergy.settings/Settings/Relay/Function" - "0"
2016-10-22 08:04:22.259488500 INFO  [Service] Stop "/service/dbus-pump-starter"
2016-10-22 08:04:22.280160500 INFO  [Service] DBus service "/service/ssh-tunnel" changed: "com.victronenergy.settings/Settings/System/RemoteSupport" - "1"
2016-10-22 08:04:22.280174500 INFO  [Service] DBus service "/service/ssh-tunnel" changed: "com.victronenergy.settings/Settings/System/VncInternet" - ""
2016-10-22 08:04:22.280180500 INFO  [Service] Start "/service/ssh-tunnel"
[... cut ...]
2016-10-22 08:04:27.154525500 TRACE serviceOwnerChanged "com.victronenergy.logger" ":1.36" ""
2016-10-22 08:04:27.275538500 TRACE serviceOwnerChanged "com.victronenergy.battery.ttyO4" "" ":1.50"
2016-10-22 08:04:28.794595500 TRACE serviceOwnerChanged "com.victronenergy.system" "" ":1.49"
2016-10-22 08:04:30.919344500 TRACE serviceOwnerChanged "com.victronenergy.logger" "" ":1.52"
2016-10-22 08:05:04.705562500 TRACE serviceOwnerChanged "com.victronenergy.grid.cgwacs_ttyUSB0_di30_mb1" "" ":1.55"
```

mpvader, 2016-11-08:
@jhofstee: above errors are no longer visible in gui v3.0.2 on the beaglebone. Is that explainable by your changes in the gui? So we can close this issue now?

jhofstee, 2016-12-08:
yes, could chance, the gui switched to qitems, which differ a bit BusItemCons, it will:
  - not request a value from a offline service put postpone it till it is on-bus again
  - merge multiple requests for the same value into a single one
The old BusItemCons would blindly forward every request to the dbus when it didn't have a cached one.

So closing this issue, I haven't seen it recently either.

closed 2016-12-08 by jhofstee