victronenergy/venus #34: kernel panic after insterting usb wifi

mpvader, 2016-05-28:
besides the main issue, two others:
- and no reset afterwards: watchdog not yet running ok?
- and shouldn't the kernel initiate reboot after panicking?

```
root@beaglebone:~# [12768.419617] usb 1-1: new high-speed USB device number 2 using musb-hdrc
[12768.575513] usb 1-1: New USB device found, idVendor=0586, idProduct=341e
[12768.582586] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12768.590094] usb 1-1: Product: 802.11 n WLAN
[12768.594481] usb 1-1: Manufacturer: Ralink
[12768.598682] usb 1-1: SerialNumber: 1.0
[12768.736212] cfg80211: Calling CRDA to update world regulatory domain
[12768.877037] cfg80211: World regulatory domain updated:
[12768.882581] cfg80211:  DFS Master region: unset
[12768.887154] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[12768.897410] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[12768.905834] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[12768.914248] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[12768.922664] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[12768.932622] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[12768.942577] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[12768.951075] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[12768.959481] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[12769.129587] usb 1-1: reset high-speed USB device number 2 using musb-hdrc
[12769.281856] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[12769.309186] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0005 detected
[12769.379835] usbcore: registered new interface driver rt2800usb
[12769.398813] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[12769.450057] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.29
[12769.906083] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

root@beaglebone:~#
root@beaglebone:~# ifconfig
eth0      Link encap:Ethernet  HWaddr EC:24:B8:9E:18:DF
          inet addr:192.168.51.62  Bcast:192.168.51.255  Mask:255.255.255.0
          inet6 addr: fe80::ee24:b8ff:fe9e:18df/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:45951 errors:0 dropped:4 overruns:0 frame:0
          TX packets:5566 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12128812 (11.5 MiB)  TX bytes:1440693 (1.3 MiB)
          Interrupt:173

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1051 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1051 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:777913 (759.6 KiB)  TX bytes:777913 (759.6 KiB)

wlan0     Link encap:Ethernet  HWaddr 10:7B:EF:C7:79:A4
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@beaglebone:~# ifconfig
eth0      Link encap:Ethernet  HWaddr EC:24:B8:9E:18:DF
          inet addr:192.168.51.62  Bcast:192.168.51.255  Mask:255.255.255.0
          inet6 addr: fe80::ee24:b8ff:fe9e:18df/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:45954 errors:0 dropped:4 overruns:0 frame:0
          TX packets:5566 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12129239 (11.5 MiB)  TX bytes:1440693 (1.3 MiB)
          Interrupt:173

[... 55 lines cut ...]
[12785.352696] 7f40: 00000000 00000001 dead4ead ffffffff ffffffff c0731ad4 c0868ca4 00000000
[12785.361273] 7f60: c05ef984 de097f64 de097f64 00000000 00000001 dead4ead ffffffff ffffffff
[12785.369849] 7f80: c0731ad4 00000000 00000000 c05ef984 de097f90 de097f90 de097fac de0287c0
[12785.378425] 7fa0: c004ea08 00000000 00000000 c000f238 00000000 00000000 00000000 00000000
[12785.387000] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[12785.395575] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 78093675 cf8e62e9
[12785.404160] [<c0066c74>] (__lock_acquire) from [<c006912c>] (lock_acquire+0xac/0x12c)
[12785.412383] [<c006912c>] (lock_acquire) from [<c04c1a18>] (_raw_spin_lock_irqsave+0x3c/0x50)
[12785.421281] [<c04c1a18>] (_raw_spin_lock_irqsave) from [<bf18db9c>] (rt2x00queue_get_entry+0x24/0x7c [rt2x00lib])
[12785.432071] [<bf18db9c>] (rt2x00queue_get_entry [rt2x00lib]) from [<bf1be9bc>] (rt2800usb_txstatus_pending+0x1c/0x70 [rt2800usb])
[12785.444301] [<bf1be9bc>] (rt2800usb_txstatus_pending [rt2800usb]) from [<bf1beb64>] (rt2800usb_tx_sta_fifo_read_completed+0xa4/0x150 [rt2800usb])
[12785.457993] [<bf1beb64>] (rt2800usb_tx_sta_fifo_read_completed [rt2800usb]) from [<bf1b7418>] (rt2x00usb_register_read_async_cb+0x20/0x48 [rt2x00usb])
[12785.472139] [<bf1b7418>] (rt2x00usb_register_read_async_cb [rt2x00usb]) from [<c0301aec>] (__usb_hcd_giveback_urb+0x94/0x134)
[12785.483997] [<c0301aec>] (__usb_hcd_giveback_urb) from [<c0301c14>] (usb_giveback_urb_bh+0x88/0xc4)
[12785.493497] [<c0301c14>] (usb_giveback_urb_bh) from [<c00362a8>] (tasklet_action+0x68/0xb8)
[12785.502258] [<c00362a8>] (tasklet_action) from [<c00358d4>] (__do_softirq+0x104/0x364)
[12785.510564] [<c00358d4>] (__do_softirq) from [<c0035b64>] (run_ksoftirqd+0x30/0x40)
[12785.518602] [<c0035b64>] (run_ksoftirqd) from [<c0051edc>] (smpboot_thread_fn+0x108/0x154)
[12785.527282] [<c0051edc>] (smpboot_thread_fn) from [<c004eae4>] (kthread+0xdc/0xf0)
[12785.535229] [<c004eae4>] (kthread) from [<c000f238>] (ret_from_fork+0x14/0x3c)
[12785.542804] Code: e1a00004 e28dd054 e8bd8ff0 e59f3e08 (e5962000)
[12785.549199] ---[ end trace e7152e0c5ad44463 ]---
[12785.554040] Kernel panic - not syncing: Fatal exception in interrupt
[12785.560705] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

mpvader, 2016-05-28:
reproduced it by rebooting again, without wifi inserted. and then inserting wifi again as soon as the login prompt showed up.

```
** Unable to read file boot.scr **
reading uEnv.txt
727 bytes read in 5 ms (141.6 KiB/s)
Loaded environment from uEnv.txt
Importing environment from mmc ...
Running uenvcmd ...
30424 bytes read in 32 ms (927.7 KiB/s)
Loaded bbb-venus.dtb
3652016 bytes read in 236 ms (14.8 MiB/s)
Kernel image @ 0x82000000 [ 0x000000 - 0x37b9b0 ]
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8fff5000, end 8ffff6d7 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15-venus (matthijs@matthijs-VirtualBox) (gcc version 5.2.0 (GCC) ) #1 Tue May 24 15:38:07 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: TI AM335x BeagleBone Black
[    0.000000] cma: Reserved 16 MiB at 0x9f000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 487632K/524288K available (4993K kernel code, 322K rwdata, 1748K rodata, 264K init, 8172K bss, 20272K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff000000   ( 488 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc069d820   (6743 kB)
[    0.000000]       .init : 0xc069e000 - 0xc06e0000   ( 264 kB)
[    0.000000]       .data : 0xc06e0000 - 0xc0730938   ( 323 kB)
[    0.000000]        .bss : 0xc0730938 - 0xc0f2bb58   (8173 kB)
[    0.000000] Running RCU self tests
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000013] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000034] clocksource timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000076] OMAP clocksource: timer1 at 24000000 Hz
[    0.000732] Console: colour dummy device 80x30
[    0.000784] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.000791] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.000797] ... MAX_LOCK_DEPTH:          48
[    0.000803] ... MAX_LOCKDEP_KEYS:        8191
[    0.000809] ... CLASSHASH_SIZE:          4096
[    0.000815] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.000821] ... MAX_LOCKDEP_CHAINS:      65536
[    0.000827] ... CHAINHASH_SIZE:          32768
[    0.000833]  memory used by lock dependency info: 5167 kB
[    0.000839]  per task-struct memory footprint: 1152 bytes
[    0.000859] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.078963] pid_max: default: 32768 minimum: 301
[    0.079169] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.079181] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.081403] CPU: Testing write buffer coherency: ok
[    0.082586] Setting up static identity map for 0x80008200 - 0x80008258
[    0.086578] devtmpfs: initialized
[... 286 lines cut ...]
[   48.802420] [<c00362a8>] (tasklet_action) from [<c00358d4>] (__do_softirq+0x104/0x364)
[   48.810726] [<c00358d4>] (__do_softirq) from [<c0035e68>] (irq_exit+0xfc/0x130)
[   48.818403] [<c0035e68>] (irq_exit) from [<c00729cc>] (__handle_domain_irq+0x58/0xa8)
[   48.826620] [<c00729cc>] (__handle_domain_irq) from [<c0009458>] (omap_intc_handle_irq+0xc8/0xd0)
[   48.835930] [<c0009458>] (omap_intc_handle_irq) from [<c04c2344>] (__irq_svc+0x44/0x5c)
[   48.844320] Exception stack(0xde257c70 to 0xde257cb8)
[   48.849614] 7c60:                                     00000001 0000000d c0eddd78 de24c040
[   48.858189] 7c80: 00000000 de7fb2d0 de7fb2d8 00000000 00000000 00000000 c072eda4 60070013
[   48.866765] 7ca0: c0759de0 de257cb8 c0731c34 c0069144 80070013 ffffffff
[   48.873704] [<c04c2344>] (__irq_s75e4c>] (kernfs_remove_by_name_ns+0x3c/0x84)
[   48.899175] [<c0175e4c>] (kernfs_remove_by_name_ns) from [<c01776d4>] (sysfs_remove_file_from_group+0x3c/0x60)
[   48.909671] [<c01776d4>] (sysfs_remove_file_from_group) from [<c030c148>] (usb_remove_sysfs_dev_files+0x40/0x50)
[   48.920345] [<c030c148>] (usb_remove_sysfs_dev_files) from [<c02fa610>] (usb_bus_notify+0x88/0x90)
[   48.929745] [<c02fa610>] (usb_bus_notify) from [<c004fa10>] (notifier_call_chain+0x4c/0x8c)
[   48.938505] [<c004fa10>] (notifier_call_chain) from [<c004fd4c>] (__blocking_notifier_call_chain+0x48/0x60)
[... cut ...]
[   48.993531] [<c0049560>] (process_one_work) from [<c004995c>] (worker_thread+0x4c/0x4fc)
[   49.002022] [<c004995c>] (worker_thread) from [<c004eae4>] (kthread+0xdc/0xf0)
[   49.009602] [<c004eae4>] (kthread) from [<c000f238>] (ret_from_fork+0x14/0x3c)
[   49.017176] Code: e1a00004 e28dd054 e8bd8ff0 e59f3e08 (e5962000)
[   49.023573] ---[ end trace 9549616f3d814bd7 ]---
[   49.028415] Kernel panic - not syncing: Fatal exception in interrupt
[   49.035079] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

mpvader, 2016-05-28:
issue happens when both ethernet and wifi are enabled at same time. 

booting up with ethernet, without wifi: OK. But once wifi inserted, the panic accurs.

booting up with without ethernet, without wifi, ok. and then insterting wifi, still ok. then inserting ethernet: panic. See this one:

```
[    1.184675] loop: module loaded
[    1.189868] vcan: Virtual CAN interface driver
[    1.194544] slcan: serial line CAN interface driver
[    1.199682] slcan: 10 dynamic interface channels.
[    1.204624] CAN device driver interface
[    1.211204] c_can_platform 481cc000.can: c_can_platform device registered (regs=fa1cc000, irq=162)
[    1.222435] c_can_platform 481d0000.can: c_can_platform device registered (regs=fa1d0000, irq=163)
[    1.233340] usbcore: registered new interface driver asix
[    1.239124] usbcore: registered new interface driver ax88179_178a
[    1.245664] usbcore: registered new interface driver cdc_ether
[    1.251942] usbcore: registered new interface driver smsc95xx
[    1.258067] usbcore: registered new interface driver net1080
[    1.264134] usbcore: registered new interface driver cdc_subset
[    1.270463] usbcore: registered new interface driver zaurus
[    1.276488] usbcore: registered new interface driver cdc_ncm
[    1.292606] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.299474] ehci-omap: OMAP-EHCI Host Controller driver
[    1.305376] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.312096] usbcore: registered new interface driver cdc_wdm
[    1.318184] usbcore: registered new interface driver usbtest
[    1.326753] 47401300.usb-phy supply vcc not found, using dummy regulator
[    1.343298] 47401b00.usb-phy supply vcc not found, using dummy regulator
[    1.355479] musb-hdrc musb-hdrc.1.auto: MUSB HDRC host driver
[    1.364760] musb-hdrc musb-hdrc.1.auto: new USB bus registered, assigned bus number 1
[    1.374842] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.382016] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.389605] usb usb1: Product: MUSB HDRC host driver
[    1.394811] usb usb1: Manufacturer: Linux 4.1.15-venus musb-hcd
[    1.401031] usb usb1: SerialNumber: musb-hdrc.1.auto
[    1.410384] hub 1-0:1.0: USB hub found
[    1.415166] hub 1-0:1.0: 1 port detected
[    1.426121] mousedev: PS/2 mouse device common for all mice
[    1.435084] i2c /dev entries driver
[    1.438831] Driver for 1-wire Dallas network protocol.
[    1.449108] omap_wdt: OMAP Watchdog Timer Rev 0x01: initial timeout 60 sec
[    1.458629] omap_hsmmc 48060000.mmc: Got CD GPIO
[    1.547110] mmc0: host does not support reading read-only switch, assuming write-enable
[    1.559009] ledtrig-cpu: registered to indicate activity on CPUs
[    1.565889] usbcore: registered new interface driver usbhid
[    1.571764] usbhid: USB HID core driver
[    1.576789] oprofile: using arm/armv7
[    1.581289] Initializing XFRM netlink socket
[    1.586078] NET: Registered protocol family 10
[    1.593289] mmc0: new high speed SDHC card at address b368
[    1.604435] mmcblk0: mmc0:b368       7.45 GiB
[    1.614599] sit: IPv6 over IPv4 tunneling driver
[    1.621778] NET: Registered protocol family 17
[    1.626520] NET: Registered protocol family 15
[    1.631257] can: controller area network core (rev 20120528 abi 9)
[    1.637860] NET: Registered protocol family 29
[    1.642603] can: raw protocol (rev 20120528)
[    1.647212] can: broadcast manager protocol (rev 20120528 t)
[    1.653659] Key type dns_resolver registered
[    1.658309] omap_voltage_late_init: Voltage driver support not added
[    1.665985] ThumbEE CPU extension supported.
[    1.672192]  mmcblk0: p1 p2 p3
[    1.724910] tps65217 0-0024: TPS65217 ID 0xe version 1.2
[    1.730691] omap_i2c 44e0b000.i2c: bus 0 rev0.11 at 400 kHz
[    1.736921] mmc1: MAN_BKOPS_EN bit is not set
[    1.761290] mmc1: new high speed MMC card at address 0001
[    1.768418] mmcblk1: mmc1:0001 MMC04G 3.65 GiB
[    1.773528] mmcblk1boot0: mmc1:0001 MMC04G partition 1 1.00 MiB
[    1.780211] mmcblk1boot1: mmc1:0001 MMC04G partition 2 1.00 MiB
[... 196 lines cut ...]
[   60.895913] 7f40: 00000000 00000001 dead4ead ffffffff ffffffff c0731ad4 c0868ca4 00000000
[   60.904488] 7f60: c05ef984 de097f64 de097f64 00000000 00000001 dead4ead ffffffff ffffffff
[   60.913063] 7f80: c0731ad4 00000000 00000000 c05ef984 de097f90 de097f90 de097fac de0287c0
[   60.921639] 7fa0: c004ea08 00000000 00000000 c000f238 00000000 00000000 00000000 00000000
[   60.930213] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   60.938789] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 78093675 cf8e62e9
[   60.947373] [<c0066c74>] (__lock_acquire) from [<c006912c>] (lock_acquire+0xac/0x12c)
[   60.955597] [<c006912c>] (lock_acquire) from [<c04c1a18>] (_raw_spin_lock_irqsave+0x3c/0x50)
[   60.964495] [<c04c1a18>] (_raw_spin_lock_irqsave) from [<bf195b9c>] (rt2x00queue_get_entry+0x24/0x7c [rt2x00lib])
[   60.975285] [<bf195b9c>] (rt2x00queue_get_entry [rt2x00lib]) from [<bf1c69bc>] (rt2800usb_txstatus_pending+0x1c/0x70 [rt2800usb])
[   60.987514] [<bf1c69bc>] (rt2800usb_txstatus_pending [rt2800usb]) from [<bf1c6b64>] (rt2800usb_tx_sta_fifo_read_completed+0xa4/0x150 [rt2800usb])
[   61.001206] [<bf1c6b64>] (rt2800usb_tx_sta_fifo_read_completed [rt2800usb]) from [<bf1bf418>] (rt2x00usb_register_read_async_cb+0x20/0x48 [rt2x00usb])
[... cut ...]
[   61.061813] [<c0035b64>] (run_ksoftirqd) from [<c0051edc>] (smpboot_thread_fn+0x108/0x154)
[   61.070491] [<c0051edc>] (smpboot_thread_fn) from [<c004eae4>] (kthread+0xdc/0xf0)
[   61.078437] [<c004eae4>] (kthread) from [<c000f238>] (ret_from_fork+0x14/0x3c)
[   61.086012] Code: e1a00004 e28dd054 e8bd8ff0 e59f3e08 (e5962000)
[   61.092407] ---[ end trace 4fad142e998efc27 ]---
[   61.097249] Kernel panic - not syncing: Fatal exception in interrupt
[   61.103913] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

jhofstee, 2016-05-28:
Hello Matthijs,

If I have to guess it is caused by adding the CRDA package.
I am sure adding the package causes problems, but I don't
recall what exactly... it might be this ;)

Regards,
Jeroen

mpvader, 2016-05-28:
Thanks, I'll try without crda. And in case that helps I'll follow instructions for the alternative:
https://wireless.wiki.kernel.org/en/developers/regulatory/crda

mpvader, 2016-07-13:
This issue made me leave crda out of the venus-bbb builds for now. Causing annoying crda update messages in the console after boot.

If anyone reading this can take the time to find out if it is really necessary / worthwhile to further look into this, please comment.

I see three options:
1) Keep crda out, and do nothing else <- least amount of work! any technical disadvantages of this?
2) Fix crda issue
3) Use the alternative instead of crda itself

mpvader, 2016-07-13:
correction: actually crda is still in the WIP-bbbx branches.

mpvader, 2016-09-06:
close: unable to reproduce. When this happened, the bbb was powered from a CCGX USB port. Which is not the strongest of USB ports with regards to its power supply. Jeroen has also seen it a long time ago, at that time his bbb was powered via USB as well, from one of the ports on his laptop.

closed 2016-09-06 by mpvader