victronenergy/venus #36: Test Raspbian and Debian packages

mpvader, 2016-06-14:
Quite a while ago we have made it possible to create packages out of the Venus system which can be installed on Raspbian and Debian.

See [this email](https://groups.google.com/forum/m/#!topic/victron-dev-venus/CzQlEF10biU) on the dev-venus mailinglist for an introduction.

Current status of this is that we can easily build them, but don't have time to test them and therefore nobody uses them.

A waste, as there are many people that I know would like to run these drivers. Especially as it is quite easy to have the data available not only on d-bus but also on mqtt. See this repo for the mqtt work:
https://github.com/victronenergy/dbus-mqtt

official feeds are here:
http://updates.victronenergy.com/feeds/debian/
http://updates.victronenergy.com/feeds/raspbian/

Recent builds are here:
https://build-feeds.victronenergy.com/mvader/venus.git/WIP-bbb4/118ee417/

Besides testing it, part of this issue is also to make documentation. So people don't have to go through the mailing list to get started. Preferably on a page in the [venus wiki](https://github.com/victronenergy/venus/wiki)

izak, 2016-06-14:
I noticed that the process used is to build a deb out of the binaries that are already compiled as part of building a venus image. Would it be useful if instead it had a proper debianisation, using something like debhelper or cdbs? I could probably whip that up in a few hours?

mpvader, 2016-06-14:
It does compile specifically for debian / raspbian. It does not use binaries out of Openembedded. The thing that it does use from Openembedded are the recipes.

We want to do it that way to keep only having to maintain those recipes in Venus. Without having to maintain some meta data for Debian package making as well..

FYI: The code used to do this is here: https://github.com/jhofstee/meta-bin-deb

See [gitlab-ci file](https://github.com/victronenergy/venus/blob/master/.gitlab-ci.yml#L67) for how such build is triggered.

A question to keep focus: why is it necessary to improve this?

izak, 2016-06-18:
I think my suggestion to do a "proper" debianisation might stem from my ignorance as to how this really works. Taking in the whole build architecture is somewhat of a steep learning curve. I already found @jhofstee 's repo, and after I looked into the way that debian itself does cross-compiling of packages (there's more than one way), I figure there is no improvement to be made here by attempting to aim for some kind of standard.

If the requirement is only to test, I can do that, at least for Jessie. We tested the mk2dbus binary at work and it does work, there was just something odd with our dbus setup, so a bit more work is required. I just need to prepare a fresh Jessie image.

izak, 2016-06-23:
Feedback time.

1. The build process does not appear to correctly generate package index files (Packages.gz/bz2). For example, [this directory][dist-dir] should not be empty. I could not use apt and manually handled dependencies.

2. After install vrmlogger and its dependencies, it would not start up because com.victronenergy.system wasn't registered on dbus. I had to install dbus-systemcalc-py to get that working (thanks @mpvader). It might not be a bug, I might just not understand the decision behind it, but I suspect vrmlogger should depend on dbus-systemcalc-py, as it cannot opperate without it?

3. Unless I'm overlooking something, the architecture for dbys-systemcalc-py should be "all" and not "armhf".

4. This is outside our domain: daemontools creates a /etc/inittab file on jessie, which is not used at all because jessie uses systemd. I initially thought this was a bug and mentioned it to @mpvader , but it seems this is deliberate. If you switch from systemd to sysvinit, an inittab will be created without the required line for `svscanboot`. So daemontools avoids the situation by creating this file in advance. Non-bug, perhaps just document it? Suspect this might be my job :-)

5. I created a vebus-interface package locally because I didn't know this existed elsewhere. I also noticed there is a vecan-mk2 in meta-ccgx.
  - What is the difference between vecan-mk2 and what one might call a non-can build?
  - Notes that this repo should be split up and migrated elsewhere. Perhaps to meta-victronenergy-private?)
  - I suspect it is best that this stays out of the general build for now?

6. I made a package for dbus-cli. I'm not sure where the best place for this is. I made a pull request to @jhofstee 's repo, but I see that repo is already cloned into the victronenergy account. Should I put in a PR to that repo instead? My changes only makes the README files a bit more sexy and adds debianisation boilerplate so it can be installed as a deb package on the Rpi. The license is fairly liberal, it should not be a problem including this in the package repo?

7. I feel uneasy about everything running as root. Perhaps that's the old sysadmin in me. With the dbus layer in the middle, and debian/raspbian/ubuntu's shared setup where you need only be in the dialout group to access serial ports, perhaps we can button this down a bit. I see this as a low-priority thing, as I don't yet know if there is a benefit seeing as rather large holes have to be drilled into dbus anyway.

That's it for now. Let me know if there is something I can tackle here.

[dist-dir]: https://build-feeds.victronenergy.com/mvader/venus.git/WIP-bbb4/ae96eaf1/build-raspbian-jessie/deploy/raspbian-jessie/reprepro/dists/jessie/main/binary-armhf/

mpvader, 2016-06-28:
@izak 

1. package index files missing.
If you can, find out what the issue is and fix it. Otherwise @jhofstee will have to look at this.

2. vrmlogger depends on dbus-systemcalc-py.
yes. please fix and send @jhofstee a pull request. Don't forget to bumb the PR. Also for all other recipe changes.

3. architecture for dbus-systemcalc-py:
yes, make a pull request. Don't forget PR bumb again.

4. /etc/inittab on jessie not being used because systems.
I don't known anything about this ;)

5. vecan-mk2 related questions
vecan-mk2 is just the historical package / recipe name for the mk2-dbus executable. We'll include it in the packages build, and leave it in meta-ccgx for now. Once we have image updating in place (#27), we'll rename the recipe.

6. dbus-cli package.
For @jhofstee to reply on this

7. everything running as root.
Low priority indeed! Lets first make it work and easy to install for the (many!) people who will want to use these packages. Especially when mk2-dbus aka vecan-mk2 is included. And then yes, if someone is still bothered by this and has time, welcome to make some pull requests that fix it.

Best, Matthijs

izak, 2016-07-04:
Finally having a look at this again.

1. I see in the most recent build the Package index is now there. So this has been resolved.

2. I need access to the vrmlogger repo :-)

3. I'm looking at that now, just getting a strange error about oe.external.

6. I'll remake the pull request to the VE repo.

7. Should be a fairly benign change. I'm thinking as step one to only move to a different user, just to get some separation between system processes. It should be easy enough, daemontools include a setuidgid binary meant for this. qmail was famous for the insane separation it used, even logging was done as a different user.

mpvader, 2016-07-05:
For 2, I meant fix the recipe, not the repo itself. But I have given you access to the vrmlogger repo as well.

For 3. Its already done, see https://github.com/victronenergy/meta-victronenergy/commit/bb61c9990c674d8f47e6d061986d71dd1a710696

And that seems to be all then, nearly there! Let me know once the documentation is ready to be proofread.

izak, 2016-07-05:
Understood on number 2. Just cannot test it if the package won't build :-)

I wrote some documentation [here](https://github.com/victronenergy/venus/wiki/Installing-VRMLogger-on-Raspbian-Jessie), but I think it needs to be tested through again, because I have this habit of omitting details that are so obvious to me that I don't even think of adding them. Get that a LOT at work. Could also see some formatting love, that was basically just writing up the content of .bash_history in human-readable language.

Additional things I'm thinking of:

I'm a bit worried that my use of the udev symlink in /dev/serial/by-id might not work. It works very well with mk2dbus, but I seem to recall that the vedirect cable isn't properly detected and linked on jessie. The idea in using the udev created link is so your device doesn't move around, because you aren't guaranteed that it will come up in the same place each time after a reboot, or if a user unplugs and re-plugs devices.

Also thinking that the apt-add-repository tool from software-properties-common could be useful in setting this up, but for that to work cleanly you'll need to sign the release file and add a whole lot of extra stuff. It will get rid of the "these packages are unverified" warning though :-) A future nice-to-have perhaps.

mpvader, 2016-07-05:
Documentation:
I've quickly read through it. Style is fine. Just needs changing once dbus-systemcalc-py is auto installed, as well as mk2-dbus and vedirect-dbus and anything else. Which they should be as that was the intention (or not @jhofstee?). 

Also rather than getting people to play with D-Bus the primary goal of most people installing that is to monitor their data remotely on the VRM Portal. So please move the D-Bus playing things to a final extra section. And first continue on how to find vrm portal id and then connect to vrm.

packages are unverified: yes nice for future not for now.

izak, 2016-07-05:
Agreed on moving the dbus stuff. I thought it was an easy way to test that things are working, but even then it probably belongs towards the end as further and/or debug steps.

I completely forgot the vrmid. That's just the hardware address of eth0, so easy enough to find without installing connman.

izak, 2016-07-05:
Alright, dbus info is moved to the end. I removed the details about the LogInterval and replaced it with more generic testing info and a link to the official CCGX documentation that explains some of this. Also documented the VRM Portal ID.

I'll fix the vrmlogger/systemcalc dependency tonight and make a PR.

izak, 2016-07-14:
Two bugs remain, it should be easy enough for me to fix.

1. The start-vedirect.sh script has /opt/color-control hardcoded in it. That fails on the rpi where it should be /opt/victronenergy.

2. vedirect-interface doesn't like symlinks, probably because it names the dbus endpoint after the tty. A simple readlink -f solves the problem, so something like:

    if test -L $TTY_DEV; then
        TTY_DEV = $(readlink -f $TTY_DEV)
    fi

The symlink mechanism is needed, because we cannot guarantee that the serial devices will always be detected in the same order.

izak, 2016-07-14:
Another small issue: When upgrading a package, the modified `run` file is overwritten and the TTY_DEV placeholder returns. Usually dpkg marks files in /etc as configuration files all by itself, but not in this case as the file is a symlink to something in /opt. Have to consider marking the run files as configuration files so dpkg diverts these files instead of breaking your setup, or move configuration to a file that actually lives in /etc. Or something.

izak, 2016-07-14:
Just submitted a merge request that fixes the two issues logged previously. Also updated the howto so it is in line with the last changes.

My last concern is whether the vedirect-usb cable is properly detected by udev. I recall vaguely seeing it not work, but time will tell, cable was supposed to be delivered today but it will probably be here tomorrow. At the moment I'm using a generic serial cable, so the howto reflects that name.

mpvader, 2016-08-09:
Hi @izak I’ve gone through the [manual](https://github.com/victronenergy/venus/wiki/Installing-VRMLogger-on-Raspbian-Jessie), some questions:
- I think the name and/or title can be more generic: Installing Venus packages on a Raspberrypi
- Then explain what it is what one will get after installing this, and support for which types of products (includes Fronius PV Inverters connecting via Ethernet/Wifi)
- List the compatible distro and hardware (raspbian wheezy, raspbian jessy, debian jessy), all for raspberrypi2 only? or normal raspberrypi as well. Perhaps not necessary to test it all, but explain what has been tested and what might work as well but hasn't been tested.

mpvader, 2016-10-16:
@izak: did you already test one of the recent feeds in https://updates.victronenergy.com/feeds/venus ?

all OK?

izak, 2016-10-18:
I tested the latest bunch -- at least that they install and configure cleanly -- and updated the [documentation](https://github.com/victronenergy/venus/wiki/Installing-Venus-packages-on-a-Raspberry-Pi).

The issue with `/opt/color-control` vs `/opt/victronenergy` is still there, but I suspect the best time to solve that issue is when the move is complete and it just goes away on its own.

mpvader, 2017-10-16:
for anyone stumbling on this issue and looking for info, see documentation, and or consult the dev-venus mailing list.

Or even better, use the full [raspberrypi Venus images](https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-image), instead of the packages.

closed 2017-10-16 by mpvader