victronenergy/venus #37: pyo files corrupt?

mpvader, 2016-07-08:
Very rare, but still happens sometimes: a python process won't run. And its fixable by removing a pyo file forcing it to recompile.

For example this log, output for vrmpubnub:
```
2016-07-08 20:50:15.175094500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:15.175125500 EOFError: EOF read where object expected
2016-07-08 20:50:18.393325500 Traceback (most recent call last):
2016-07-08 20:50:18.397170500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:18.416976500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:18.417007500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:18.429183500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:18.429214500 EOFError: EOF read where object expected
2016-07-08 20:50:21.648941500 Traceback (most recent call last):
2016-07-08 20:50:21.651168500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:21.659500500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:21.659500500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:21.659530500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:21.659530500 EOFError: EOF read where object expected
2016-07-08 20:50:24.310501500 Traceback (most recent call last):
2016-07-08 20:50:24.312576500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:24.323562500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:24.323593500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:24.323593500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:24.323593500 EOFError: EOF read where object expected
2016-07-08 20:50:27.143386500 Traceback (most recent call last):
2016-07-08 20:50:27.147689500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:27.147720500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:27.147720500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:27.147720500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:27.147720500 EOFError: EOF read where object expected
2016-07-08 20:50:29.828720500 Traceback (most recent call last):
2016-07-08 20:50:29.833328500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:29.833358500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:29.833358500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:29.833358500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:29.837448500 EOFError: EOF read where object expected
2016-07-08 20:50:32.486587500 Traceback (most recent call last):
2016-07-08 20:50:32.495437500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:32.495468500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:32.495468500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:32.495468500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:32.495468500 EOFError: EOF read where object expected
2016-07-08 20:50:34.911544500 Traceback (most recent call last):
2016-07-08 20:50:34.914230500   File "/opt/color-control/vrmpubnub/vrmpubnub.py", line 46, in <module>
2016-07-08 20:50:34.917587500     from dbusmonitor import DbusMonitor
2016-07-08 20:50:34.917587500   File "/opt/color-control/vrmpubnub/./ext/velib_python/dbusmonitor.py", line 32, in <module>
2016-07-08 20:50:34.920364500     from vedbus import VeDbusItemExport, VeDbusItemImport
2016-07-08 20:50:34.920364500 EOFError: EOF read where object expected
2016-07-08 20:50:40.097610500 INFO:__main__:/opt/color-control/vrmpubnub/vrmpubnub.py v0.96 is starting up
2016-07-08 20:50:40.101242500 INFO:__main__:Loglevel set to INFO
2016-07-08 20:50:43.766525500 INFO:__main__:Auth-packages have been sent to VRM
2016-07-08 20:50:44.261215500 INFO:__main__:Starting mainloop, responding on only events
2016-07-08 20:50:45.124740500 INFO:__main__:Send to pubnub :[1, u'Sent', u'14680110450763114']
root@ccgx:~# 
```

What you see is the process being started and then crashing over and over again. Line 32 in dbusmonitor.py reads:
```
from vedbus import VeDbusItemExport, VeDbusItemImport
```

And removing vedbus.pyo is what made the process properly start again.

izak, 2016-07-08:
Two ways that I can think of. The one way is one I already suggested, that perhaps the processes shouldn't run as root. If they don't run as root, they cannot write a pyc or pyo file in the first place and no corruption can occur. It has to recompile it every time you start it, but that is negligible. The second way is what the Debian Python Policy recommends: You have to compile the modules in the postinstall step and remove them in the prerm step. If you ship image-based installs and upgrades with already-compiled modules that should also avoid corruption. If you can combine this with also not running as root, there is no possibility of future accidental recompilation.

That's just off the top of my head.

mpvader, 2016-07-08:
About shipping pre-compiled: yes that is an option. As long as we keep shipping the sources too I am OK with that.

Not running as root: perhaps a solution as well, though I am not immediately enthousiastic about the extra required config and work on the recipes. What is the extra value after doing step 1 already?

And I'd think that there must be built-in functions in python that auto-re-compile after seeing a corrupt pyo file with .py file as well?

izak, 2016-07-08:
The debian python policy says to ship the source, and then compile a module at install time. So your package doesn't really "ship" a compiled version, rather it compiles one at install time for the particular architecture, only to speed things up in future. When you ship image-based updates though, it will have the side effect of shipping those compiled binaries.

But let me look into running non-root, I'm fairly positive it is not going to be a lot of work, it's possibly going to be less work than figuring out a way to precompile or recompile things.

jhofstee, 2016-07-09:
alternatively we can set this system wide:

PYTHONDONTWRITEBYTECODE

If this is set to a non-empty string, Python won’t try to write .pyc or .pyo files on the import of source modules. This is equivalent to specifying the -B option.

mpvader, 2016-07-13:
@izak: Jeroen and myself discussed this, and we prefer, at least for now, to use the PYTHONDONTWRITEBYTECODE. It is a super simple fix and fast. And corrupt pyo files can create such vague issues for customers, that ultimately end up on my desk, or Jeroens desk, thus are a major waste of time: would be good to fix this asap.

Only two questions we have before applying that option, and I was hoping you could find the answers for us:

1) When enabled, does it also not look at existing .pyc or .pyo files? Or not, which makes it necessary to clean them all up.
2) The only advantage of byte code is reducing startup time, right? Can you check if, for vrmlogger for example, if there is a noticeable difference in startup time on the CCGX. Since already just starting Python takes 2 seconds, how much worse can the startup time become having to compile a few hundred lines of python code.

Matthijs

izak, 2016-07-14:
I agree that a first level fix is to add a -B option or PYTHONDONTWRITEBYTECODE env var. I've never had to do this, so I had to check the manual, didn't even know it existed!

It could be as simple as setting the #! string in the script to contain the extra -B.

I'll run a few performance test. You're right about it only being for performance reasons, and for large frameworks (such as a zope/plone stack) you will see a noticeable difference, but I doubt there will be much in this case.

izak, 2016-07-14:
Good info here. Startup is longer, speed is the same.

http://stackoverflow.com/questions/471191/why-compile-python-code

With the Rpi devices at work, I've had similar interesting things happening with flash. In our case, it's usually the DHCP lease file that gets corrupted, or more specifically, truncated. This is where dhcp stores its lease info so it can negotiate the same IP address next time. I'm not certain why it happens, but I think it has something to do with the way blocks are stored in flash, the block sizes (often quite large) and how that might align with the filesystem mechanics (often ext4). One would not expect a journalling FS to do this, but somehow it does so more often on flash than not.

mpvader, 2016-07-14:
ok. And are you able to test difference in startup time? 

testing will be a bit time consuming, since as written here that command line option prevents python from writing them, but not from using them:
http://stackoverflow.com/questions/10789165/how-to-prevent-python-from-using-orphaned-pyc-files-ones-with-no-matching-py

Might be usefull to retest that yourself though.

Also check the pyo files of all the python system libraries: what to do with them?

izak, 2016-07-18:
Very quick hacky-test. Put a sys.exit(1) right after the imports in the main file (at which point compilation will be finished). Then use the time command to check startup speed.

On a Raspberry Pi2:
With compiled files, around 1.4 seconds to start up.
Without compiling files, 1.7 seconds start up.

So the penalty is 0.3 seconds on vrmpubnub. The numbers are similar for vrmlogger. I think that is acceptable.

izak, 2016-07-24:
Merge Request submitted for this on GitLab.

mpvader, 2017-01-10:
@jhofstee: is it doable to implement the PYTHONDONTWRITEBYTECODE env variable for all python programs? I remember something about daemontools blocking all env vars?

mpvader, 2017-01-10:
@izak is it useful for @jhofstee to dig up your old merge request again? 

We've this fix on hold back in July, waiting for swupdate to be in place. Which is in place now, and I had another customer with this issue today (in v2.00), so its finally but also a bit urgently time to get this fixed now.

jhofstee, 2017-01-11:
hi guys, see above commit message. Ubuntu folks had the same vague / rare problems as this issue. They ended up backporting creating the pyc file first and thereafter rename it, instead of fiddling around with the magic header in the pyc file itself, so there can no longer be a half written pyc file. It is unclear to me why the magic header test fails, but given that Ubuntu folks solved their issues with it I guess it likely solves ours as well. If we encounter the issue again we can always decide to completely disable the pyc completely.

it is included in v2.01~6. Closing this issue for now, if we see it again, just reopen it to completely disable pyc support.

jhofstee, 2017-01-20:
for what it is worth, encountered this on a v2.00 version:

@4000000058820ad3266260bc Traceback (most recent call last):
@4000000058820ad327e21734   File "/opt/color-control/localsettings/localsettings.py", line 46, in <module>
@4000000058820ad328048b84     from lxml import etree
@4000000058820ad32857e75c   File "lxml.etree.pyx", line 39, in init lxml.etree (src/lxml/lxml.etree.c:140909)
@4000000058820ad328585c8c EOFError: EOF read where object expected

interestingly lxml is shipped with pyc files. removing them made localsettings boot again....

mpvader, 2017-04-20:
it still happens after v2.01: https://github.com/victronenergy/venus/issues/137.

re-opening this one, and closing the other one.

jhofstee, 2017-11-30:
shipping pyc files now

closed 2017-11-30 by jhofstee