victronenergy/venus #38: (draft) create eventlogger

mpvader, 2016-07-13:
## idea
tbd

## what this solves
- The two lists we have are not in sync: confusing. (notification list in gui is generated by gui itself, and VRM Portal also has its own (different) set of rules generating the VRM Alarm Log)
- Notifications shown in the gui are not stored to non-volatile memory. Erased on a reboot
- Code in gui needs to be added for each thing you want to have as an alarm. Instead of adding that logic to place where it belongs: dbus-vecan, dbus-vedirect, dbus-mk2 etc.

## List of events
Existing:
- (to be added, there must be a list in the gui somewhere)

New events which can/could be added:
- Mains power failure
- Genset started/stopped
- VE.Bus system config changed remotely (Remote VEConfigure)
- CCGX Firmware updated from .. to ..
- VE.Can Firmware for device X updated (remotely)
- Connection to device X (for example a BMV) lost. / found again
- Alarm xx started, followed by alarm xx cleared
- Warning xx started, followed by warning xx cleared

And even more events, of the type that not necessarily need to be shown to the user. But can still be very useful to us
- Service xx crashed  (data comes from daemontools)
- Watchdog reboot

mpvader, 2016-09-19:
## Could we use MQTT for this?
services would then throw the events at the local mqtt broker running on the ccgx. A local db can take them from that broker and store them for local use, such as showing in the gui. And the broker will forward the messages to the online VRM Broker. And same there, some service can take them from that broker and store them in the database.

Some initial considerations:
- mqtt can queue the messages while the online broker is unreachable. This is native mosquitto functionality. But is that queue stored in non-volatile memory?

jlark, 2016-09-21:
We could use something like https://www.graylog.org/ locally ad pipe everything to syslog. This would provide a central place for all the events/logs to aggregate.

mpvader, 2018-07-12:
we'll look at this once `the stars are in the right position`

closed 2018-07-12 by mpvader