victronenergy/venus #43: configure gpio pins

mpvader, 2016-08-08:
The used gpio pins (for the buzzer, and relays) are defined in the the [beaglebone.conf file](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-bsp/conf/machine/beaglebone.conf).

But before they can be used, they need to be configured. During testing we configured them manually, see [here in the wiki](https://github.com/victronenergy/venus/wiki/bbb-gpio).

Since they are not configured, errors show up in the dbus-systemcalc-log:

```
root@beaglebone:/dev# cat /var/log/dbus-systemcalc-py/current | tai64nlocal
2016-08-08 23:00:56.363865500   File "/opt/color-control/dbus-systemcalc-py/delegates.py", line 266, in _update_relay_state
2016-08-08 23:00:57.073019500     with open(file_path, 'rt') as r:
2016-08-08 23:00:57.073025500 IOError: [Errno 2] No such file or directory: '/sys/class/gpio/gpio68/value'
```

mpvader, 2016-08-09:
also the buzzer is not yet included in the beaglebone.conf file, that needs to be done as well. (And tested with the capes)

luctius, 2016-08-24:
commit 512301fb4a2040b3fff9e2e44d66ad8a04101df4 in meta-victronenergy adds a script to set gpio pins.
It is configured by the file /etc/venus/gpio_list.

closed 2016-08-24 by luctius