victronenergy/venus #44: make the usb work as an ethernet device

mpvader, 2016-08-09:
To configure the bbb in a wifi network, it is now first necessary to:
- hook it up to a wired (ethernet) network
- find it in the network (which is quite easy because of simple-upnpd)
- type the ipaddress in a webbrowser, which opens up the Remote Console on LAN
- finally configure wifi
- and then you can remove the wired ethernet cable again.

Which is not always to easy. So therefore it would be nice if configuring via USB is possible. And Ethernet over USB sounds like the simplest solution.

Note that at some point it might also be possible to configure the device via BLE with VictronConnect, which is even easier. And even without BLE, instead of using a webbrowser it will soon be possible to use VictronConnect: it will use upnp to find the device, and then connects to it over mqtt.

And another alternative would be to auto-configure the wifi as an access point in case it is not configured or not connected.

Ethernet over USB works in the default Beaglebone image from the beaglebone organisation.

mpvader, 2016-09-06:
The bbb beaglebone image uses this:

https://www.kernel.org/doc/Documentation/usb/gadget_multi.txt

mpvader, 2016-09-19:
https://github.com/RobertCNelson/boot-scripts/blob/master/boot/autoconfigure_usb0.sh

jlark, 2016-09-20:
This could also be done using the <hostname>.local in the browser (no need for ip address) https://en.wikipedia.org/wiki/.local

mpvader, 2017-03-20:
closing, since we have other solutions like #51.

and prefer to use this usb port to increase limit of connectable usb devices, see #130.

closed 2017-03-20 by mpvader