victronenergy/venus #45: dbus-motordrive: rpm is missing from the gui

mpvader, 2016-08-09:
See screenshot:
[image]

The problem is that the D-Bus items have no texts. Try 'dbus -y com.victronenergy.motordrive.piktronik0 /Motor/RPM GetText' and you will get an empty string. Whevener a D-Bus item is displayed in the GUI and no unit is specified you will see the text of the D-Bus item instead of the value. If there is no text '--' is used instead.

To enable text in D-Bus items, use the 'veItemAddQuantity' function from velib instead of 'veItemAddChildByUid'. This function requires an extra parameter of type VeVariantUnitFmt. This is a struct where you can specify the precision and the unit. You can set the unit to NULL if there is none.

Velib contains an example project which illustrates the process: examples/dbus/c_service/task.c.

mpvader, 2016-08-09:
assigned to Oceanvolt

mpvader, 2016-08-09:
note that in CCGX v1.70 we still have dbus-motordrive v1.6, which also has an issue causing the temperature not to be displayed on GUI and also not on VRM. I've added v1.7 to our todo list for next CCGX version.

jhofstee, 2016-08-09:
Hello Matthijs,

On 09-08-16 09:41, mpvader wrote:
>
> See screenshot:
> image 
> <https://cloud.githubusercontent.com/assets/5200296/17508389/0924b8a0-5e15-11e6-8460-793aa8395e58.png>
>
> The problem is that the D-Bus items have no texts. Try 'dbus -y 
> com.victronenergy.motordrive.piktronik0 /Motor/RPM GetText' and you 
> will get an empty string. Whevener a D-Bus item is displayed in the 
> GUI and no unit is specified you will see the text of the D-Bus item 
> instead of the value. If there is no text '--' is used instead.
>
> To enable text in D-Bus items, use the 'veItemAddQuantity' function 
> from velib instead of 'veItemAddChildByUid'. This function requires an 
> extra parameter of type VeVariantUnitFmt. This is a struct where you 
> can specify the precision and the unit. You can set the unit to NULL 
> if there is none.
>
I doubt the last statement is correct. Preferably use veItemAddBasic if 
you had an item
and want to add it or veItemCreateBasic to create and add it. Both 
versions will use
veUnitNone instead of NULL.

Regards,
Jeroen

RealAth, 2016-08-17:
RPM bug is now fixed in version 1.7 (found in git with tag '1.7').

mpvader, 2016-08-17:
Hi, thanks. I added it to our venus todo list with this change log:

dbus-motordrive v1.7
* fix temperature missing from GUI and VRM by changing the path for the controller temperature from /Dc/0/Temperature into /Controller/Temperature
* fix RPM missing from GUI by adding GetText for it

Please confirm that that is ok like that.

Matthijs

RealAth, 2016-08-17:
Sounds good.

However in firmware version 1.70 GUI (file gui/qml/PageMotorDrive.qml) uses path /Dc/0/Temperature for controller temperature. Is this changing or already changed in some development version?

mpvader, 2016-08-17:
thanks for that: we've also added that gui change to the list of necessary changes.

I'll close this issue now. You'll see a test version (v1.80~xx) appearing in the coming weeks. We'll email the change log to the [venus-dev mailing group](https://groups.google.com/forum/#!forum/victron-dev-venus).

closed 2016-08-17 by mpvader