victronenergy/venus #46: dbus-valence: bms service has wrong product id

martinbosma, 2016-08-09:
The product id is set to 0x142, which is the Lynx Ion. It should be 0xB001.

martinbosma, 2016-08-09:
`dbus -y com.victronenergy.bms.valence0 / GetValue
value = {'Balancing': 0,
 'Connected': 1,
 'DeviceInstance': 0,
 'FirmwareVersion': 0,
 'HardwareVersion': '',
 'Io/AllowToCharge': 1,
 'Io/AllowToDischarge': 1,
 'Mgmt/Connection': 'VE.Can',
 'Mgmt/ProcessName': 'dbus_valence',
 'Mgmt/ProcessVersion': '1.4',
 'N2KUniqueNumber': 10,
 'ProductId': 322,
 'ProductName': 'Super B BMS',
 'Serial': '',
 'State': 9,
 'System/BatteriesParallel': 0,
 'System/BatteriesSeries': 4,
 'System/MaxCellVoltage': 0.0,
 'System/MinCellVoltage': 0.0,
 'System/NrOfBatteries': 0,
 'System/NrOfCellsPerBattery': 4}
`

mpvader, 2016-08-09:
assigned to oceanvolt

RealAth, 2016-08-17:
dbus-valence found in latest release candidate stream couldn't found D-Bus services. I updated the velib and it works again. However I couldn't reproduce this product id bug. Command  "dbus -y com.victronenergy.bms.valence0 / GetValue" outputs 'ProductId': 45057,".

New dbus-valence is now available in git (tag is 1.10). executable-name branch is also merged into it (binary name is now dbus-valence instead of dbus_valence).

mpvader, 2016-08-17:
ok. I'll add it to the venus todo list. What is the change log for dbus-valence 1.10?

RealAth, 2016-08-17:
dbus_valence 1.10
- rename the executable dbus-valence
- update velib

mpvader, 2016-08-17:
thanks! I'll close this issue now. You'll see a test version (v1.80~xx) appearing in the coming weeks. We'll email the change log to the [venus-dev mailing group](https://groups.google.com/forum/#!forum/victron-dev-venus).

closed 2016-08-17 by mpvader