victronenergy/venus #47: dbus-generator is littering the localsettingslog

mpvader, 2016-08-12:
which makes it impossible to look back when the user changed a certain setting.

For example:
```
2016-08-11 16:52:37.279154500 2016-08-11 16:52:37,271 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561027, New: 2561037
2016-08-11 16:52:37.310618500 2016-08-11 16:52:37,309 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54950}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54960}
2016-08-11 16:52:47.255808500 2016-08-11 16:52:47,253 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561037, New: 2561047
2016-08-11 16:52:47.283701500 2016-08-11 16:52:47,282 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54960}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54970}
2016-08-11 16:52:57.294016500 2016-08-11 16:52:57,292 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561047, New: 2561057
2016-08-11 16:52:57.328287500 2016-08-11 16:52:57,326 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54970}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54980}
2016-08-11 16:53:07.441050500 2016-08-11 16:53:07,439 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561057, New: 2561067
2016-08-11 16:53:07.495493500 2016-08-11 16:53:07,491 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54980}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54990}
2016-08-11 16:53:17.321482500 2016-08-11 16:53:17,319 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561067, New: 2561077
2016-08-11 16:53:17.348246500 2016-08-11 16:53:17,346 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 54990}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 55000}
2016-08-11 16:53:27.368174500 2016-08-11 16:53:27,365 INFO Setting /Settings/Generator0/AccumulatedTotal changed. Old: 2561077, New: 2561087
2016-08-11 16:53:27.396280500 2016-08-11 16:53:27,394 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "1469577600": 52724, "1469664000": 58900, "1469750400": 66767, "1469836800": 62799, "1469923200": 0, "1470009600": 33186, "1470096000": 72371, "1470182400": 71516, "1470268800": 55057, "1470355200": 59054, "1470441600": 58224, "1470528000": 0, "1470614400": 52372, "1470700800": 41045, "1470787200": 40879, "1470873600": 55000}, New: {"1468368000": 0, "1468454400": 0, "1468540800": 0, "1468627200": 0, "1468713600": 0, "1468800000": 0, "1468886400": 30669, "1468972800": 0, "1469059200": 28794, "1469145600": 28997, "1469232000": 86400, "1469318400": 55094, "1469404800": 5783, "1469491200": 62864, "14695 [... cut ...]
2016-08-11 23:00:00.985477500 2016-08-11 23:00:00,983 INFO Setting /Settings/Generator0/AccumulatedDaily changed. Old: {"1470873600" : 0}, New: {"1470873600": 0, "1470960000": 0}
2016-08-12 07:00:05.112003500 2016-08-12 07:00:05,083 INFO Setting /Settings/Gui/StartWithMenuView changed. Old: 0, New: 1
2016-08-12 07:02:20.277134500 2016-08-12 07:02:20,275 INFO Setting /Settings/Services/VeCan changed. Old: 0, New: 1
```

And looks like StartWithMenuView might do the same. Not sure how to fix this: either add an option for localsettings not to write changes to certain paths to the log. Or don't use localsettings for these kinds of things. Or .. ?

mpvader, 2016-08-12:
for the above example, there is so much written to the localsettings log that one can't even look back more than a day

jhofstee, 2016-12-08:
as a start, we can at least decrease the interval, also for cpu usage.

jhofstee, 2017-01-25:
closing this issue, more common solution can be found here https://github.com/victronenergy/venus/issues/123

closed 2017-01-25 by jhofstee