victronenergy/venus #50: ve-cape: use eeprom

mpvader, 2016-09-14:
1) Store the following fields in the eeprom during manufacturing:
- eeprom format version (0x00)
- hw revision number (0x0000)
- product id (0x????)
- analog input calibration parameters (if and when we decide to calibrate them)
- crc / md5sum or something

2) Make the values available in the rootfs
Make some script or executable that reads these values (early!) during the boot process, and writes them to /data.

Note that, in case the same value also exists on the CCGX, it needs to be in the same location.

3) Less priority, extend this. Candidates:
- unique can number
- serial number (HQ ....)
- VE.Smart PUK code

4) Access via /etc/venus?
It might be a good idea to have other processes access these values via symlinks in /etc/venus: when installed as a package, for example on an rpi, there is no /data. (?)

jhofstee, 2019-02-27:
closing this, old

mpvader, 2020-12-01:
@mansr there are two new fields that I want to store in the eeprom of the GX Device:

- OpenPaygo Private key (128 bits)
- OpenPaygo Starting code (int, max 9 digits, ie. an uint32)

Pedro needs them to finish what he is working on now. Can you add them to [update_eeprom.py](https://github.com/victronenergy/meta-victronenergy-private/blob/master/recipes-ve/prodtest/update-eeprom.py)?

mpvader, 2021-02-25:
Today I tested the latest eeprom script. Result is all passed wrt the script.

In the production test sequence there is one bug: it correctly retrieves a mac address from the pool and writes it to the eeprom, but what it then somehow does archive the 02-eth-mac address into VictronSN.

And there are a few things to do to clean up, but they don't need to any longer delay production of the SHS product:
1. the script now makes /data/venus/hw-revision, which is double with the `hw-revision` binary, and also both give a different result.
2. similarly, /data/venus/product-id vs `productid-`.

All the details:

Tests done with this build:

meta-multimedia      
meta-networking      
meta-oe              
meta-python          
meta-webserver       = "b2.60:eab5c1c81827cf0ebca9bbaccfa8c3c8da2a25ee"
meta-qt4             = "b2.60:ec8e0800ef1c33b72f3d3924412dabc48c2a55ac"
meta-qt5             = "b2.60:fefe9112dcf17147dba28dd5ac7f071b909042b5"
meta-swupdate        = "b2.60:757d735b7099256e8f18aa048b4f4bd29222fb0b"
meta-alternatives    
meta-bsp             
meta-third-party     
meta-venus           
meta-ve-software     = "matthijs/eeprom:c1efead39458b70cebb1bcba9f995e0c4b8ae3a2"
meta-victronenergy-private = "mans/eeprom-2.60:98fcd1eac7486d36a7420699a7f00235f486d5ea"
meta                 = "b2.60:37b998e25e009a585aeef0f867de358dd666d413"

### TEST A) Installed the SWU file on a Cerbo programmed with v0 in production a while ago.

```
root@einstein:/data/venus# cat /opt/victronenergy/version 
v2.63-eeprom-test
Victron Energy
20210225110306

root@einstein:/data/venus# /opt/victronenergy/venus-eeprom/eeprom --check
format-version 0
product-id C00A
hw-rev 1
part-number BPP900450100
serial-number HQ20283D6HA
wpa-psk 7y5zwdns
installer-version 20200706160114 v2.56
crc32 d585c088
```

1) the eeprom --read call at boot didn't touch the mqtt_password & vrm_auth token. Which is PASSED, as they are not in the eeprom and thus the info on /data needs to be preserved:

```
root@einstein:/data/conf# ls -al *
-rw-r--r--    1 root     root            32 Feb 21 21:25 mqtt_password.txt
-rw-r--r--    1 root     root         29488 Feb 25 14:36 settings.xml
-rw-r--r--    1 root     root             1 Jan 31 19:41 vncpassword.txt
-rw-r--r--    1 root     root            16 Jan 19 13:44 vrm_auth_token.txt
```

2) the eeprom --read call at boot did overwrite the right files in /data/venus (which means PASSED):

```
root@einstein:/data/venus# head *
==> bluetooth-addr <==
hci0 ec:06:44:05:79:41

==> hw-revision <==
1

==> installer-version <==
v2.56
Victron Energy
[... 463 lines cut ...]
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:98 errors:0 dropped:0 overruns:0 frame:0
          TX packets:98 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10944 (10.6 KiB)  TX bytes:10944 (10.6 KiB)

wifi0     Link encap:Ethernet  HWaddr C0:84:7D:C9:A7:45
          UP BROADCAST MULTICAST  MTU:1450  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

get-unique-id still returns the wifi id, which is PASS:
```
root@einstein:~# get-unique-id
c0847dc9a745
```

Information on the printed label is all OK: vrm portal id == wifi mac. And ethernet address is the C0619...1D numbrt.

### Test E) Wifi access point PASS. it works.

both using the printed key as well as using the qr code.

mpvader, 2021-02-25:
ps; I also tested that vrmportal & mqtt processes properly work with the two password files created for them.

mpvader, 2021-02-26:
Retested, after the updated that no longer writes the hw-rev and product-id files to /data/venus.

Diff of the change:
```
@jeroen mans pushed a new version of the branches. ~(reverse)~ difference to the version I tested earlier:

```diff --git b/recipes-ve/venus-eeprom/eeprom a/recipes-ve/venus-eeprom/eeprom
index bfda458..c4aff05 100755
--- b/recipes-ve/venus-eeprom/eeprom
+++ a/recipes-ve/venus-eeprom/eeprom
@@ -92,14 +92,15 @@ EEPROM_FIELDS = (
      ('crc32',               'I',   '{:08x}')),
 )
 
-FILES = {
+# name, filename, type
+FIELDS = {
     'installer-version': ['/data/venus/installer-version', version_str],
-    'hw-rev':            ['/data/venus/hw-revision', auto_int],
+    'hw-rev':            [None, auto_int],
     'mqtt-passwd':       ['/data/conf/mqtt_password.txt', hexstr(16)],
     'part-number':       ['/data/venus/part-number', str],
     'paygo-priv-key':    ['/data/venus/paygo-priv-key', hexstr(16)],
     'paygo-start-code':  ['/data/venus/paygo-start-code', auto_int],
-    'product-id':        ['/data/venus/product-id', hex_int],
+    'product-id':        [None, hex_int],
     'serial-number':     ['/data/venus/serial-number', str],
     'vrm-auth':          ['/data/conf/vrm_auth_token.txt', hexstr(8)],
     'wpa-psk':           ['/data/venus/wpa-psk', str],
@@ -226,14 +227,14 @@ def write_file(f, s, v):
     if len(f) > 3:
         s = f[3](v)
 
-    file = FILES.get(f[0])
-    if not file:
+    field = FIELDS.get(f[0])
+    if not field or not field[0]:
         return
 
     if all(x == chr(255) for x in struct.pack(f[1], v)):
         return
 
-    fname = file[0]
+    fname = field[0]
     dname = os.path.dirname(fname)
     if not os.path.exists(dname):
         os.makedirs(dname)
@@ -271,7 +272,7 @@ def main():
     parser.add_argument('--format-version', type=auto_int)
     parser.add_argument('--eth-addr', type=parse_eth_addr)
 
-    for f, d in FILES.items():
+    for f, d in FIELDS.items():
         parser.add_argument('--' + f, type=d[1])
 
     args = parser.parse_args()
```

### TEST A) Updating a Cerbo GX programmed with eeprom v0 in production.

Before:
```
root@einstein:~# ls -al /data/venus/*
-rw-r--r--    1 root     root            23 Feb 25 14:42 /data/venus/bluetooth-addr
-rw-r--r--    1 root     root             2 Feb 25 14:42 /data/venus/hw-revision
-rw-r--r--    1 root     root            36 Feb 25 14:42 /data/venus/installer-version
-rw-r--r--    1 root     root            13 Feb 25 14:42 /data/venus/part-number
-rw-r--r--    1 root     root             5 Feb 25 14:42 /data/venus/product-id
-rw-r--r--    1 root     root            12 Feb 25 14:42 /data/venus/serial-number
-rw-r--r--    1 root     root            13 Feb 25 14:42 /data/venus/unique-id
[... 51 lines cut ...]

```
root@einstein:~# cat /opt/victronenergy/version 
v2.63-e3
Victron Energy
20210226130336

root@einstein:~# /opt/victronenergy/venus-eeprom/eeprom --check
Unknown EEPROM format version 255

root@einstein:~# ls -al /data/venus/*
-rw-r--r--    1 root     root            23 Jan  1  1970 /data/venus/bluetooth-addr
-rw-r--r--    1 root     root            38 Jan  1  1970 /data/venus/installer-version
-rw-r--r--    1 root     root            13 Jan  3  2020 /data/venus/part-number
-rw-r--r--    1 root     root            12 Jan  3  2020 /data/venus/serial-number
-rw-r--r--    1 root     root            13 Feb 26 13:07 /data/venus/unique-id
-rw-r--r--    1 root     root             9 Jan  3  2020 /data/venus/wpa-psk

root@einstein:~# ls -al /data/conf/*
-rw-r--r--    1 root     root            32 Jan  8  2020 /data/conf/mqtt_password.txt
-rw-r--r--    1 root     root         19301 Feb 27  2020 /data/conf/settin_old.xml
-rw-r--r--    1 root     root         30551 Feb 26 13:08 /data/conf/settings.xml
-rw-r--r--    1 root     root             0 Feb 26 13:07 /data/conf/vncpassword.txt
-rw-r--r--    1 root     root            16 Jan  6  2020 /data/conf/vrm_auth_token.txt
```