victronenergy/venus #51: make a wifi access point

mpvader, 2016-09-15:
The BBB has no display what-so-ever. Which makes it quite the black box when you want to configure it. It would be great if it acts as a WiFi access point. So one can connect to that, and set it up. Or even use it in case it is in some remote place with no other existing wifi or lan network.

Some wifi chipsets support to be both an AP and connected to another network at the same time. Some don't :smiley:.

In case the chipset cannot do both at the same time, the logic which needs to be followed is not straight forward:
- of course on first boot, or in any other situation where there is no wifi config made, it can start as an access point
- but then, once there is a config, it should try to connect to that configged networks(s). And never ever give up.

So that last point makes it a bit hard to get to that access point again. Some possible solution would be to be an access point only the first xx seconds during power up. Perhaps others too, suggestions welcome.

But, lets first double check if the wifi chipset in the [BeagleBone Enhanced](https://github.com/victronenergy/venus/wiki/bbb-hardware-spec#victron-version-of-the-beaglebone-enhanced) can do both at the same time. Since that would already make it a lot less urgent to come up with a real good solution for above issue.

mpvader, 2016-09-15:
reading material:
- Long discussion on the beagleboard mailing list:
https://groups.google.com/forum/m/#!category-topic/beagleboard/v4_A2x1I5gs
- Wiki page about using connman as an accesspoint
https://github.com/IntelOpenDesign/MakerNode/wiki/connman:-using-as-an-access-point
- Repo containing there end result
https://github.com/IntelOpenDesign/MakerNode/tree/master/conf

mpvader, 2016-09-16:
- Enabling STA+AP mode for Realtek chipsets:
https://randomstuffidosometimes.blogspot.nl/2016/03/rtl8192cu-and-rtl8188cus-in-station-and.html?m=1

mpvader, 2016-10-14:
[Realtek_WiFi_concurrent_mode_Introduction.pdf](https://github.com/victronenergy/venus/files/528866/Realtek_WiFi_concurrent_mode_Introduction.pdf)

mpvader, 2016-12-24:
The chipset in the bbe is the rtl8723bu. The driver we use for it is [this one](https://github.com/lwfinger/rtl8723bu).

It is compiled with the CONFIG_CONCURRENT_MODE, and as a result there are two wlan interfaces:

```
wlan0     Link encap:Ethernet  HWaddr 38:1D:D9:38:2A:D9
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2 errors:0 dropped:157 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan1     Link encap:Ethernet  HWaddr 3A:1D:D9:38:2A:D9
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:38 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

Now, there are two options to use wlan1 as an accesspoint:
- use connman for it, connman calls this tethering. (tethering is of course more than just running an accesspoint, but could be a nice extra bonus)
- make connman ignore that interface, and use something else.

Using connman seems the easiest route. But, enabling tethering for wifi doesn't work. Enabling tethering with connmanctl returns very minimal error message: `Error enabling wifi tethering: Not supported`. Running connman in debug mode (add -d), yields a bit more info:

```
connmanctl> connmand[4383]: /media/ram/builds/a9bbdc1a/1/mirrors/venus/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/connman/1.33-r2.0/connman-1.33/src/technology.c:set_property() conn 0xccd28
connmand[4383]: /media/ram/builds/a9bbdc1a/1/mirrors/venus/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/connman/1.33-r2.0/connman-1.33/src/technology.c:set_property() property Tethering
connmand[4383]: Missing support for 802.1d ethernet bridging
Error enabling wifi tethering: Not supported
```

Relevant code is here (tethering.c):

```
const char *__connman_tethering_get_bridge(void)
{
	int sk, err;
	unsigned long args[3];

	sk = socket(AF_INET, SOCK_STREAM, 0);
	if (sk < 0)
		return NULL;

	args[0] = BRCTL_GET_VERSION;
	args[1] = args[2] = 0;
	err = ioctl(sk, SIOCGIFBR, &args);
	close(sk);
	if (err == -1) {
		connman_error("Missing support for 802.1d ethernet bridging");
		return NULL;
	}

	return BRIDGE_NAME;
}
```

So, CONFIG_BRIDGE was missing in the kernel defconfig. After adding that, enabling tethering again yields this:

```
connmanctl> tether wifi on myap hoihoihoi
Wifi SSID set
Wifi passphrase set
Enabled tethering for wifi
[  194.150050] connmand wrote to ip_forward when file position was not 0!
[  194.150050] This will not be supported in the future. To silence this
[  194.150050] warning, set kernel.sysctl_writes_strict = -1
l2_packet_init: ioctl[SIOCGIFINDEX]: No such device
wlan0: Failed to open l2_packet connection for the bridge interface 'tether'
```

to be continued..

mpvader, 2016-12-24:
Above `No such device` error was probably caused by the wifi interface being already associated as a client: disconnecting it before enabling tethering makes it work:

From windows I can connect to 'myap'. Windows machine gets an ip address, and it can connect to the Remote Console on 192.168.0.1. Which is the ip on the 'tether' interface:

```
# ifconfig
can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
...

eth0      Link encap:Ethernet  HWaddr B0:D5:CC:F1:B3:56
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:174

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3818 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3818 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:385399 (376.3 KiB)  TX bytes:385399 (376.3 KiB)

tether    Link encap:Ethernet  HWaddr 38:1D:D9:38:2A:D9
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::b8ca:2eff:fe8f:ecce/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:902 errors:0 dropped:0 overruns:0 frame:0
          TX packets:279 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:81413 (79.5 KiB)  TX bytes:28226 (27.5 KiB)

wlan0     Link encap:Ethernet  HWaddr 38:1D:D9:38:2A:D9
          inet6 addr: fe80::3a1d:d9ff:fe38:2ad9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:909 errors:0 dropped:50 overruns:0 frame:0
          TX packets:286 errors:0 dropped:4 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:168387 (164.4 KiB)  TX bytes:68777 (67.1 KiB)
```

Connecting from my iphone doesn't work though. And a new issue is coming up: connman is all about wifi / ethernet / gadget etcetera. But not about wlan1, wlan0. So using it to use wlan0 as a client while wlan1 as an AP for tethering might be tricky. I'll ask the connman people.

mpvader, 2016-12-27:
Email to connman:
https://lists.01.org/pipermail/connman/2016-December/021360.html

And in the meantime I am setting it up with hostapd and dnsmasq instead. Which works.

mpvader, 2017-01-29:
@mansr: as discussed, please take this one from me. Most previous comments in this issue are about using connman, but that didn't work out:
https://lists.01.org/pipermail/connman/2017-January/021368.html

Might be better anyway to use hostapd for this.

Latest status is here:
https://github.com/victronenergy/meta-victronenergy/tree/matthijs/add-hostapd

todo, besides testing:
- SSID needs to contain the product serial number, so it can be recognized when there are multiple in range. Our serialnumber is always 11 characters. For example HQ1702DE2W3. Which would be a product manufactured in 2017, week 2. Store it in /data/venus/serial-number
- Add a password. Which can be randomly generated, stored in /data/somehwere and printed on a label on the back in the factory.
- See if something can be done with DNS, to user can just type a name (venusgx) in his browser instead of an ipaddress.
- Default enable the AP, but have an option to disable it from the GUI. If convenient, @jepefe can do the GUI work again.
- To my opinion it would be overkill to give the user an option the change the password in the gui.
- I'm not sure what to do when the file containing the password is not there. Just don't enable the AP?

jhofstee, 2017-02-21:
hello @mansr, @mpvader,

So it seems we take the wifi / bluetooth / upnp route right? So any objection if I turn the USB otg into a host mode? A single musb can currently support up to 10 devices FTDI usb serial port converters per port, so using two ports that makes it 20 instead of 10, but attaching a gadget driver will no longer be possible.

@mansr when looking into this again, it waits on https://github.com/victronenergy/venus/issues/104, please do add an ### BEGIN INIT INFO header, we might not use it now, but systemd will use it as backwards compatible afaik.

Regards,
Jeroen

mpvader, 2017-02-21:
@jhofstee:

> So any objection if I turn the USB otg into a host mode?

No objection, go ahead.

mpvader, 2017-03-16:
@mansr: took me a while, but I finally looked at the patches. Also all the rest is done now: the product is ready to start being produced. And would be nice to be able to include this from the start.

Two more things that need to be done for this wifi issue:

1) Make it possible for user to disable the access point.

    suggestion: (feel free to come with a better one obviously): run hostapd under daemontools. Then either   you if you have time or @jepefe can easily add it to the gui: it already has a framework to start/stop daemontools services based on user settings.

2) Add the last 3 chars of the vrm-portal-id to de SSID. So the SSID becomes:

    HQ0000AAAAA-edc, where edc are those three last chars.

    the vrm portal id can be found in /data/venus/unique-id, which is created by [machine-runtime-conf](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-bsp/recipes-bsp/machine-runtime-conf/machine-runtime-conf_0.3.bb)

    reasoning: we don't want to broadcast the full vrm portal id, since it is sort-of used as an authentication mechanism for the VRM Portal. But still its handy to see / recognize it when looking for access points, since the same id is used in upnp and also when adding sites to https://vrm.victronenergy.com

mpvader, 2017-03-20:
@jepefe: now that above has been done, up to you to complete it:

- add enabling/disabling the access point to the gui and a setting in localsettings.
- default enabled. Note that the daemontools service is default disabled.
- I am not sure where to put it in the gui: a new submenu or adding it to Settings -> Wifi.

Don't let the user change the password, its also printed on the box, and there is no need to make it changeable.

The hostapd service is /service/hostapd

swu file containing the access-point:
https://www.dropbox.com/s/9myxkeyawwgsmnt/venus-swu-beaglebone-20170320195827-v2.03~wlan.swu?dl=0

jepefe, 2017-03-21:
Done and pushed to branch: https://github.com/victronenergy/gui/commit/3c073674e9ef4a1f1f9fff993ffc36628c08075d

Enable/disable option added to Settings -> Wi-Fi. Wi-Fi networks list has been moved to a submenu for beagle bone and kept on the main Wi-Fi page for CCGX. (screenshot)

Created /Settings/Service/AccessPoint setting in local settings which is enabled by default.

[image]

mansr, 2017-03-21:
The GUI control works as expected here.

jhofstee, 2017-03-21:
Thanks, hostapd and gui are included in v2.04~1.

closed 2017-03-21 by jhofstee

zorrobyte, 2019-02-23:
Access Point (HostAPD) on wlan1 for Raspberry Pi
https://groups.google.com/forum/#!topic/victron-dev-venus/vB0XUbYeHC4