victronenergy/venus #52: connman: upgrade to latest version

mpvader, 2016-09-15:
Currently used version in Venus is v1.25. The latest available version is v1.33. Between those, there are lots of dhcp, dns and other relevant fixes which might improve network / connection stability.

There are also user complaints now and then. Some reproducible, some not. But instead of debugging / investigating them now, we'll first work with a newer version of ConnMan and see what then happens.

mpvader, 2016-10-06:
The upgrade work is completed and ready for testing.
- Details in [meta-victronenergy/connman](https://github.com/victronenergy/meta-victronenergy/commits/connman).
- Test swu files available in [mva/connman on the build-feeds server](https://build-feeds.victronenergy.com/venus/mva/connman/images/ccgx/).
- [ci pipeline here](https://git.victronenergy.com/mirrors/venus/pipelines/867)
 
Before adding this into master there is some more testing needed.
 
While briefly testing myself I noticed: blank lines in the gui page showing all the networks. Was this also happening in venus v1.71 with older connman? I didn't have time to check that.

mpvader, 2016-10-07:
To revert a CCGX to v1.71, you can use the [production image](https://www.dropbox.com/s/dkw1bzcov1hqyih/ccgx-install-v1.71-prod.zip?dl=1)

mansr, 2016-10-07:
Both versions show blank entries for networks with hidden SSID.

mpvader, 2016-10-07:
Ok. @ThiemoVanEngelen did some more tests today, with a Ccgx with v1.71 and one with new Connman side by side: not a single difference found.

I'll finish what I was doing: put it in master and make a new test release.

mpvader, 2016-10-07:
All visible awkwardness with new v1.33 (empty lines, gateway 0.0.0.0 for inactive connection) was also on v1.71.

mansr, 2016-10-07:
If you're making a list of problems, the MAC address displayed for wifi connections is that of the local interface, not the access point.

mpvader, 2016-10-08:
connman v1.33 is in meta-victronenergy/master, and part of the v1.80~12 test release which is in develop

mpvader, 2016-10-10:
One thing fixed in v1.30 is that kernel counters are no longer reset to 0:
https://lists.01.org/pipermail/connman/2015-April/019839.html

closed 2016-10-16 by mpvader