victronenergy/venus #53: connman: revisit patches and (again) try upstream the wifi patch

mpvader, 2016-09-15:
Currently used version in Venus is v1.25, and we have 4 patches on it. Besides upgrading connman to a newer version, see #52, the patches need to be revisited. 

At a minimum better document those patches in the meta-victronenergy repository. Even better is to get rid of some or all. Most importantly the wifi patch, that one would be best to get into upstream.

## 1) add hwclock when time is decoded
[0001-Add-set-hwclock-when-time-is-decoded.patch](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-alternatives/recipes-connectivity/connman/connman/0001-Add-set-hwclock-when-time-is-decoded.patch)

I'm not certain what it is for. Perhaps to make sure NTP time updates are also stored into the RTC? Make me wonder if it is really necessary to do that with a patch like this. Is there no more common way to do that? The, only?, other way to change the time on a Venus device is via the gui, and that also relates to connman: [gui/connman/clockmodel.cpp](https://git.victronenergy.com/ccgx/gui/blob/master/connman/clockmodel.cpp)

## 2) ntp max poll set to 16
[0001-ntp-max-poll-set-to-16.patch](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-alternatives/recipes-connectivity/connman/connman/0001-ntp-max-poll-set-to-16.patch)

I'm not certain what it is for. Perhaps to reduce generated network traffic?

## 3) gweb multi-header work around
[gweb-multi-header-workaround.patch](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-alternatives/recipes-connectivity/connman/connman/gweb-multi-header-workaround.patch)

I don't know what this is for.

## 4) ignore invalid key on AR failure
[connman_IgnoreInvalidKey_Arfailure.patch](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-alternatives/recipes-connectivity/connman/connman/connman_IgnoreInvalidKey_Arfailure.patch)

Fixes some issues we have with the way connman handles wifi.

This patch was made by Thiemo. Thiemo: problem with the wifi is the following:
Setting up a Wifi connection consists of a number of phases. When wpa_supplicant makes a certain state transition (G_SUPPLICANT_STATE_4WAY_HANDSHAKE -> G_SUPPLICANT_STATE_DISCONNECTED), connman draws the conclusion that the wifi password was wrong. This is a correct conclusion when Wifi reception is good, but this can also occur on marginal wifi reception. Stock connman will not try to reconnect once it thinks that the password has changed. The reason is that on some networks the account/network card is blocked after to many retries.
See this mail: http://connman.connman.narkive.com/AkqCDM7z/connman-behavior-with-marginally-stabe-wifi
 
This also describes the way that they want the issue to be resolved: Wpa supplicant will report a wifi network gone when it is not seen for 2 minutes. Connman performs regularly a scan for networks when it is not connected. The interval time is doubled with each scan. So when the interval gets more than 2 minutes, wpa_supplicant removes the network and when the next scan rediscovers the network, connman will try to reconnect with the stored passphrase. (at least it was like this in version 1.25... things have changed lately in this area but I don't know what the end result is). So it can take upto something like 6 minutes before connman tries to reconnect to the network (which also has a relative large chance of failing due to the marginal wifi recception).
 
Besides this, there is also an issue when the gui is in the Wifi menu. When it is in this menu, a Wifi scan is started every 10 seconds or so, so that the user gets fast updates on the available networks. But this also means that the network is continuously seen by wpa_supplicant which will not remove it from the list so connman will keep the error set and will not try to reconnect.
 
To work around these problems, the following changes are made in the patch:

1. It adds a plugin called "arfailure", which can reset a failure on a service after a certain amount of time.
Normally, when a service has a failure (like when connman thinks the Wifi password is wrong) it will not automatically try to reconnect to that service.
When enabled, this plugin will reset the failure after a configurable number of seconds and starts an auto connect (but only when in the past there has been a succesfull connection attempt).
 
2. It adds a setting IgnoreInvalidKey.
Normally, when a service has the CONNMAN_SERVICE_ERROR_INVALID_KEY error set, a call to service_connect will not try to connect and return with an error.
When the IgnoreInvalidKey is set, it will ignore this error and still try to connect.
 
I know that these patches do not apply any more with the latest connman. The plugin part might still apply but this alone is probably not enough to solve the issue.

Maybe interesting commits between 1.25 and 1.32:
 
- 2015-04-15 - [service: Do not clear passphrase or identity on error](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=346609214f05b3b4eb49a2b1b03844d94e9eb59d) - Patrik Flykt
- 2015-03-23 - [service: Let service error indicate invalid WiFi PSK passphrase](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=edc02deab0225b3410179c8937e7cfb4058ced7c) - Patrik Flykt
- 2014-10-10 - [service: When connecting, ignore previous invalid key errors](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=026690ff7c5d8a434b26aaedbe70e3580aa5027a) - Patrik Flykt
- 2014-10-10 - [service: Don't save or load reason for failure](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=afb38d5f7b6db1a3446101602a39d026ab583426) - Patrik Flykt
- 2014-09-05 - [service: Use proper function when setting service Passphrase attribute](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=52212edaaa75250b5d73bab503ffe0d81894d605) - Patrik Flykt
- 2014-09-05 - [service: Remove __connman_service_add_passphrase()](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=2d909f4dd2c3df4498627cdb476280080fa673fd) - Patrik Flykt
- 2014-09-05 - [service: Clean up __connman_service_set_passphrase()](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=eb694de89baca81b9054491538635984d2b03c56) - Patrik Flykt
- 2014-09-05 - [service: Make a specifig function for clearing a passphrase on error](http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=bfa49a6928de961349d6f490ea4ba47f497939b6) - Patrik Flykt

mpvader, 2016-09-15:
Regarding 1. hwclock: read this to see ConnMans maintainers opinion on syncing to the hwclock:
https://bugs.tizen.org/jira/si/jira.issueviews:issue-html/TC-165/TC-165.html

Also, rtc and all related datetime mgmt works fine on the ccgx. No complaints and no issues. So just adding a few explanation lines on that patch, for example a link to this issue, is more than enough.

@jhofstee: you'll be happy to know that systemd is coming and all will be better. See Tizen link :smile:.

mpvader, 2016-09-18:
Some extra info:

about 1) "normally" the hwclock is saved during shutdown / reboot. Since
a color control can just be removed from power, we always update the rtc as well.

about 3) g-web multi header:

it belongs to this thread:
- https://www.mail-archive.com/connman@connman.net/msg15637.html
- https://www.mail-archive.com/connman@connman.net/msg15731.html

mansr, 2016-09-20:
Some findings:

## 1) hwclock
The patch is not suitable for upstream, but it's small and should be easy to maintain. An alternative could be a cron job that periodically syncs the hardware clock.

## 2) ntp max poll
From RFC5905:
> Poll: 8-bit signed integer representing the maximum interval between successive messages, in log2 seconds.  Suggested default limits for minimum and maximum poll intervals are 6 and 10, respectively.

Increasing the value from 10 to 16 changes the maximum interval from 17 minutes to 18 hours although there actual intervals may be shorter depending on network conditions and clock stability.

The patch as is would probably not be accepted upstream as it changes behaviour without a good reason. A new option allowing this value to be set in the config file might have a better chance.

## 3) gweb multi-header
An attempt to ask about this on the connman mailing list has received no response, and the message to the list appears to have been black-holed.

mpvader, 2016-09-20:
@mansr 

1) hwclock: let's leave that as a patch

2) ntp max poll: I'd expect the rationale is to reduce network traffic. Let's leave as a patch as well

3) gweb multi-header: have you looked into reproducing that yet?

4) wifi: ignore invalid key on AR failure:
you already made the patch apply to the latest Connman version right? Is there much work involved in testing its effectiveness?

I'd prefer to first update our Connman to latest version including these patches; except possibly be 3. And then see about mainlining patch (4).

mansr, 2016-10-05:
The gweb patch is now upstream: http://git.kernel.org/cgit/network/connman/connman.git/commit/?id=22901212105055b24f73504a74f5a57eee809777

mpvader, 2016-11-10:
@mansr how is the upstreaming going?

mpvader, 2017-08-03:
just for record keeping, the last attempt at upstreaming the 'dont ever give up' wifi patch is here:
https://lists.01.org/pipermail/connman/2017-January/021470.html

And the patch is also [here](https://github.com/victronenergy/meta-victronenergy/commit/c4a316116f359ddbbf832c741b0cde29aa757a12). It is scheduled to replace what we have in meta-victronenergy/master now soon.

mpvader, 2017-08-04:
I've sent connman a reminder email: https://lists.01.org/pipermail/connman/2017-August/021993.html

jhofstee, 2018-04-20:
Since upstreaming the remaining patches seems impossible, I am closing this issue.

closed 2018-04-20 by jhofstee