victronenergy/venus #54: make CONFIG=jethro on clean checkout error

mpvader, 2016-09-17:
See log at line ~14. Probably something small, but I don't have time to look further into this now.

Also it doesn't seem to prevent building. 

```
matthijs@matthijs-VirtualBox:/media/venusbuilds$ git clone git@github.com:victronenergy/venus.git venus-jethro
Cloning into 'venus-jethro'...
remote: Counting objects: 606, done.
remote: Total 606 (delta 0), reused 0 (delta 0), pack-reused 606
Receiving objects: 100% (606/606), 87.63 KiB | 0 bytes/s, done.
Resolving deltas: 100% (335/335), done.
Checking connectivity... done.
matthijs@matthijs-VirtualBox:/media/venusbuilds$ cd venus-jethro/
matthijs@matthijs-VirtualBox:/media/venusbuilds/venus-jethro$ make CONFIG=jethro
ln -s configs/jethro conf
find: `sources': No such file or directory
matthijs@matthijs-VirtualBox:/media/venusbuilds/venus-jethro$ make fetch

--------------- sources/bitbake - https://github.com/victronenergy/bitbake.git -------------
Cloning into 'sources/bitbake'...
remote: Counting objects: 43713, done.
Receiving objects: 100% (43713/43713), 21.09 MiB | 4.54 MiB/s, done.
remote: Total 43713 (delta 0), reused 0 (delta 0), pack-reused 43713
Resolving deltas: 100% (28924/28924), done.
Checking connectivity... done.
Branch 1.28 set up to track remote branch 1.28 from origin.
Switched to a new branch '1.28'
From https://github.com/openembedded/bitbake
 * branch            1.28       -> FETCH_HEAD
 * [new branch]      1.28       -> upstream/1.28
Branch 1.28 set up to track remote branch 1.28 from upstream.

--------------- sources/meta-openembedded - https://github.com/victronenergy/meta-openembedded.git -------------
Cloning into 'sources/meta-openembedded'...
remote: Counting objects: 61442, done.
remote: Total 61442 (delta 0), reused 0 (delta 0), pack-reused 61441
Receiving objects: 100% (61442/61442), 26.51 MiB | 4.13 MiB/s, done.
Resolving deltas: 100% (34781/34781), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
From https://github.com/victronenergy/meta-openembedded
 * branch            jethro     -> FETCH_HEAD
Branch jethro_venus set up to track remote branch jethro from origin.

[....]
```

sancloud, 2016-09-17:
Still doesn't build.  I have created a virgin machine (Ubuntu 14.04). Get a fresh checkout and attempt to build:
`sancloud@ubuntu:~/Devel/Victron$ rm -rf venus/
sancloud@ubuntu:~/Devel/Victron$ 
sancloud@ubuntu:~/Devel/Victron$ git clone https://github.com/victronenergy/venus.git
Cloning into 'venus'...
remote: Counting objects: 606, done.
remote: Total 606 (delta 0), reused 0 (delta 0), pack-reused 606
Receiving objects: 100% (606/606), 87.63 KiB | 0 bytes/s, done.
Resolving deltas: 100% (335/335), done.
Checking connectivity... done.
sancloud@ubuntu:~/Devel/Victron$ cd venus/
sancloud@ubuntu:~/Devel/Victron/venus$ 
sancloud@ubuntu:~/Devel/Victron/venus$ 
sancloud@ubuntu:~/Devel/Victron/venus$ make CONFIG=jethro
ln -s configs/jethro conf
find: `sources': No such file or directory
sancloud@ubuntu:~/Devel/Victron/venus$ make fetch

--------------- sources/bitbake - https://github.com/victronenergy/bitbake.git -------------
Cloning into 'sources/bitbake'...
remote: Counting objects: 43713, done.
remote: Total 43713 (delta 0), reused 0 (delta 0), pack-reused 43713
Receiving objects: 100% (43713/43713), 21.09 MiB | 5.42 MiB/s, done.
Resolving deltas: 100% (28924/28924), done.
Checking connectivity... done.
Branch 1.28 set up to track remote branch 1.28 from origin.
Switched to a new branch '1.28'
From https://github.com/openembedded/bitbake
 * branch            1.28       -> FETCH_HEAD
 * [new branch]      1.28       -> upstream/1.28
Branch 1.28 set up to track remote branch 1.28 from upstream.

--------------- sources/meta-openembedded - https://github.com/victronenergy/meta-openembedded.git -------------
Cloning into 'sources/meta-openembedded'...
remote: Counting objects: 61442, done.
remote: Total 61442 (delta 0), reused 0 (delta 0), pack-reused 61441
Receiving objects: 100% (61442/61442), 26.51 MiB | 5.99 MiB/s, done.
Resolving deltas: 100% (34781/34781), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
From https://github.com/victronenergy/meta-openembedded
 * branch            jethro     -> FETCH_HEAD
Branch jethro_venus set up to track remote branch jethro from origin.

--------------- sources/meta-qt5 - https://github.com/victronenergy/meta-qt5.git -------------
Cloning into 'sources/meta-qt5'...
remote: Counting objects: 6646, done.
remote: Total 6646 (delta 0), reused 0 (delta 0), pack-reused 6646
Receiving objects: 100% (6646/6646), 1.91 MiB | 1.48 MiB/s, done.
Resolving deltas: 100% (4044/4044), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
From https://github.com/meta-qt5/meta-qt5
 * branch            jethro     -> FETCH_HEAD
 * [new branch]      jethro     -> upstream/jethro
Branch jethro_venus set up to track remote branch jethro from upstream.

--------------- sources/meta-swupdate - https://github.com/victronenergy/meta-swupdate.git -------------
Cloning into 'sources/meta-swupdate'...
remote: Counting objects: 346, done.
remote: Total 346 (delta 1), reused 1 (delta 1), pack-reused 344
Receiving objects: 100% (346/346), 55.51 KiB | 0 bytes/s, done.
Resolving deltas: 100% (109/109), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
From https://github.com/victronenergy/meta-swupdate
 * branch            master     -> FETCH_HEAD
Branch master set up to track remote branch master from origin.

--------------- sources/meta-victronenergy - https://github.com/victronenergy/meta-victronenergy.git -------------
Cloning into 'sources/meta-victronenergy'...
remote: Counting objects: 4254, done.
remote: Compressing objects: 100% (233/233), done.
remote: Total 4254 (delta 124), reused 0 (delta 0), pack-reused 4013
Receiving objects: 100% (4254/4254), 795.17 KiB | 1.50 MiB/s, done.
Resolving deltas: 100% (2089/2089), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Branch master set up to track remote branch master from origin.

--------------- sources/openembedded-core - https://github.com/victronenergy/openembedded-core.git -------------
Cloning into 'sources/openembedded-core'...
remote: Counting objects: 292226, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 292226 (delta 5), reused 5 (delta 5), pack-reused 292217
Receiving objects: 100% (292226/292226), 111.05 MiB | 5.93 MiB/s, done.
Resolving deltas: 100% (202680/202680), done.
Checking connectivity... done.
Branch jethro_venus set up to track remote branch jethro_venus from origin.
Switched to a new branch 'jethro_venus'
From https://github.com/victronenergy/openembedded-core
 * branch            jethro     -> FETCH_HEAD
Branch jethro_venus set up to track remote branch jethro from origin.
sancloud@ubuntu:~/Devel/Victron/venus$ make beaglebone-venus-image
export MACHINE=beaglebone && . ./sources/openembedded-core/oe-init-build-env build sources/bitbake && bitbake venus-image
/bin/sh: 47: .: Can't open /home/sancloud/Devel/Victron/venus/scripts/oe-buildenv-internal
make: *** [beaglebone-venus-image] Error 2
sancloud@ubuntu:~/Devel/Victron/venus$`

jhofstee, 2016-09-17:
for what it is worth `make CONFIG=jethro fetch` doesn't display the error.

jhofstee, 2016-09-17:
that looks weird, are you sure you use bash instead of dash?

mpvader, 2016-09-17:
@jhofstee is it an idea to add dash detection in the makefile somehow? Would save people lots of time.

jhofstee, 2016-09-17:
it is an OE thing, not venus related,

./sources/openembedded-core/oe-init-build-env is the proper place to fix it:

#
# Normally this is called as '. ./oe-init-build-env builddir'
#
# This works in most shells (not dash), but not all of them pass arg1 when
# being sourced.   To workaround the shell limitation use "set arg1" prior
# to sourcing this script.
#
if [ -n "$BASH_SOURCE" ]; then
    OEROOT="`dirname $BASH_SOURCE`"
elif [ -n "$ZSH_NAME" ]; then
    OEROOT="`dirname $0`"
else
    OEROOT="`pwd`"
fi

sancloud, 2016-09-17:
I am using bash
sancloud@ubuntu:~/Devel/Victron/venus$ ps ch -o %c $$
bash

sancloud, 2016-09-17:
I have tried the checkout with 'make CONFIG=jethro fetch' and it still fails when I try to build.

There is something weird here or I am not using the correct build command once I have fetched the repo's.

Why is it trying to open 'scripts/oe-buildenv-internal' when building?
/bin/sh: 47: .: Can't open /home/sancloud/Devel/Victron/venus/scripts/oe-buildenv-internal

Using:
# make beaglebone-venus-image

jhofstee, 2016-09-18:
Hi, that is you current shell. Where is /bin/sh pointing to?

jhofstee, 2016-09-18:
Hi,

On 17-09-16 21:36, SanCloud wrote:
>
> I have tried the checkout with 'make CONFIG=jethro fetch' and it still 
> fails when I try to build.
>

It was just to prevent the (harmless) error Matthijs initially reported.

> There is something weird here or I am not using the correct build 
> command once I have fetched the repo's.
>
the build command is fine..

> Why is it trying to open 'scripts/oe-buildenv-internal' when building?
> /bin/sh: 47: .: Can't open 
> /home/sancloud/Devel/Victron/venus/scripts/oe-buildenv-internal
>
see sources/openembedded-core/oe-init-build-env, this part fails:

# This works in most shells (not dash), but not all of them pass arg1 when
# being sourced.   To workaround the shell limitation use "set arg1" prior
# to sourcing this script.
#
if [ -n "$BASH_SOURCE" ]; then
    OEROOT="`dirname $BASH_SOURCE`"
elif [ -n "$ZSH_NAME" ]; then
    OEROOT="`dirname $0`"
else
    OEROOT="`pwd`"
fi

you take the last else, so you end up with OEROOT = venus-path and that of
course fails miserable since it start looking for oe scripts inside 
venus etc.

jhofstee, 2016-09-18:
better idea, just always use bash:

https://github.com/victronenergy/venus/commit/c043b75e4058a7c387976af0744ddabde698f1cf

mpvader, 2016-09-18:
:smile:

@sancloud you have fixed it in the end, right? was it indeed dash/bash issue?

jhofstee, 2016-09-18:
On 17-09-16 10:00, mpvader wrote:
> See log at line ~14. Probably something small, but I don't have time
> to look further into this now.
>
> Also it doesn't seem to prevent building.
>
> |matthijs@matthijs-VirtualBox:/media/venusbuilds$ git [...]

About the original issue, you lack a conf target here, so make will just 
pick
the first one, which is build/conf/bblayers.conf and that needs the sources.

The most useful default command I can up with is 
https://github.com/victronenergy/venus/commits/help...
Any comments?

mpvader, 2016-09-18:
> About the original issue, you lack a conf target here, so make will just 
> pick
> the first one, which is build/conf/bblayers.conf and that needs the sources.

Aha

> 
> The most useful default command I can come up with is 
> https://github.com/victronenergy/venus/commits/help...
> Any comments?

It's great! Now clean out the readme and add a link to wiki home to it?

Wiki is a lot easier to update since it's not in the repo contents.

sancloud, 2016-09-18:
Woohoo, thank you just got head and it seems to be building.. at least its started to parse recipes.

Just 3000 more tasks to run.

FYI if you install a virgin machine (Ubuntu 14.04) you need to add the additional components:
sudo apt-get install g++
sudo apt-get install gawk
sudo apt-get install chrpath
sudo apt-get install texinfo

mpvader, 2016-09-18:
On 18 Sep 2016, at 15:12, SanCloud <notifications@github.com<mailto:notifications@github.com>> wrote:

Woohoo, thank you just got head and it seems to be building.. at least its started to parse recipes.

Great!

Just 3000 more tasks to run.

FYI if you install a virgin machine (Ubuntu 14.04) you need to add the additional components:
sudo apt-get install g++
sudo apt-get install gawk
sudo apt-get install chrpath
sudo apt-get install texinfo

See make prereq

Matthijs

mpvader, 2016-09-26:
@jhofstee: probably the help branch in Venus should be deleted now?

closed 2016-09-26 by mpvader