victronenergy/venus #55: dummy issue to hold pictures

mpvader, 2016-09-19:
in case you want to add a picture to the wiki, just paste it here and grab the url.

mpvader, 2016-09-19:
[image]

mpvader, 2016-09-19:
[image]

closed 2016-09-26 by mpvader

jepefe, 2016-10-11:
[image]
[image]

jepefe, 2016-10-11:
[image]

mpvader, 2017-03-15:
[image]

mpvader, 2017-12-28:
[image]

3dGrabber, 2018-02-05:
[image]

3dGrabber, 2018-02-05:
[image]

3dGrabber, 2018-02-05:
[image]

3dGrabber, 2018-02-05:
[image]

3dGrabber, 2018-02-05:
[image]

mpvader, 2018-02-12:
[image]

AlexImb, 2018-10-26:
[image]