victronenergy/venus #58: uncaught exception in vrmlogger related to umount/mount

mpvader, 2016-10-02:
While testing/fixing sdcard and usbstick mount issues on the bbb, one thing lead to another and I had a look in the vrmlogger log file.

```
(clean boot, no card inserted while booting)
2016-10-02 19:21:48.841896500 MainThread-vrmlogger: /opt/color-control/vrmlogger/vrmlogger.py v2.55 is starting up
2016-10-02 19:21:48.852861500 MainThread-vrmlogger: Loglevel set to INFO
2016-10-02 19:21:48.902564500 MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
2016-10-02 19:21:49.003355500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
2016-10-02 19:21:49.057322500 MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
2016-10-02 19:21:49.099631500 MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
2016-10-02 19:21:52.738600500 MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2016-10-02 19:21:52.935952500 MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
2016-10-02 19:21:54.023924500 MainThread-dbusmonitor: Found: com.victronenergy.tank.builtin_adc2_di2 matches com.victronenergy.tank, scanning and storing items
2016-10-02 19:21:54.030030500 MainThread-dbusmonitor:        com.victronenergy.tank.builtin_adc2_di2 has device instance 22
2016-10-02 19:21:54.079282500 MainThread-dbusmonitor: Found: com.victronenergy.tank.builtin_adc4_di0 matches com.victronenergy.tank, scanning and storing items
2016-10-02 19:21:54.085280500 MainThread-dbusmonitor:        com.victronenergy.tank.builtin_adc4_di0 has device instance 20
2016-10-02 19:21:54.138449500 MainThread-dbusmonitor: Found: com.victronenergy.tank.builtin_adc6_di1 matches com.victronenergy.tank, scanning and storing items
2016-10-02 19:21:54.144644500 MainThread-dbusmonitor:        com.victronenergy.tank.builtin_adc6_di1 has device instance 21
2016-10-02 19:21:54.204117500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
2016-10-02 19:21:54.299341500 MainThread-kwhdeltas: active battery service: None
2016-10-02 19:21:54.319379500 MainThread-kwhdeltas: active battery service: None
2016-10-02 19:21:54.421632500 MainThread-vrmlogger: Starting to log, logmode == LogToVRM
2016-10-02 19:21:54.447147500 MainThread-http_endpoint: Internal backlog buffer initialised. Count = 0
2016-10-02 19:21:54.448491500 MainThread-http_endpoint: Found '' as remembered external path
2016-10-02 19:21:54.449853500 MainThread-http_endpoint: Checking '' for suitability for external storage
2016-10-02 19:21:54.450765500 MainThread-http_endpoint: Mount point to check was empty.
2016-10-02 19:21:54.451614500 MainThread-http_endpoint: Because we haven't found a suitable remembered external storage path, we're scanning for one to use.
2016-10-02 19:21:54.453771500 MainThread-http_endpoint: Checking '/media/mmcblk0p1' for suitability for external storage
2016-10-02 19:21:54.460931500 MainThread-http_endpoint: Marking /media/mmcblk0p1 as candidate for guessed mount point
2016-10-02 19:21:54.462641500 MainThread-http_endpoint: Changing remembered external storage path from '' to '/media/mmcblk0p1'
2016-10-02 19:21:54.479327500 MainThread-http_endpoint: Found /media/mmcblk0p1 as external storage dir
2016-10-02 19:21:54.929884500 MainThread-http_endpoint: Extra SQLite DB made at /media/mmcblk0p1/vrmlogger-backlog.sqlite3
2016-10-02 19:21:55.071478500 MainThread-http_endpoint: Made /media/mmcblk0p1/vrmlogger-backlog.sqlite3 the active buffer
2016-10-02 19:21:55.170166500 MainThread-http_endpoint: Starting sender thread
2016-10-02 19:21:55.205074500 MainThread-http_endpoint: Started sender thread
2016-10-02 19:21:55.216901500 MainThread-http_endpoint: HttpEndpoint initialised, http worker thread running
2016-10-02 19:21:55.638463500 MainThread-vrmlogger: Starting mainloop, responding on only events
2016-10-02 19:32:23.938250500 MainThread-dbusmonitor: umount event!
2016-10-02 19:32:23.944554500 MainThread-http_endpoint: Umount event:/dev/mmcblk0p1 -> /media/mmcblk0p1
2016-10-02 19:32:24.152796500 MainThread-connection: Exception in handler for D-Bus signal:
2016-10-02 19:32:24.152809500 Traceback (most recent call last):
2016-10-02 19:32:24.152813500   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
2016-10-02 19:32:24.152818500     self._handler(*args, **kwargs)
2016-10-02 19:32:24.152821500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 342, in _umount_event
2016-10-02 19:32:24.152826500     self.stop()
2016-10-02 19:32:24.152828500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 424, in stop
2016-10-02 19:32:24.152981500     self._process_results() # Otherwise your last processed job won't be saved if it failed.
2016-10-02 19:32:24.152987500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 480, in _process_results
2016-10-02 19:32:24.152991500     self._feed_from_buffer_to_queue()
2016-10-02 19:32:24.152995500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 430, in _feed_from_buffer_to_queue
2016-10-02 19:32:24.153067500     self._enough_free_space_on_storage() # To update free space display
2016-10-02 19:32:24.153072500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 359, in _enough_free_space_on_storage
2016-10-02 19:32:24.153077500     stat = os.statvfs(f)
2016-10-02 19:32:24.153080500 OSError: [Errno 2] No such file or directory: '/media/mmcblk0p1'
2016-10-02 19:32:24.173876500 MainThread-dbusmonitor: umount event!
2016-10-02 19:32:24.175326500 MainThread-http_endpoint: Umount event:/dev/mmcblk0 -> /media/mmcblk0
2016-10-02 19:32:24.179123500 MainThread-connection: Exception in handler for D-Bus signal:
2016-10-02 19:32:24.179134500 Traceback (most recent call last):
2016-10-02 19:32:24.179137500   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
2016-10-02 19:32:24.179142500     self._handler(*args, **kwargs)
2016-10-02 19:32:24.179146500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 342, in _umount_event
2016-10-02 19:32:24.179150500     self.stop()
2016-10-02 19:32:24.179153500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 424, in stop
2016-10-02 19:32:24.179274500     self._process_results() # Otherwise your last processed job won't be saved if it failed.
2016-10-02 19:32:24.179280500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 480, in _process_results
2016-10-02 19:32:24.179285500     self._feed_from_buffer_to_queue()
2016-10-02 19:32:24.179288500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 430, in _feed_from_buffer_to_queue
2016-10-02 19:32:24.179327500     self._enough_free_space_on_storage() # To update free space display
2016-10-02 19:32:24.179332500   File "/opt/color-control/vrmlogger/http_endpoint.py", line 359, in _enough_free_space_on_storage
2016-10-02 19:32:24.179337500     stat = os.statvfs(f)
2016-10-02 19:32:24.179340500 OSError: [Errno 2] No such file or directory: '/media/mmcblk0p1'
root@beaglebone:/media#
```

wiebeytec, 2016-10-03:
On what device is this? Is there remote access to it?

Is that the entire log? I would expect a fallback on the internal storage.

mpvader, 2016-10-11:
since this is double with #59, I'll close it.

closed 2016-10-11 by mpvader