victronenergy/venus #59: umount fails when taking sdcard out without ejecting from gui

mpvader, 2016-10-02:
Reproducable with an sdcard *not* containing any swu or other files that make vrmlogger ignore it.

1st insert OK
1st removal fails to unmount
2nd insert fails
2nd removal OK to unmount
3rd insert OK
3rd removal fail
etc 

I expect the reason for the first unmount to fail is that vrmlogger still has the file open. The umount fails because of that. But still vrmlogger will receive the dbus-signal from mount.sh that it is time to unmount. So, a few seconds later the device is no longer busy, and it would be possible to umount. Doesn't happen though: udev doesn't try again a few seconds later. Also there is no -l (Lazy unmount) flag used. Yet (?).

Then next insert fails or doesn't happen, because the device with that name is still mounted.

The next umount succeeds: vrmlogger does not have a file open.

etc.

One solution would be to use the -l flag:
```
-l Lazy unmount. Detach the filesystem from the filesystem hierarchy now, and cleanup all references to the filesystem as soon as it is not busy anymore. This option allows a "busy" filesystem to be unmounted. (Requires kernel 2.4.11 or later.)
```

It would be applied here:
```
diff --git a/recipes-bpp3/recipes-core/udev/files/mount.sh b/recipes-bpp3/recipes-core/udev/files/mount.sh
index 5c4bfa2..2a6dd54 100644
--- a/recipes-bpp3/recipes-core/udev/files/mount.sh
+++ b/recipes-bpp3/recipes-core/udev/files/mount.sh
@@ -59,7 +59,7 @@ fi
 if [ "$ACTION" = "remove" ] && [ -x "$UMOUNT" ] && [ -n "$DEVNAME" ]; then
    for mnt in `cat /proc/mounts | grep "$DEVNAME" | cut -f 2 -d " " `
    do
-       $UMOUNT $mnt
+       $UMOUNT -l $mnt
    done
    
    # Remove empty directories from auto-mounter
``

Which actually [has been discussed on the openembedded-devel mailinglist](http://lists.openembedded.org/pipermail/openembedded-devel/2010-June/066241.html). But somehow that was never applied.

mpvader, 2016-10-02:
relates to #58

mpvader, 2016-10-02:
while looking at this, also consider if adding -o sync to the mount commands might be a good idea.

wiebeytec, 2016-10-03:
I'm not sure `-o sync` will be a good idea. I don't remember the details, but some distribution decided to disable cache for USB sticks, which led to rapid destruction of the sectors on which the FAT was located, because it was updated every operation.

mansr, 2016-10-03:
With or without -o sync (which does cause cards to wear out quickly), removing a storage device without unmounting it first is just wrong. Lazy umount doesn't really help either since it would just cause a subsequent mount of a newly inserted card (the same or a different one) to fail. Moreover, pulling the card while the kernel has unfinished writes will result in nasty errors (possibly preventing further use of the card slot before a reboot) at best, a permanently ruined card at worst.

mpvader, 2016-10-03:
> With or without -o sync (which does cause cards to wear out quickly)

allright thanks both: forget about it.

> removing a storage device without unmounting it first is just wrong. 

Yes agreed. But unfortunately I cannot stop a user from doing that.

We can teach him though! Quite easy, now that I think of it: in mount.sh detect that this happened, and then show a toast in the gui. I'll discuss with @jepefe how he needs this signal to arrive in the gui.

> Lazy umount doesn't really help either since it would just cause a subsequent
> mount of a newly inserted card (the same or a different one) to fail.

I am not so sure that a subsequent mount will fail: with lazy umount enabled, umount will be finalized a few seconds later (the dbus signal you see in mount.sh triggers the process to release the file). So a subsequent mount of a newly inserted card will work fine. I'll test it later today.

mansr, 2016-10-03:
mpvader <notifications@github.com> writes:

> Yes agreed. But unfortunately I cannot stop a user from doing that.
>
> We can teach him though! Quite easy, now that I think of it: in mount.sh detect that this happened, and then show a toast in the [...]

Yes, the best we can do is complain if we notice someone doing it.  Even
Windows requires you to "safely remove hardware."

> I am not so sure that a subsequent mount will fail: with lazy umount enabled, umount will be finalized a few seconds later (the dbus signal you see in mount.sh triggers the process to release the [...]

Unless there were pending writes in the kernel.  I've seen a badly timed
removal cause the unmount to fail such that any process touching the
mount point or device is left in permanent D state.

-- 
M�ns Rullg�rd

jepefe, 2016-10-06:
> We can teach him though! Quite easy, now that I think of it: in mount.sh detect that this happened, and then show a toast in the gui. I'll discuss with @jepefe how he needs this signal to arrive in the gui.

For this purpose we can add a new state to the /Storage/MountState path in the logger DBus service.
0 = Not mounted
1 = Mounted
2 = Unmount desired
3 = Bussy unmounting
**4 = Disk not  ejected properly**

Create a new DBus path also in other service still a valid option if is easier to implement on your side.

mpvader, 2016-10-06:
Instead of a state, it would be better that this is an event. The umount and mount events are sent from [/etc/udev/scripts/mount.sh](https://git.victronenergy.com/ccgx/meta-ccgx/blob/master/recipes-bpp3/recipes-core/udev/files/mount.sh):

```
dbus-send --system --type=signal / com.victronenergy.udev.mount string:$DEVNAME string:"/media/$name"
dbus-send --system --type=signal / com.victronenergy.udev.umount string:$DEVNAME string:"/media/$name"
```

I suppose we could just add a third parameter to the unmount signal, for example uint16:0 on a clean unmount, and uint16:1 on a dirty eject.

And then in the gui listen for this signal: there is no reason to route this through the vrmlogger.

Fyi: the code in vrmlogger listening for this signal is here:
https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/http_endpoint.py#L82

@jepefe: you can test the gui implementation simply by running this from the command line:
```
dbus-send --system --type=signal / com.victronenergy.udev.umount string:mmctest string:"/media/mmctest" uint16:1
```

mansr, 2016-10-06:
That seems like a simpler approach.

mpvader, 2016-10-11:
I [sent an email to the oe-core mailing list](http://lists.openembedded.org/pipermail/openembedded-core/2016-October/127438.html) about lazy umount, hoping to find out why they never added the -l patch.

And after that I dicussed it some more with @wiebeytec: he will futher look into vrmlogger and lazy umount, to fix the subsequent mount fails issue.

wiebeytec, 2016-10-12:
In the mount script, we can detect dirty mounts because the 'remove' event takes place when the mount still exists. To that end, I have made the umount lazy (and fixed vrmlogger to be able to handle that) and added this parameter. I will make the necessary pull requests / poke package maintainers.

mpvader, 2016-12-02:
https://github.com/victronenergy/meta-victronenergy/commit/fe77ae178f6bc04974298516746f2c76976e86b0

closed 2016-12-02 by mpvader