victronenergy/venus #62: Properly handle failure to mount /data

mpvader, 2016-10-10:
- Ccgx: show failure or something like that on the display
- Bbb: could do something with its led

And preferably keep vrmlogger running so the error is visible remotely.

As is, the processes will start probably use the mount point (/data) in the rootfs. Possibly just continuing like all is fine and using default settings.

mpvader, 2019-01-11:
First part is done and scheduled for v2.30~23. Over to @izak for vrmlogger and the gui.

izak, 2019-01-15:
The existence of /run/data-partition-failed is logged as `dpf=1` (data partition failed), and `dpf=0` otherwise. GuI already done by jhostee. Over to @wiebeytec .

mpvader, 2019-01-15:
@wiebeytec this is quite urgent; see above for whats special here; besides just adding the new dataattribute.

wiebeytec, 2019-01-15:
What does `/run/data-partition-failed` mean as of yet? Mounted read-only or not mounted?

Remounting read-only can happen at any time, dynamically. Does whatever writes `/run/data-partition-failed` check `/data` continuously, or only on boot?

Also, the current implementation in vrmlogger likely won't work properly: when backlog is used (even when very briefly) and the sqlite file is read only, any use of it will make it crash, or go into undefined behavior, and `dpf` won't be send out. Plus, any significant backlog will make it appear late. It would be convoluted to write a bypass for the `dpf` attribute, and allow for sending it while read-only.

It's  better to call `vrmhttp.send_to_vrm()` directly, as is done for the auth keys by the old `vrmpubnub.py`, and the to-be-used `vnc_ssh_sender.py`.

`send_to_vrm()` takes a command type (`ANNOUNCE, SENDDATA, CONFIGCHANGE, HOURLYDELTAS, STORE_AUTH_PACKAGES, STORE_SSH_VNC_PRIVATE_KEY, TEST_POST`), and I suggest making a new one. On the server, I can deal with these separately then, also before any auth token checks are done.

wiebeytec, 2019-01-15:
I'm not allowed to edit the ticket? No rights?

Anyway, we discussed in real life/telephone:

- Jeroen will verify a corrupt file system is mounted read-only from the start, vs _will become_ read-only later. If read-only from the start, that's what we're going to report; no constant checks for `/data` being read-only. I suspect once an fs is corrupt and has become read-only, subsequent mounts will just fail.
- Jeroen's script that makes `/run/data-partition-failed` will also post this event to VRM using curl, with a new command type. Not using vrmlogger, because it won't function at all with a read-only data partition.
- Wiebe will fix vrmlogger to handle a non-functional sqlite. When that functions properly, we can remove the above curl call.

Questions:

Do we revert adding the `dpf` attribute commit?

jhofstee, 2019-01-22:
Turns out that ubifs can make itself read-only during any io, so also after mounting. So more things have to be done to be able to log this events, details are here, https://github.com/victronenergy/meta-victronenergy/commits/data-ro-check at the moment.

/run/data-partition-state can contain the following states (and vrmlogger will use the number)
```
0: fine
1: failed-to-mount
2: failed-once
3: recovered
4: failed
```
"failed" is logged by curl. vrmlogger should log the others. The "failed-once" -> "recovered" can change at runtime.

wiebeytec, 2019-02-04:
(Start of) [Grafana dashboard](https://grafana.victronenergy.com/d/pyBDBKfmz/venus-data-partition).

closed 2019-08-23 by mpvader