victronenergy/venus #63: CCGX Groningen-hub2 stuck early in the morning of the 9th

mpvader, 2016-10-10:
- on the 8th during the day, version v1.80~12 was installed.
- early morning on the 9th, the unit updated, and rebooted, all still ok
- after or during the boot there are issues, the unit was no longer accessible:
  - display stuck, and not responding to button presses
  - not accessible via console on vedirect 1: but that makes sense, console was not enabled. It is now.
  - not accessible via ssh: does not make sense, as remote support was enabled.
  - not logging to the vrm portal. Latest message 
  - local settings was not started: swupdate could not reach it to determine the update setting

Then around 9am on the 10th, the unit was manually rebooted. and accessible again.

a quick scan of below logs shows that the unit was extremely busy or slow. Which for some reason did not lead to a reboot. also there is a strange error in the messages (WARNING: at net/sched/sch_generic.c:255 dev_watchdog+0x26c/0x28c()). But I don't know if that is a result of the slowness, or causing it.

Below are the messages and swupdate logs:

messages log:
```
Oct  9 02:05:01 CCGX cron.info /USR/SBIN/CROND[21788]: (root) CMD (/opt/victronenergy/swupdate-scripts/check-updates.sh -auto -delay)
Oct  9 02:05:01 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (*** Checking for updates ***)
Oct  9 02:05:01 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (arguments: -auto -delay)
Oct  9 02:05:01 CCGX cron.err /USR/SBIN/CROND[21796]: (CRON) EXEC FAILED (/usr/sbin/sendmail): No such file or directory
Oct  9 02:05:01 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Sleeping for 847 seconds)
Oct  9 02:13:50 CCGX daemon.info connmand[1614]: ntp: time slew +0.052286 s
Oct  9 02:19:08 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Retrieving latest version (feed=develop)...)
Oct  9 02:19:14 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (installed: 201610031109 v1.80~10)
Oct  9 02:19:14 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (available: 201610081208 v1.80~12)
Oct  9 02:19:14 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Starting swupdate to install version 201610081208 v1.80~12 ...)
Oct  9 02:19:15 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (software set: stable mode: copy1)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Swupdate v2016.07.0)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ()
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Licensed under GPLv2. See source distribution for detailed copyright notices.)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ()
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Searching for custom LUA handlers :[string "require ("swupdate_handlers")"]:1: module 'swupdate_handlers' not found:)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no field package.preload['swupdate_handlers'])
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/share/lua/5.2/swupdate_handlers.lua')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/share/lua/5.2/swupdate_handlers/init.lua')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/lib/lua/5.2/swupdate_handlers.lua')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/lib/lua/5.2/swupdate_handlers/init.lua')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file './swupdate_handlers.lua')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/lib/lua/5.2/swupdate_handlers.so')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file '/usr/lib/lua/5.2/loadall.so')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	no file './swupdate_handlers.so')
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Custom handlers not found, no error, skipping...)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ()
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Registered handlers:)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	archive)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	tar)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	flash)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	flash-hamming1)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	raw)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	rawfile)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	ubivol)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	ubipartition)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	lua)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	shellscript)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	preinstall)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	postinstall)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (	uboot)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Main loop Daemon)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : SWUPDATE started :  Software Update started !)
Oct  9 02:19:16 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : SWUPDATE downloading :  )
Oct  9 02:19:21 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (scan_for_ubi_devices)
Oct  9 02:19:21 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (Version 0.1.0)
Oct  9 02:19:21 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : SWUPDATE running :  Installing image bpp3-rootfs-ccgx.ubifs into volume /dev/ubi0_0(rootfs1))
Oct  9 02:20:29 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : SWUPDATE running :  Installation in progress)
Oct  9 02:20:30 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : SWUPDATE successful ! SWUPDATE successful !)
Oct  9 02:20:30 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT ([NOTIFY] : No SWUPDATE running :  Waiting for requests...)
Oct  9 02:20:33 CCGX cron.info /USR/SBIN/CROND[21787]: (root) CMDOUT (do_swupdate completed OK. Rebooting)
Oct  9 02:20:33 CCGX user.notice shutdown[27411]: shutting down for system reboot
Oct  9 02:20:33 CCGX daemon.info init: Switching to runlevel: 6
[... 1515 lines cut ...]
2016-10-09 16:22:02.018463500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-09 16:22:02.018493500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-09 18:21:55.179321500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-09 18:21:55.182098500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-09 20:21:55.024963500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-09 20:21:55.024993500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-09 22:22:11.557739500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-09 22:22:11.557739500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-10 00:22:17.525573500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-10 00:22:17.525573500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-10 02:22:29.579925500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-10 02:22:29.582031500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-10 02:44:35.004974500 TRACE ColorControl::updateAvailableChanged: status 0 - version "" offline version "" 
2016-10-10 02:44:51.512847500 TRACE ColorControl::updateAvailableChanged: status 1 - version "" offline version "" 
2016-10-10 02:44:51.512847500 TRACE ColorControl::updateAvailableChanged: status 1 - version "" offline version "" 
2016-10-10 02:48:10.905212500 TRACE ColorControl::updateAvailableChanged: status -1 - version "" offline version "" 
2016-10-10 02:48:11.620574500 TRACE ColorControl::updateAvailableChanged: status -1 - version "" offline version "" 
[... cut ...]
2016-10-10 04:22:27.612121500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-10 06:22:42.578216500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-10 06:22:42.578247500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
2016-10-10 08:22:40.501464500 TRACE connman/cmservice.cpp Property changed "/net/connman/service/ethernet_847e405f9b43_cable" 
2016-10-10 08:22:40.501495500 TRACE connman/cmservice.cpp Update Property "Timeservers" "" 
```

mpvader, 2016-10-11:
@mansr: any thoughts? Could this be related to the Connman update? Since there are several eth0 related messages and errors.

mansr, 2016-10-11:
I doubt it. Something deeper has got stuck, and the network is just one casualty.

jhofstee, 2016-10-11:
I have a hard time making any sense of this. For completeness I have never seen this in the 3 years.

notice this line:

Oct  9 02:21:41 CCGX user.warn kernel: [   45.007720] [sched_delayed] sched: RT throttling activated

Which seemed to indicate something inside linux takes too long. The only things I am aware of we do scheduling related in userland is the watchdog (perhaps we shouldn't do that, so it gets more likely a system under very high loads causes a wd-reset?):

-------
root@ccgx:/log# cat /etc/watchdog.conf 
watchdog-device = /dev/watchdog

# This greatly decreases the chance that watchdog won't be scheduled before
# your machine is really loaded
realtime		= yes
priority		= 1

-------

/service/gui/run
...  nice -n-15 /opt/color-control/gui/gui -qws -nomouse

Which only intend is that if there are multiple process in a ready state, the gui is picked first, so animation run a bit more smoothly.

I fail to understand how we can make even the tcp stack starve from userland. Even a ping no longer works..... I will add some more info in a next reply.

jhofstee, 2016-10-11:
"Groningen hub1" experienced the same behavior when _downgraden_ to an image including the older conmann. Localsettings was up as well. It is not completely stuck, although tcp stack / button are no longer working. Console was disable at the time, so there was no way to get any 
more information.

What I will do is have a look at boot order. IIRC the ccgx starts the gui first, there after networking. The bpp3 would first start networking and gui in the end. I have no logical explanation for why that would matter at the moment, it just one of the only things I can think of which affects startup.

mpvader, 2016-10-12:
an update:

### facts:
- pressing a button doesn't have any affect
- speed is slow: in the logs of various processes you can see that starting up a service which normally takes a second now takes much longer (minutes)
- network doesn't work
- it happened on two different units (Groningen Hub-1 and Groningen Hub-2)
- both times this was after a soft restart, initiated from check-updates.sh
- gui is running and updating the lcd (you can see the time changing)
- first log entry that is different in from a normal boot is `Oct  9 02:21:41 CCGX user.warn kernel: [   45.007720] [sched_delayed] sched: RT throttling activated`. All entries above that are normal, also in their timing.

user space shouldn't be able to cause this, unless there is a kernel bug

### candidates / recent changes
- ~~connman update~~ 2nd incident, Groningen Hub1, was when booting an image with the old connman
- swupdate
  - boot through u-boot instead of falcon mode
  - (any other changes in boot process because of swupdate?)
- change from bpp3 to ccgx machine
  - meta
  - boot order is different:
    - On the bpp3 machine, startapp.sh started daemontools. And startap was on rc5 S95. See [here](https://git.victronenergy.com/ccgx/meta-ccgx/blob/master/recipes-bpp3/recipes-core/initscripts_1.1.bb#L128).
    - On the ccgx machine there is no startapp.sh [see gl issue #8](https://git.victronenergy.com/mirrors/venus/issues/8). Instead, daemontools is started from [daemontools-run-venus.bb](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-third-party/recipes-core/daemontools-run/daemontools-run-venus.bb), at rc5 S30. Which is before connman, php, networking, cron, webservice.

### boot order on v1.80~14 ccgx machine
```
root@ccgx:/etc/rc5.d# ls -l
S02dbus-1 -> ../init.d/dbus-1
S20syslog -> ../init.d/syslog
S21application -> ../init.d/application
S30svscanboot.sh -> ../init.d/svscanboot.sh
S60hostname.sh -> ../init.d/hostname.sh
S60php-fpm -> ../init.d/php-fpm
S61networking -> ../init.d/networking
S70connman -> ../init.d/connman
S70hiawatha -> ../init.d/hiawatha
S70swupdate -> ../init.d/swupdate
S90crond -> ../init.d/crond
S90scan-versions.sh -> ../init.d/scan-versions.sh
S99check-updates.sh -> ../init.d/check-updates.sh
S99simple-upnpd -> ../init.d/simple-upnpd
S99stop-bootlogd -> ../init.d/stop-bootlogd
```

Let me know if I missed anything, so I'll update above list.

jhofstee, 2016-10-13:
Starting daemontools late workarounds this issue / makes it work like it used to do. With daemontools starting early a test setup got into this state at the 3th boot, when starting late it makes it till 73, but that is a different issue, https://github.com/victronenergy/venus/issues/70. Patch is here, https://github.com/victronenergy/meta-victronenergy/commit/91cc83aa48ea941ca1adb117a6c00b0f4d402297

Closing for now, since it is more interesting to check if a newer kernel behaves the same / have a look at a systemd boot.

mpvader, 2017-11-22:
Report from a partner company that is building their own application on top of Venus about a similar issue:

> We're using a AM3517 (CCGX)  as a CANopen NMT master using SocketCAN and are sending a CANopen heartbeat message every 100ms and get replies from various other devices.
via CAN. The CANbus is configured for 250 kbit/s. This causes the AM3517 with the ti-hecc driver ( linux 3.7.1 ) to randomly:
> 
> - generate high soft-IRQ load
> - massive delays on the entire linux network stack that also affects the IP network traffic
> - kernel timeouts on other driver IRQs like ethernet f.e.
> 
> During normal operation idle is about 67 - 88%, but randomly the soft-IRQ starts consuming up to 99% of the CPU. The problem only happens randomly when the CAN interface is used (Sometimes after a few seconds, sometimes after a couple of hours) without using the CAN interface the system runs stable for many days.

Next step they’ll take  in fixing that is trying out the new kernel for ccgx (4.9).

There issue might or might not be related to this one. From our side we’ll do nothing, and await their test results with the new kernel.

I briefly tried reversing the boot order again, on Groningen Hub-1 which is still connected in a canbus system. And rebooted it a few times. All was fine. I’ve restored the boot order afterwards.

jhofstee, 2017-11-25:
turns out we have problems ourselves with the CAN-bus on 4.9, but it is not random and quite reproducible. Seems fixed with https://github.com/victronenergy/linux/commit/e308b68b599888d0010c99e97fc1bf0da16108c7, so they might want to check again. Can be build with https://github.com/victronenergy/meta-victronenergy/tree/jeroen/linux-ccgx-4.9 for now...

mpvader, 2017-12-13:
A ccgx with linux 4.9 build with above issues fixed is available [here as a dropbox download](https://www.dropbox.com/s/c4xexlyjuvd5cu6/venus-swu-ccgx-20171211163318-v2.12_20-linux4.9.swu?dl=1).

I'll ask the german partner company to test this with their software.

closed 2018-07-05 by jhofstee

mpvader, 2018-07-05:
We never heard back from the german partner company