victronenergy/venus #64: improve overviews

mpvader, 2016-10-11:
The different things that can be improved are:

## General:
- during grid failures it would be great to gray out the active-in field and corresponding lines. And, for grid-parallel, also gray out the unprotected AC-load box.
- the multi has 8 leds, not 6. (MultiCompact only has 3)
- make the multi leds blink? led status is already available on the d-bus.

## No PV:
[image]

## Hub-1
- is not showing ac-coupled PV
[image]

## Hub-2
- is not showing dc-coupled PV:
[image]

## Hub-4
- only shows PV on ac input: DC coupled PV and PV on AC output are missing. also systems with only dc-coupled pv are using the hub-4 assistant now.
- is no indication of loads on the output: some people will have those, others don't.
- there will also be hub-4 like systems without pv at all
[image]

mpvader, 2016-10-11:
The gui has several diagrams, and on the diagram certain blocks are shown/hidden based on the a config:
- dc system is shown/hidden based on the user setting
- pv blocks are shown/hidden based on their presence

The diagram selected based on com.victronenergy.system/Hub, determination is done in [delegates.py](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates.py#L84):
```
		hub = None
		vebus_path = newvalues.get('/VebusService')
		if self._dbusmonitor.get_value(vebus_path, '/Hub4/AcPowerSetpoint') != None:
			hub = 4
		elif self._dbusmonitor.get_value(vebus_path, '/Hub1/ChargeVoltage') != None or \
			newvalues.get('/Dc/Pv/Power') != None:
			hub = 1
		elif newvalues.get('/Ac/PvOnOutput/Total/Power') != None:
			hub = 2
		elif newvalues.get('/Ac/PvOnGrid/Total/Power') != None or \
			newvalues.get('/Ac/PvOnGenset/Total/Power') != None:
			hub = 3
		newvalues['/Hub'] = hub
```

Note that that also has a hub = 3, which might need to be removed. Depends on how many different diagrams we'll be using in the new implementation

mpvader, 2016-11-08:
@jepefe: on v2.00~2, there seems to be something wrong with the LED aligment:

[image]

Matthijs Thuis / 1495.

(really cool by the way, a pleasant surprise after updating my bbb!)

mpvader, 2016-11-08:
And this is really cool too:

[image]

mpvader, 2016-11-08:
@jepefe: also I see this error in the gui log (gui v3.02)

```
2016-11-08 19:48:31.308150500 
2016-11-08 19:48:31.308153500 QVNCServer No password configured, accepting connection
2016-11-08 19:48:34.200090500 file:///opt/color-control/gui/qml/OverviewSolarCharger.qml:47:2: QML Image: Failed to get image from provider: image://theme/overview-victron-logo-small
2016-11-08 19:48:42.243668500 QVNCServer disconnected from "127.0.0.1" 
```

jepefe, 2016-11-08:
@mpvader Theme on this build is outdated so the new "live" leds are drawn over the old icon.
Just added the themes repository to the todo so @jhofstee can include it on the next build.

closed 2016-12-02 by mpvader