victronenergy/venus #69: adding cellular support into Venus

mpvader, 2016-10-13:
Many Victron installs are remote, so to get internet they need cellular uplink to internet, sometimes a  special radio connection, or even satellite.

The 'solution' we offer today is for customers to add an external 3g router; see 1.4.3 in the [ccgx manual](https://www.victronenergy.com/live/ccgx:start#internet_connectivity).

I have been discussing the various options around integrating this with several different people in the last few months. And, even though we might not start development on this any time soon, the purpose of this issue is to hold the info together.

### 2G vs 3G
Probably 3G only:
- [In Australia the 2G networks are going to be switched off soon](http://whatphone.com.au/guide/how-the-2g-mobile-shutdown-will-affect-users-and-the-best-3g-options/)
- In installs where the last 20 or 30 USD is no issue, some customers will want 3G because of Remote Console for example

### USB Modem vs. mini PCIe vs. other module type options
Info from Hal:
```
For example we had some Huawei USB modems we were using. The problem is
they run their own OS, so they can appear as a cdc-acm Ethernet device (and a
USB storage device to Windows to install drivers).

The bad part is: they have a retry limit defined in their OS, and if the cell network
is down and the retry limit is exceeded, you have to power cycle them before they
will connect again. Very unfortunate behaviour for Africa and unattended installations.

If you used a mini PCIe module that appears as a USB serial device and is a bit more
"dumb" you can control the reconnection inside Linux and you don't have such a
bad time.
```

Disadvantage of mini PCIe and other modules will probably be their cost. And power cycling the USB device might be possible, depends on schematics of the BBE, and CCGX: needs looking into.

Another alternative (thanks Hal!)
```
You can get mini PCI-e to USB adapters, designed specifically for turning an internal 3G modem into an external modem.

https://m.aliexpress.com/item/32657444139.html

They are quite cheap, as they as only passive devices. We have done this with
some internal modems and the combination usually works much better than
the "retail" USB 3G modems.

The cost is only around $40USD for the modem ($35) + adapter ($5). This is
only a bit more expensive than a retail USB 3G modem.

The only issue is that you would need a case to enclose them.
```

### Connman / Ofono etc.
to be added

halmartin, 2016-10-13:
`In installs where the last 20 or 30 USD is no issue, some customers will want 3G because of Remote Console for example`

I wouldn't advocate installing a 2G modem. They're not easy to buy anymore, everyone is selling 3/4G modems, and they are backwards compatible with 2G networks anyway.

Customers should perform a cursory survey of their installation site with a smartphone before they purchase a modem and SIM card for Venus communications. In many rural regions, even in some developed countries, only 2G service is available.

Remote console can be used over 2G (we do it in Africa) but it's painful, and I would not recommend it!

mpvader, 2017-07-06:
Test results of the EF555DA from Yifan Industry Co. Ltd:

> Plugging one into a USB
> port on a PC, it shows up as a CDROM device containing a Windows driver
> and a PDF with instructions for Linux.  These instructions don't seem to
> match the actual hardware, although a quick Google search and a peek at
> the Linux drivers suggests it shouldn't be too hard to get working.
> 
> However, plugging the dongle into the BBE only results in an error
> message:
> 
> musb-hdrc musb-hdrc.1.auto: VBUS_ERROR in a_host (91, <VBusValid), retry #1, port1 00000507
> 
> It seems that this device draws more power than it should or is
> otherwise misbehaving.  One of my USB hubs also disliked it to the point
> of resetting itself.  With a powered hub connected to the BBE the dongle
> is successfully detected, but I don't think that's a requirement we want.

halmartin, 2017-07-07:
> It seems that this device draws more power than it should or is otherwise misbehaving. One of my USB hubs also disliked it to the point of resetting itself. With a powered hub connected to the BBE the dongle is successfully detected, but I don't think that's a requirement we want.

Sorry, I should have mentioned earlier. Yes, these devices consume a lot of power. It's not uncommon to see them consume up to 1A while transmitting.

If the BBE cannot reliably provide this much power to a single USB device, then I don't think you're going to have a fun time with any USB 3G modem.

mansr, 2018-02-28:
An externally powered modem based on the Simcom SIM5360E module will be used. This device presents as multiple tty devices in Linux: two functionally identical AT command / data interfaces and one for GPS NMEA messages.

In Venus, one AT interface is managed by pppd to provide the data connection. The other is used by the dbus-modem application to retrieve status information and publish on the system dbus. The following read-only values are exported under the com.victronenergy.modem service.

Path | Description
-----|-------------
/Model | modem model
/IMEI | International Mobile Equipment Identity
/NetworkName | name of registered mobile network
/NetworkType | type of mobile network (GSM, UMTS, ...)
/SignalStrength | signal strength (0-31)
/Roaming | currently roaming (0/1)
/Connected | data link active (0/1)
/IP | IP address (when connected)

The following localsettings values are used. These are monitored and changes acted upon.

Setting | Description
--------|------------
/Settings/Modem/Connect | establish data connection (0/1)
/Settings/Modem/RoamingPermitted | connect when roaming (0/1)

When the data connection is active, it is configured with a high routing metric. This way, the Linux kernel prioritises Ethernet or Wifi when these are available. A dnsmasq proxy forwards DNS lookups to the correct name server on Ethernet/Wifi or mobile data.

mpvader, 2018-04-03:
After messing around a little, or just after doing a reboot, udev still ends up starting the modem script on other ports:

```
# svstat /service/dbus-modem*
/service/dbus-modem.ttyUSB2: down 364 seconds
/service/dbus-modem.ttyUSB3: up (pid 6401) 1 seconds, normally down
/service/dbus-modem.ttyUSB4: down 203 seconds
```

Looking at dmesg, it seems to skip the ports that dbus-modem is running on:
```
[  145.648959] usb 2-1.3: new high-speed USB device number 10 using musb-hdrc
[  145.780863] usb 2-1.3: New USB device found, idVendor=05c6, idProduct=9000
[  145.788082] usb 2-1.3: New USB device strings: Mfr=3, Product=2, SerialNumber=0
[  145.795771] usb 2-1.3: Product: SimTech SIM5360
[  145.800523] usb 2-1.3: Manufacturer: SimTech, Incorporated
[  145.809603] usb 2-1.3: no of_node; not parsing pinctrl DT
[  145.825207] option 2-1.3:1.0: no of_node; not parsing pinctrl DT
[  145.831754] option 2-1.3:1.0: GSM modem (1-port) converter detected
[  145.838491] option1 ttyUSB0: no of_node; not parsing pinctrl DT
[  145.846983] usb 2-1.3: GSM modem (1-port) converter now attached to ttyUSB0
[  145.854920] option 2-1.3:1.1: no of_node; not parsing pinctrl DT
[  145.861409] option 2-1.3:1.1: GSM modem (1-port) converter detected
[  145.868144] option1 ttyUSB1: no of_node; not parsing pinctrl DT
[  145.877147] usb 2-1.3: GSM modem (1-port) converter now attached to ttyUSB1
[  145.885101] option 2-1.3:1.2: no of_node; not parsing pinctrl DT
[  145.891582] option 2-1.3:1.2: GSM modem (1-port) converter detected
[  145.898320] option1 ttyUSB6: no of_node; not parsing pinctrl DT
[  145.909411] usb 2-1.3: GSM modem (1-port) converter now attached to ttyUSB6
[  145.917422] option 2-1.3:1.3: no of_node; not parsing pinctrl DT
[  145.923882] option 2-1.3:1.3: GSM modem (1-port) converter detected
[  145.930654] option1 ttyUSB7: no of_node; not parsing pinctrl DT
[  145.939738] usb 2-1.3: GSM modem (1-port) converter now attached to ttyUSB7
[  145.947831] option 2-1.3:1.4: no of_node; not parsing pinctrl DT
[  145.954319] option 2-1.3:1.4: GSM modem (1-port) converter detected
[  145.961067] option1 ttyUSB8: no of_node; not parsing pinctrl DT
[  145.969546] usb 2-1.3: GSM modem (1-port) converter now attached to ttyUSB8

# svstat /service/dbus-mod*
/service/dbus-modbustcp: down 271 seconds
/service/dbus-modem.ttyUSB2: down 221 seconds
/service/dbus-modem.ttyUSB3: down 196 seconds
/service/dbus-modem.ttyUSB4: down 171 seconds
/service/dbus-modem.ttyUSB5: down 146 seconds
/service/dbus-modem.ttyUSB6: up (pid 3141) 134 seconds, normally down

```

Could be that dbus-modem isn't releasing the port quick enough between the disconnect & reconnect, causing udev to skip that tty. I think we've seen this before with vedirect or similar.

mpvader, 2018-04-03:
Besides the serial port being locked~, also pppd is not stopped. I ended up having lots of instances running. poff (which calls killall) fixes it. Possible fix for that here:

https://github.com/victronenergy/dbus-modem/commit/bf1c61b4d34d5c2271ef23187bfa8d9d8d3de1e8

Above serialport not being reused issue might relate to ser.close() never being called. But I haven't been able to pin that down yet. Also, it feels like these clean-up things might be better being taken care of by udev rather than (only?) the script.

mpvader, 2018-05-18:
support sofar is in Venus since v2.20~something. Left-over todos:

- look into using the watchdog to reset the simcom in case there is an issue
- add modem to gui
- code review of dbus-modem

And then, probably at a later stage: add sending / receiving text messages.

mpvader, 2018-05-22:
for some unknown reason, the gps wasn't enabled on my modem:

```
AT+CGPS?
0,1   (the 0 means that its disabled, should have been a 1)
```

dbus-modem, that enables this during init, was running OK.

I power cycled the modem, and at the same time rebooted the venus device, and then all worked OK. Discussed action taken is to add error logging to dbus-modem, and thereafter look further.

jepefe, 2018-06-12:
I made some progress on the modem GUI.
  - Add modem settings page
  - Add network type and signal strength to the status bar
  - Add sim locked icon to the status bar
  - To make the status bar icons behave same as smartphones I also added Wi-Fi status icon:
    - Only modem: Show signal strength + network type
    - Modem & Wi-Fi(Wi-Fi has priority): Show modem & Wi-Fi signal strength, hide network type. 

I miss APN settings in dbus-modem, I need to set APN name to "internet" to be able to use my Lebara NL sim card in Spain (roaming).

Some screenshots:

Modem only:
[image]

Wi-Fi + modem:
[image]

Modem connected, sim locked(PIN):
[image]

[image]

[image]

[image]

jepefe, 2018-06-13:
I made some changes to the gui:
  * Hide network type (2G, 3G...) on the status bar when not connected (a
  *  Show an "R" on the status bar when roaming
  * Add APN name setting on the settings page

[image]
[image]
[image]
[image]

imval, 2019-02-11:
I performed several tests this weekend during a trip to France. CCGX with v2.22 and USB modem, Vodafone NL sim.

- On the way to France, roaming enabled. Successful connection to foreign networks in Belgium and France. Good network stability.
- While in France, disabling the roaming disconnected the internet connection. Updated to latest release candidate 2.30~27 at home before driving back to Netherlands.
- On the way back to the Netherlands, roaming disabled. CCGX stayed disconnected in France and Belgium, connected correctly when arriving in the Netherlands.
- When loosing signal in home country, it reconnected correctly to the right network after reinstalling the antenna.

mpvader, 2019-02-11:
Thanks Valentin; closing this issue now.

closed 2019-02-11 by mpvader