victronenergy/venus #70: linux fails to start

jhofstee, 2016-10-13:
A test setup, continuously updating with swupdate, failed the 73 time, details below 

```
U-Boot SPL 2013.01.01 (Oct 11 2016 - 15:20:59)

U-Boot 2013.01.01 (Oct 11 2016 - 15:20:59)

AM35XX-GP ES2.0, CPU-OPP2, L3-165MHz, Max CPU Clock 600 Mhz
BPP3 + LPDDR/NAND
I2C:   ready
DRAM:  256 MiB
NAND:  512 MiB
MMC:   OMAP SD/MMC: 0
In:    serial
Out:   serial
Err:   serial
Die ID #242e000100000000016b3a8b0e00c01c
Net:   DaVinci-EMAC

NAND read: device 0 offset 0x340000, size 0x7f800
 522240 bytes read: OK
gpio: pin 26 (gpio 26) value is 1

gpio: pin 29 (gpio 29) value is 1

NAND read: device 0 offset 0x400000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 80300000 ...
   Image Name:   Linux-3.7.1
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2850992 Bytes = 2.7 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.7.1 (jeroen@jeroen-VirtualBox) (gcc version 4.7.2 (GCC) ) #1 Mon Oct 10 16:48:39 CEST 2016
[    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine: Victron BPP3
[    0.000000] Reserving 2097152 bytes SDRAM for VRAM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] AM3517 ES1.1 (l2cache sgx neon )
[    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 64256
[    0.000000] Kernel command line: root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 console=ttyO0,115200n8 omapdss.def_disp=lcd omapfb.vram=0:2M vram=2M mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),496m(ubisystem)
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253MB = 253MB total
[    0.000000] Memory: 246112k/246112k available, 16032k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc04df310   (4957 kB)
[    0.000000]       .init : 0xc04e0000 - 0xc050caf8   ( 179 kB)
[    0.000000]       .data : 0xc050e000 - 0xc05513a0   ( 269 kB)
[    0.000000]        .bss : 0xc05513c4 - 0xc0a67f2c   (5211 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 4.0) with 96 interrupts
[    0.000000] Total of 96 interrupts on 1 active controller
[    0.000000] OMAP clockevent source: GPTIMER1 at 32768 Hz
[... 376 lines cut ...]
Out:   serial
Err:   serial
Die ID #242e000100000000016b3a8b0e00c01c
Net:   DaVinci-EMAC

NAND read: device 0 offset 0x340000, size 0x7f800
 522240 bytes read: OK
gpio: pin 26 (gpio 26) value is 1

gpio: pin 29 (gpio 29) value is 1

NAND read: device 0 offset 0xa00000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 80300000 ...
   Image Name:   Linux-3.7.1
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2850992 Bytes = 2.7 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...
```

jhofstee, 2016-10-14:
I pushed the test script here, so others can reproduce this https://github.com/victronenergy/meta-victronenergy/commit/3114d2d25d051dd9071436815fbc43dc81954a21

@mansr Can you have a look at this?

jhofstee, 2016-10-18:
For completeness: when the device is restarted after above issue occurs, it boots normally. So a watchdog can catch this situation. From the logs "Verifying Checksum ... OK", so it is not a nand read issue.

I made an image with the 4.4 kernel on it (slightly different test, since it only reboots, doesn't update). It has been rebooting, on the same device as the issue occurred, since yesterday afternoon and still working. (that's more than 1000 reboots or so). 

So likely something goes wrong in linux before, "0.524597] omap_uart.0: ttyO0 at MMIO 0x4806a000 (irq = 88) is a OMAP UART0". The logs before that are only printed when the uart has been probed in this configuration.

mansr, 2016-10-18:
I have reproduced the hang with earlyprintk enabled and got this log:
```
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.7.1 (mru@unicorn) (gcc version 5.2.0 (GCC) ) #2 Mon Oct 17 21:43:36 BST 2016
[    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine: Victron BPP3
[    0.000000] bootconsole [earlycon0] enabled
[    0.000000] Reserving 2097152 bytes SDRAM for VRAM
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] AM3517 ES1.1 (l2cache sgx neon )
[    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 64256
[    0.000000] Kernel command line: root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 console=ttyO0,115200 omapdss.def_disp=lcd omapfb.vram=0:2M vram=2M fbskip earlyprintk mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),-(ubisystem)
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253MB = 253MB total
[    0.000000] Memory: 246044k/246044k available, 16100k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc04efd10   (5024 kB)
[    0.000000]       .init : 0xc04f0000 - 0xc051cef8   ( 180 kB)
[    0.000000]       .data : 0xc051e000 - 0xc05615e0   ( 270 kB)
[    0.000000]        .bss : 0xc0561604 - 0xc0a78128   (5211 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 4.0) with 96 interrupts
[    0.000000] Total of 96 interrupts on 1 active controller
[    0.000000] OMAP clockevent source: GPTIMER1 at 32768 Hz
[    0.000000] sched_clock: 32 bits at 32kHz, resolution 30517ns, wraps every 131071999ms
[    0.000000] OMAP clocksource: 32k_counter at 32768 Hz
[    0.000000] Console: colour dummy device 80x30
[    0.000000] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.000000] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.000000] ... MAX_LOCK_DEPTH:          48
[    0.000000] ... MAX_LOCKDEP_KEYS:        8191
[    0.000000] ... CLASSHASH_SIZE:          4096
[    0.000000] ... MAX_LOCKDEP_ENTRIES:     16384
[    0.000000] ... MAX_LOCKDEP_CHAINS:      32768
[    0.000000] ... CHAINHASH_SIZE:          16384
[    0.000000]  memory used by lock dependency info: 3695 kB
[    0.000000]  per task-struct memory footprint: 1152 bytes
[    0.062194] Calibrating delay loop... 397.57 BogoMIPS (lpj=1554432)
[    0.129669] pid_max: default: 32768 minimum: 301
[    0.134948] Mount-cache hash table entries: 512
[    0.143951] CPU: Testing write buffer coherency: ok
[    0.150054] Setting up static identity map for 0x803a5e38 - 0x803a5e90
[    0.160980] devtmpfs: initialized
[    0.212371] pinctrl core: initialized pinctrl subsystem
[    0.223510] regulator-dummy: no parameters
[    0.229675] NET: Registered protocol family 16
[    0.235809] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.244903] omap-gpmc omap-gpmc: GPMC revision 5.0
[    0.260009] OMAP GPIO hardware version 2.5
[    0.279754] omap_mux_init: Add partition: #1: core, flags: 0
[    0.308197] Found NAND on CS0
[    0.311370] Registering NAND on CS0
[    0.317016] _omap_mux_init_gpio: Multiple gpio paths (3) for gpio126
[    0.325805] _omap_mux_init_gpio: Multiple gpio paths (2) for gpio99
[    0.332397] _omap_mux_init_gpio: Multiple gpio paths (2) for gpio104
[    0.343017] hw-breakpoint: debug architecture 0x4 unsupported.
[    0.358520] OMAP DMA hardware revision 4.0
[    0.366790]  arm-pmu: alias fck already exists
[    0.415527] bio: create slab <bio-0> at 0
[    0.497863] omap-dma-engine omap-dma-engine: OMAP DMA engine driver
[... 45 lines cut ...]
[    0.866302] Process kworker/0:1 (pid: 30, stack limit = 0xcf8c2230)
[    0.872863] Stack: (0xcf8c3e78 to 0xcf8c4000)
[    0.877410] 3e60:                                                       c01c15f8 c0a6ba64
[    0.885925] 3e80: cf867640 c0c7e400 cf8c3ec0 c053d218 cf8c2020 00000000 00000002 c00451d0
[    0.894439] 3ea0: 00000002 00000000 c004515c 00000080 00000000 c01c15f8 00000000 c053ff50
[    0.902954] 3ec0: c0a6ca94 c06212fc 00000000 c0488548 cf860980 c053d340 cf867650 cf8c2038
[    0.911468] 3ee0: c053d348 c10c9715 c053d528 c0574f4c cf867640 c00468b8 c053d218 c0561544
[    0.919982] 3f00: 60000153 cf841e48 60000153 cf841e34 00000000 cf867640 c00466a4 00000000
[    0.928466] 3f20: 00000000 00000000 00000000 c004aabc 00000000 00000000 cf8c3f9c cf867640
[    0.936981] 3f40: 00000000 00000001 dead4ead ffffffff ffffffff c05752c4 00000000 00000000
[    0.945495] 3f60: c046a70c cf8c3f64 cf8c3f64 00000000 00000001 dead4ead ffffffff ffffffff
[... cut ...]
[    1.007598] [<c00468b8>] (worker_thread+0x214/0x5b4) from [<c004aabc>] (kthread+0xb4/0xb8)
[    1.016204] [<c004aabc>] (kthread+0xb4/0xb8) from [<c000e270>] (ret_from_fork+0x14/0x24)
[    1.024627] Code: e3a01000 e5970030 ebffffb2 e597303c (e5930028) 
[    1.031127] ---[ end trace fbfff8460398e860 ]---
```
After this, all hell breaks loose.

The crash occurs in the display driver. The code where it happens is later changed by commit torvalds/linux@595470a7853848cb971d5ee3fed443b1e3aa0d1b though without mention of a crash. There are also hundreds of other changes to this driver.

jhofstee, 2016-10-18:
On 18-10-16 13:14, Mans Rullgard wrote:
>
> I have reproduced the hang with earlyprintk enabled and got this log:
> ....
>
great!

||
> ||
>
> After this, all hell breaks loose.
>
> The crash occurs in the display driver. The code where it happens is 
> later changed by commit torvalds/linux@595470a 
> <https://github.com/torvalds/linux/commit/595470a7853848cb971d5ee3fed443b1e3aa0d1b> 
> though without mention of a crash. There are also hundreds of other 
> changes to this driver.
>

If cherry-picking that patch prevents the crash, that seems
good enough to me. We are looking into a 4.4 / 4.9 kernel,
so we get the other hundreds of changes as well, instead of
backporting them all.

Happen to know how this can relate to falcon mode / u-boot boot?

mansr, 2016-10-18:
I'm running the test again with that patch included. We'll see if it helps in a few hours.

U-boot things with the display controller that SPL doesn't. If that leaves it in a different state when starting the kernel is hard to tell.

mansr, 2016-10-18:
Good news and bad. Backporting that patch makes it not crash. The bad news is that the display is occasionally stuck showing solid blue along with some FIFO UNDERFLOW errors in the kernel log.

jhofstee, 2016-10-19:
mmm, perhaps we should revert https://github.com/victronenergy/linux/commit/b2c64b6b936b6f50b3e947a82ca88f2176a47b1c and check if that makes things any better?

mansr, 2016-10-19:
Testing now with the DSS parts of that patch reverted. It has the somewhat ugly side effect of the display going all white for about a second during boot.

mansr, 2016-10-19:
Still running without issues.

jhofstee, 2016-10-19:
better, so we got a new problem, the white flash.

The fb location of u-boot and linux is the same (RAM end - 2MB) [1]. Linux
video ram is passed as a boot arg. We keep the framebuffer contents by
setting fbskip. So am tempted to think we can reset the dss, without 
distorting
the fb contents. (but you have it in front of you so it is easy for you 
to see).

I guess that the white screen comes from the lcd_enable gpio's, so what we
could try is not resetting the gpio's, but do reset the dss... or 
backlight off,
reset dss, backlight on if that causes display artifacts. These are the 
gpio's
involved (from u-boot):

# define BACKLIGHT_OFF_GPIO        42
# define BACKLIGHT_ON_GPIO        53
# define DSP_POWER_GPIO            139

Note that ON is the backlight level, and changed to a PWM somewhere along
the linux boot. Any better idea's? Can you have a look otherwise what 
happens
if we keep driving the gpio pins, but do reset the dss?

[1] 
https://github.com/victronenergy/u-boot/blob/color_control/board/victron/bpp3/bpp3.c#L245
[2] 
https://github.com/victronenergy/linux/commit/ace818c7fbec33afff8b24b9914cef9b312fea4b

jhofstee, 2016-10-19:
ah damn, I just noticed you wrote "Testing now with the DSS parts of that patch reverted". So that only leaves the "Any better idea's?" for now...

jhofstee, 2016-10-19:
for what it is worth (likely nothing), the 4.4 can reset the dss without any annoying distorting and then displays the penguin because I haven't killed it yet...

mansr, 2016-10-19:
If the 4.4 kernel works without glitches, I'd be tempted to live with the white flash until we move to a new kernel anyway. Figuring out which of the hundreds of commits is/are responsible and backporting them doesn't seem worth the effort.

For what it's worth, it the test still hasn't failed after more than 9 hours of continuous reboots.

mpvader, 2016-10-19:
At a minimum we should use this fix / white flash to go back to where we were: increase number of people that test v1.80 by announcing it to the release-candidate group of people.

Besides swupdate, v1.80 is full of other changes: testing / bug fixing / improving will take a while.

I'd have to see it myself to know if we can also leave the white flash in place for the real release.

mansr, 2016-10-20:
Kernel patches pushed to swupdate branch of meta-victronenergy.

jhofstee, 2016-10-21:
it is in develop now. Regarding the 4.4 linux version, it has the white splash as well, but for a short period, the dispc is up within a second, see https://github.com/jhofstee/linux/wiki/bootlog-4.4

mpvader, 2016-11-08:
Fixed in https://github.com/victronenergy/meta-victronenergy/commit/8ee30d2f305f486670807f5319e39839a48d18f5

White splash is still there, but we accept to release that as is. No easy fix, and a there is work being done on a newer kernel already anyway.

closed 2016-11-08 by mpvader