victronenergy/venus #71: make sure bbb gets not stuck in u-boot

mpvader, 2016-10-17:
https://github.com/victronenergy/u-boot/blob/master/doc/README.autoboot

mansr, 2016-12-06:
Why not check the user button similarly to how it's done on ccgx?

mansr, 2016-12-07:
It might be hard to reach with a cape in place.

mpvader, 2016-12-14:
work is done by @mansr 

https://github.com/victronenergy/meta-victronenergy/tree/bootretry

mpvader, 2016-12-22:
https://github.com/victronenergy/u-boot/commit/40e61ddc78d42bd869fa740896ff722b4cfdb757

part of https://github.com/victronenergy/u-boot/releases/tag/v2015.07-bbb-v4

mpvader, 2016-12-22:
@jhofstee: I'll leave this for you to close once included in master

mpvader, 2016-12-22:
Note that the change made does not fix the case that a stray character comes in every second. Which is no problem, what is done now is an easy fix to catches most of the situations: good enough.

jhofstee, 2016-12-27:
https://github.com/victronenergy/meta-victronenergy/commit/c8696f1d76a26f6d41c5e4bfc9c18821e0086c67

closed 2016-12-27 by jhofstee