victronenergy/venus #72: make 4 LEDs blink in fixed pattern when in u-boot

mpvader, 2016-10-17:
The goal is to easily recognize that a unit is in bootloader modus.

Besides the four default user LEDs on the bbb and bbe boards themselves, there is also one bicolor LED on the cape: LD101 on page 1 of the [vecape rev 4 schematic](https://github.com/victronenergy/bbb-hardware/blob/master/VE_BBB_Cape_v0.4_released.pdf)

mpvader, 2016-12-07:
By the way, when all mounted together with the cape and in the enclosure, those four LEDs are no longer visible. The red/green on the cape LED is though. See LD101 on the [v0.61 schematic](https://github.com/victronenergy/bbb-hardware/blob/master/VE_BBB_Cape_v0.61_released.pdf).

mpvader, 2016-12-22:
Done: https://github.com/victronenergy/u-boot/commit/40e61ddc78d42bd869fa740896ff722b4cfdb757

Part of https://github.com/victronenergy/u-boot/releases/tag/v2015.07-bbb-v4

mansr, 2016-12-22:
LED indications:
- USR3 / cape red: u-boot started
- USR2 / cape green: booting kernel

jhofstee, 2016-12-27:
ok included, slightly rewrite (recipes was renamed)
https://github.com/victronenergy/meta-victronenergy/commit/c8696f1d76a26f6d41c5e4bfc9c18821e0086c67

and with a fix for gpio number...
https://github.com/victronenergy/meta-victronenergy/commit/2eef420ab6cf013f7452e6718254b76ae67b5568

(linux will reset the gpio directly though, so they are turned almost off again, but lets make that a separate issue, closing this)

closed 2016-12-27 by jhofstee