victronenergy/venus #73: bbb: start watchdog from u-boot (spl/mlo?) instead of from userland

mpvader, 2016-10-18:


mpvader, 2016-10-18:
right now (I assume) the bootloader is only started from userland. And therefore not protecting against hangs while starting Linux.

https://github.com/victronenergy/u-boot/blob/beaglebone/doc/README.watchdog

https://e2e.ti.com/support/arm/sitara_arm/f/791/p/495092/1792799

jhofstee, 2016-12-28:
hello Matthijs,

Starting the watchdog is already done in u-boot. However linux disables it directly again. We can prevent that if we want https://github.com/victronenergy/linux/commit/b3f0c4a6f188a04609a2e561f031c85d07b73826. Assigning to @mansr. 

Mans, any thought on this?

jhofstee, 2017-01-16:
@mansr 

lets back-port these:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/watchdog/omap_wdt.c?id=b2102eb36e7909c779e46f66595fda75aa219f4c
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/watchdog/omap_wdt.c?id=8605fec1ce2a397f738f1c394be7847dacf31b0c

It boils down to same thing, but there is a bootparam to control behavior so we can make sure it is only enabled when booting from onboard mmc and not when updating etc.

jhofstee, 2017-01-20:
included in v2.01~10. You need to opt-in for keeping the watchdog alive during boot, see https://github.com/victronenergy/meta-victronenergy/commit/69f38b3beeb6ba98bbe688dc49a627c45b5d1fc4

closed 2017-01-20 by jhofstee