victronenergy/venus #75: gui: not able to unmount an update image

jhofstee, 2016-10-20:
The upcoming release warns about unclean disk removal and tells you to go to the VRM settings menu. In case of a firmware update image, there is no option there to unmount the disk though.....

mpvader, 2018-07-12:
Solution: add the unmount option to Settings -> General. Which duplicates it, but thats fine.

jepefe, 2018-07-13:
This requires more than gui work. 
Vrmlogger script doesn't support unmount when the inserted sdcard is not storing the log file, for example when it contains a Venus firmware file.

Shall I take a look to it and fix it?

Another option is to don't warn when improperly removed if sdcard contains an update image as no one is writing to it (I'm not 100% sure).

petajoulecorp, 2019-11-22:
> Which duplicates it, but thats fine.

@mpvader Most of the time, it's not. It's a frequent topic in UX design decisions.

https://www.nngroup.com/articles/duplicate-links/

> What’s the big deal about having a few duplicate links on the page?
> 
> Each additional link increases the interaction cost required to process the link because it rises the number of choices people must process. The fewer the choices, the faster the processing time.
> 
> Each additional link depletes users’ attention because it competes with all others. Users only have so much attention to give and often don’t see stuff that’s right on the screen. So when you grab more attention for one link, you lose it for the others: there’s substantial opportunity cost to extra linking.
> 
> Each additional link places an extra load on users’ working memory because it causes people to have to remember whether they have seen the link before or it is a new link. Are the two links the same or different? Users often wonder if there is a difference that they missed. In usability studies, we often observe participants pause and ponder which they should click. The more courageous users click on both links only to be disappointed when they discover that the links lead to the same page. Repetitive links often set user up to fail.
> 
> Extra links waste users’ time whenever users don’t realize that two links lead to the same place: if they click both links, then the second click is wasteful at best. At worst, users also don’t recognize that they’ve already visited the destination page, causing them to waste even more time on a second visit to that page. (Remember that to you, the distinctions between the different pages on your site are obvious. Not so for users: we often see people visit the same page a second time without realizing that they’ve already been there.)
> 
> ​Only show what’s needed. Nothing more.

jhofstee, 2019-11-23:
@petajoulecorp that is about repeating links in a html file. This about some else. Can you please stop polluting issues..

mpvader, 2019-11-25:
on hold; other things have priority.

closed 2021-07-21 by jhofstee