victronenergy/venus #80: Conman and friends using more CPU power than expected

edevries-victron, 2016-11-07:
Sometimes the connmand daemon and related services take far more CPU power than expected. See the top-output below. 
According to /log/messages conman is permanently adding and removed routes to/from eth0. There is also a lot of DHCP activity ( /bin/sh /etc/udhcpc.d/50default renew). Note that the gui logger (multilog t s99999 n8 /log/gui) is also very busy, because the gui logs connman events.

[log_messages.txt](https://github.com/victronenergy/venus/files/575556/log_messages.txt)
[top_output.txt](https://github.com/victronenergy/venus/files/575557/top_output.txt)

mpvader, 2016-11-07:
when using connman, udhcpc should not be installed / used. I already have a half finished patch for this for the beaglebone / Jethro. I'll finish that.

For @jhofstee it might interesting to find out how this suddenly came up in Danny. We already confirmed that it is new, in v1.73 there is no udhcpc in the process list.

jhofstee, 2016-11-08:
for the ccgx this was already fixed. A default /etc/network/interfaces was accidently shipped which enabled dhpc for eth0, which it shouldn't do, since connman handles the dhcp request. It should be fine again in v2.00~2.

mpvader, 2016-11-08:
thanks @jhofstee, that finally solved the puzzle for me: after figuring out that udhcpc was conflicting with connman on the beaglebone a while back, I tried removing udhcp from the busybox packages. Which can obviously be done but then I ended up in ifup or ifdown errors since it couldn't load udhcpc anymore.

I had never thought of looking for `dhcp` in /etc/network/interfaces. I did now, see the result here:

https://github.com/victronenergy/meta-victronenergy/commits/fix-udhcpc-connman

I did two things there:

1. Fix the issue on the beaglebone
2. Fix the issue for other boards running Jethro. Which, unlike the beaglebone machine, don't perse need their own variant of /etc/network/interfaces. An example of this will be the rpi2. I already spoke to @izak and he confirmed he had this same issue.

@izak, can you recompile your image with this branch and confirm for us that /etc/network/interfaces now specifies `manual` instead of `dhcp`?

@jhofstee pending Izak his confirmation, I'll already add the beaglebone commit to your todo list.

izak, 2016-11-09:
The patch works correctly on the Raspberry Pi.

mpvader, 2016-11-09:
ok I'll close the issue then, and add this on the todolist to put in master.

closed 2016-11-09 by mpvader