victronenergy/venus #81: remove dbus-redflow

mpvader, 2016-11-10:
## Background
The new way of connecting is that the installation as a Redflow product called the Z-Cell BMS. A RaspberryPi. Which communicates its SOC over ModbusTCP into our VE.Bus SOC field. Which is all it does, and which works beautifully.

The Z-Cell BMS is about to support the canbus-bms protocol, and besides that it might also be possible to communicatie those same BMS parameters over modbustcp.

## Action
All existing installations have been converted to using the Z-Cell BMS. So time to clean up and take this code out.

mpvader, 2016-12-06:
- [ ] remove the service
- [ ] remove it from serial starter
- [ ] remove it from the gui code
- [ ] remove it from documentation

edevries-victron, 2016-12-08:
We should also remove all references to the com.victronenergy.zbmnode (created by dbus-redflow) service from VRM logger.
BTW. This service is not supported by modbus TCP.

mpvader, 2017-03-28:
@edevries-victron: fyi this is on hold. Its still actively used in South Africa.

jhofstee, 2019-02-27:
can this be removed by now or should it be closed?

mpvader, 2019-06-22:
Yes it can be removed now.

mpvader, 2019-11-28:
I checked the CCGX manual as well as battery compatibility pages: its no longer mentioned; which ticks the last box. closing.

closed 2019-11-28 by mpvader