victronenergy/venus #83: ssl error in vrmpubnub

mpvader, 2016-11-22:
```
DEBUG:__main__:livefeed channel initialized. Name: b0d5ccfd7052_77ef18e8249314a42464d90c88454bb9, Cipher: 3b9ecefd681bfbf0c3beaca5671ac8c9
DEBUG:vrmhttp:[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)
DEBUG:__main__:Sending auth-packages to vrm failed, retry on next timer tick
DEBUG:__main__:Sending to reader: {"pID": "b0d5ccfd7052", "ts": 1479833155, "opResponse": ["announce", {"msgnr": 0, "commandid": "None", "ttm": 2, "ttl": 300}]}
INFO:__main__:Starting mainloop, responding on only events
INFO:__main__:Send to pubnub :[1, u'Sent', u'14798331549232466']
DEBUG:vrmhttp:[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)
DEBUG:__main__:Sending auth-packages to vrm failed, retry on next timer tick
```

Probably there is a certificate missing in the beaglebone build.

Because of this, VRM Two Communication doesn't work. Also known as Remote VEConfigure and Remote Firmware updates, with the VE Power Setup tool.

mpvader, 2016-11-23:
@jhofstee: ccgx-ca.crt was added to openembedded-core/danny-venus here:
https://github.com/victronenergy/openembedded-core/commit/a879eaaa5009c27da6c45e09abc91fb055c448d6

I've assigned this to you.

jhofstee, 2016-12-19:
https://github.com/victronenergy/meta-victronenergy/commit/cd33b11280bf5ae5711f7c161a69c65cc831b47e

closed 2016-12-19 by jhofstee