victronenergy/venus #85: booting into backup rootfs doesn't work

mpvader, 2016-11-23:
Running version v2.00~10, trying to boot back into v2.00~7.

Details emailed to Måns.

Reproducible on the customers unit, a CCGX. Perhaps else where too: I tried one other unit running the same versions but then ran in to a different problem.

root@ccgx:/var/volatile/run# cat versions 
201611211641 v2.00~10
201611141615 v2.00~7

jhofstee, 2016-11-24:
Hi, this is caused by the following two variables:

miscargs=console=ttyO0,115200 root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M vram=2M
nandboot=setenv bootargs $miscargs $mtdparts && nand read ${loadaddr} kernel1 && bootm ${loadaddr}

which should be:
miscargs omapdss.def_disp=lcd omapfb.vram=0:2M vram=2M fbskip
nandboot nand read ${loadaddr} ${kernel} && bootm ${loadaddr}

so we need to figure out were these values are coming from .....

jhofstee, 2016-11-24:
see https://github.com/victronenergy/meta-victronenergy/commit/78335a3833ed99738fe08cff7a7b782f46d8adb8

closed 2016-11-24 by jhofstee