victronenergy/venus #89: add VE.Bus BMS status

vejpasop, 2016-12-01:


mpvader, 2016-12-02:
What logic is applied to the AD value by the VE.Bus devices?

mpvader, 2016-12-06:
@vejpasop: can you answer my question on that logic? FYI I updated your comment, added a small todo list in there.

vejpasop, 2016-12-06:
When bit 7 gets high the "High cell" flag will be set.

vejpasop, 2016-12-08:
Done implementing the  above on branch: 

https://git.victronenergy.com/ve-products/vebus_interface/commits/WIP_VEBUS_BMS_status

@edevries-victron: I copied "mk2-dbus-bms" on your test setup, so you can play with it.

jhofstee, 2017-02-27:
gui showing is included in v2.01. Do we need additional alarms for this?

mpvader, 2017-02-27:
no we do not. At least not without any further thought or testing: the overvoltage signal is part of the normal balancing procedure.

jhofstee, 2017-02-27:
ok closing this then. If we need more alarms it can be reopened again.

closed 2017-02-27 by jhofstee