victronenergy/venus #91: vrmlogger exception

jhofstee, 2016-12-02:
5940 v1.74

2016-12-02 05:39:13.653482500 MainThread-vrmlogger: Starting mainloop, responding on only events
2016-12-02 06:36:16.109130500 exit_on_error: there was an exception. Printing stacktrace will be tryed and then exit
2016-12-02 06:36:16.116088500 Traceback (most recent call last):
2016-12-02 06:36:16.116088500   File "/opt/color-control/vrmlogger/./ext/velib_python/ve_utils.py", line 16, in exit_on_error
2016-12-02 06:36:16.116118500     return func(*args, **kwargs)
2016-12-02 06:36:16.116118500   File "/opt/color-control/vrmlogger/vrmlogger.py", line 217, in send_data
2016-12-02 06:36:16.161620500     data = dbusMonitor.get_values(['onIntervalAlwaysAndOnEvent', 'onIntervalAlways'], Conversions())
2016-12-02 06:36:16.194213500   File "/opt/color-control/vrmlogger/./ext/velib_python/dbusmonitor.py", line 248, in get_values
2016-12-02 06:36:16.224395500     result.update(self.get_values_for_service(categoryfilter, serviceName, converter))

jhofstee, 2016-12-02:
Seems to be out of memory issues (perhaps 100MB of softlimit?)

@40000000583fddfe1b0aff1c Traceback (most recent call last):
@40000000583fddfe1ba60e6c   File "/opt/color-control/vrmlogger/http_endpoint.py", line 410, in _worker
@40000000583fddfe25c8b8cc     self._worker_inner(logurl, interval)
@40000000583fddfe2653070c   File "/opt/color-control/vrmlogger/http_endpoint.py", line 435, in _worker_inner
@40000000583fddfe26fe6624     result = self._send_jobs_to_vrm(job, vrmhttp)
@40000000583fddfe26fe6624   File "/opt/color-control/vrmlogger/http_endpoint.py", line 467, in _send_jobs_to_vrm
@40000000583fddfe26fe6624     compressed_data = zlib.compress(data,9)
@40000000583fddfe28190d84 MemoryError: Out of memory while compressing data

halmartin, 2016-12-02:
Using `zlib.compressobj` instead of `zlib.compress` might be an improvement as `compressobj` can handle large data streams which don't fit in memory. I'd be happy to take a look, but I'm not sure where the vrmlogger repository is, perhaps someone could point me to it?

jhofstee, 2016-12-02:
thanks @halmartin 

I guess the issues some other code is holding a reference or something to objects under some circumstances which causes all kinds of memory issues. See https://github.com/victronenergy/venus/issues/90 for example. So it is likely not related to zlib. It occurs mainly on 2 out of thousands installations, so no major issue for now.

The code is indeed not on github.

closed 2018-04-20 by jhofstee