victronenergy/venus #92: remote console over LAN no longer works

jhofstee, 2016-12-02:
Make sure to remove cached files when trying to reproduce!
Console on VRM still works fine.

jhofstee, 2016-12-05:
Both venus version, v1.74 and v2.00, have this problem

mpvader, 2016-12-06:
@jepefe: please look into this. Probably as a first step update to latest libraries and also the new layout as used on https://acceptancevrm.victronenergy.com/.

mpvader, 2016-12-07:
@jhofstee is this still a real issue? I heard it was fixed by a restart? Then I'll close this issue, and make one to update the javascript libraries and the rest of the client side things we are using first.

jhofstee, 2016-12-07:
yes, still an issue. Jesus had the issue as well and a customer reported 
it as well.
It is unknown at the moment what triggers it.

Restarting _might_ solve it. Sometimes it does, sometimes it doesn't. We 
don't
know why that helps. So I would keep it open, just set it to a low 
priority, so it
won't be forgotten to check this issue after updating the javascript 
libraries etc.

It might e.g. be a bug in firefox as well... We simply have no clue at 
the moment
(besides that it only occurs with Firefox on multiple OS-es, so at least 
it smells
like a Firefox bug).

mpvader, 2017-05-17:
I've not heard about this from customers for a long time.

closed 2017-05-17 by mpvader