victronenergy/venus #93: swupdate is not doing anything (Almere Hub-1 ~10)

mpvader, 2016-12-02:
It downloads the new file, but doesn't apply anything. What you see as a user is an endlessly rebooting device. With a 5 minute orso interval. If you disable the autoupdate (during this interval) the reboot loop stops.

To see whats going on, I added -vvvv into the do_swupdate() function in machine.sh. 

Then, on Almere Hub1, 192.168.1.111, running ~10, it just skips everything:

```
root@ccgx:~# /opt/victronenergy/swupdate-scripts/check-updates.sh -force -update
*** Checking for updates ***
arguments: -force -update
Retrieving latest version (feed=develop)...
installed: 201611211641 v2.00~10
available: 201611301632 v2.00~13
Starting swupdate to install version 201611301632 v2.00~13 ...
software set: stable mode: copy1
2
Swupdate v2016.07.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Searching for custom LUA handlers :[string "require ("swupdate_handlers")"]:1: module 'swupdate_handlers' not found:
	no field package.preload['swupdate_handlers']
	no file '/usr/share/lua/5.2/swupdate_handlers.lua'
	no file '/usr/share/lua/5.2/swupdate_handlers/init.lua'
	no file '/usr/lib/lua/5.2/swupdate_handlers.lua'
	no file '/usr/lib/lua/5.2/swupdate_handlers/init.lua'
	no file './swupdate_handlers.lua'
	no file '/usr/lib/lua/5.2/swupdate_handlers.so'
	no file '/usr/lib/lua/5.2/loadall.so'
	no file './swupdate_handlers.so'
Custom handlers not found, no error, skipping...

Registered handlers:
	archive
	tar
	flash
	flash-hamming1
	raw
	rawfile
	ubivol
	ubipartition
	lua
	shellscript
	preinstall
	postinstall
	uboot
Main loop Daemon
[NOTIFY] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
[NOTIFY] : SWUPDATE started :  Software Update started !
[NOTIFY] : SWUPDATE downloading :  
[NOTIFY] : SWUPDATE running :  [download_from_url] : Image download started : https://updates.victronenergy.com/feeds/venus/develop/images/ccgx/venus-swu-ccgx-201611301632-v2.00~13.swu
scan_for_ubi_devices
[NOTIFY] : SWUPDATE running :  [scan_ubi_volumes] : mtd12:	Volume found : 	rootfs1
[NOTIFY] : SWUPDATE running :  [scan_ubi_volumes] : mtd12:	Volume found : 	rootfs2
[NOTIFY] : SWUPDATE running :  [scan_ubi_volumes] : mtd12:	Volume found : 	data
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 0 / 0
[NOTIFY] : SWUPDATE running :  [extract_file_to_tmp] : Found file:
	filename sw-description
	size 1848
Version 0.1.0
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename venus-image-ccgx.ubifs
	size 70318080 Not required: skipping
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 1015808 / 74204160
...
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 70025216 / 74204160
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename uImage
	size 2850992 Not required: skipping
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 70893568 / 74204160
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 71794688 / 74204160
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 72630272 / 74204160
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename u-boot.img
	size 465692 Not required: skipping
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 73547776 / 74204160
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename MLO
	size 44080 Not required: skipping
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename splash.bgra
	size 522240 Not required: skipping
[NOTIFY] : SWUPDATE running :  [network_initializer] : Valid image found: copying to FLASH
[NOTIFY] : SWUPDATE running :  [download_from_url] : Success : 74204160 bytes
[NOTIFY] : SWUPDATE running :  Installation in progress
[NOTIFY] : SWUPDATE successful ! SWUPDATE successful !
[NOTIFY] : SWUPDATE running :  [network_initializer] : Main thread sleep again !
[NOTIFY] : No SWUPDATE running :  Waiting for requests...
Main loop Daemon
do_swupdate completed OK. Rebooting

```

So, besides not making those changes to the u-boot env. It is also not installing the kernel uImage and the rootfs image. Doesn’t matter how I call swupdate, with an offline image, or downloading the .swu file from the develop feed. Result is always the same: swupdate is not doing anything.

### Almere Hub-4
On another machine, Almere Hub-4, 192.168.1.115, running ~13, the update runs fine. Looking at the logs, there is a difference: the 5 parse_something lines you see here are missing on Almere Hub-1.

```
[NOTIFY] : SWUPDATE running :  [scan_ubi_volumes] : mtd12:   Volume found :                data
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 0 / 0
[NOTIFY] : SWUPDATE running :  [extract_file_to_tmp] : Found file:
                filename sw-description
                size 1848
Version 0.1.0
[NOTIFY] : SWUPDATE running :  [parse_images] : Found Image  : venus-image-ccgx.ubifs in volume : rootfs2 for handler ubivol (installed from stream) 
[NOTIFY] : SWUPDATE running :  [parse_images] : Found Image  : uImage in device : /dev/mtd11 for handler flash 
[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: nandboot = nand read ${loadaddr} ${kernel} && bootm ${loadaddr}
[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: miscargs = omapdss.def_disp=lcd omapfb.vram=0:2M vram=2M fbskip
[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: version = 2
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
                filename venus-image-ccgx.ubifs
                size 70318080 required
```

mpvader, 2016-12-02:
[swupdate binary from Almere Hub-1.zip](https://github.com/victronenergy/venus/files/627939/swupdate.from.Almere.Hub-1.zip)

mpvader, 2016-12-02:
I copied that swupdate binary over to another machine that has no problems (Almere Hub-4), and that does not break the update mechanism. So conclusion is that the issue is not in the binary.

Would also be strange that it is in the binary, because the build logs show that both builds (~10 and ~13) use the same git hash for the meta-swupade repo.

jhofstee, 2016-12-02:
Turns out to be a silly debugging leftover. To check what went wrong with ~11 I added an echo of the rootfs to the function below. Since the echo is actually used, it ended up trying to update the current image instead of the alternative one. Removing the echo makes it work as intended...
 
# print the number of the backup rootfs
get_altrootfs() {
    rootfs=$(get_rootfs)
    case $rootfs in
        1) echo 2 ;;
        2) echo 1 ;;
    esac
}

closed 2016-12-02 by jhofstee