victronenergy/venus #94: add disable PV Inverter flag

mpvader, 2016-12-05:
Some battery systems (redflow for example) will block charging current into the battery once the battery is full. In a AC-Coupled system this gives issues: the battery is needed as a buffer for sudden power changes. For example load drops.

There will be a flag to make the Multi change its output frequency to 53Hz (or whichever other frequency is configured). Which will cause the PV Inverter to shut down.

This flag will be implemented in two Assistants:

- [x] ESS Assistant
- [ ] PV Inverter support

In Venus, this needs to done in the following projects:

- [ ] Add flag to mk2-dbus
- [ ] Add flag to ModbusTCP
- [ ] Add flag to vrmlogger

This is not related to ESS or Hub-4, so the path generated by mk2-dbus should not start with /hub4

mpvader, 2016-12-05:
The flag will only be effective when the system is in Inverter mode.

Any system using this, for example the Redflow BMS, needs to write the value to the CCGX on an interval. Not only on change.

mpvader, 2016-12-06:
its in vecan-mk2, aka mk2-dbus, v2.42, which will be in v2.00~16

SimonHackett, 2016-12-07:
Thank you for making these changes - they'll be really useful to us.

I have some questions about it, which would be very helpful to understand the answers to:

1) Does the statement about needing to write this "on an interval' mean it will self-reset back to normal operation after a while, as a fail-safe to restore 'standard' operation?

2) At what interval (how often) do we need to be updating the (boolean?) value of this flag via MODBUS-TCP to keep the new function working as we intend?

3) Which MODBUS-TCP register/unit number will it be? 

4) Will be be a boolean, i.e.  0 for normal PV operation and 1 for inhibit PV generation ?

Thank you!

mpvader, 2016-12-07:
Hello Simon,

> 1. Does the statement about needing to write this "on an interval' mean it will self-reset back to normal operation after a while, as a fail-safe to restore 'standard' operation?

It will only reset after the CCGX resets, or after the Multi resets.

> 2. At what interval (how often) do we need to be updating the (boolean?) value
> of this flag via MODBUS-TCP to keep the new function working as we intend?

Once a minute for example? Or once every 10 seconds.

> 3. Which MODBUS-TCP register/unit number will it be?

Looks like it is registerid 56, and the unit id of the vebus device. See this commit for details:
https://github.com/victronenergy/dbus_modbustcp/commit/df19cafe0f774b09596cb508925df4e6edaad2df

> Will be be a boolean, i.e. 0 for normal PV operation and 1 for inhibit PV generation ?

Something like that yes, for exact definition see that commit.

mpvader, 2016-12-07:
feature will be finished with the release of v2.00~19

SimonHackett, 2016-12-25:
Hi guys,

Thank you for putting this feature in. I have just been testing it in my home system after loading the ESS software and it works as expected.

Actually it works better than expected - the PV Disable setting is persistent (including across reboots of the CCGX) so (in my experience so far) I don't strictly seem to have to keep 'refreshing' it - I can just change it when I need to change it (in an event-based manner) too. 

I think its actually useful this way - and it would be nice if it stayed that way - i.e. it would be great if it was persistent in general as a supported aspect of the setting. 

I think it would be great, also, to show the setting on the CCGX GUI in the future (to let a user see this value and also to let them manually configure it there if they want to). It could be called something like "Disable PV output when grid is down" (as a boolean setting).

I can think of cases where an installer might like to have the PV installed on the output side of the Multi but where they might want to have "PV Disable" active 'by design' (for instance, where they can not respect the 'factor 1.0' rule in the system design when the grid is down). 

So - I think this is a really nice setting in general, and if we can keep it working like it works now (persistent, including across CCGX restarts) I think that would be great.

Long story short: I think its working great  - with thanks for implementing it (because we definitely need it with the Redflow ZBM battery systems to handle specific edge cases where all ZBM's might be offline or full at the same time).

SimonHackett, 2017-01-03:
I have been testing the operation of this new flag on my home system and I've run into a problem with it that I would appreciate some help with. The executive summary is that it doesn't (yet) solve the problem.

The flag works as described - but the limitation noted above (that it only works when the system is in inverter mode) is a problem at face value.

In the exact scenario where I need this to work:
- Batteries full
- Sun shining strongly
- Not much house AC load

... if I disconnect the grid, then the system instantly overloads and shuts down (and yes: with the PV Disable flag set!)

I guess this is because the system can't shift out of charger mode and into inverter mode and then start to wind the frequency up, fast enough to avoid the system overload-shutdown happening first.

Can you tell me if there is a workaround here?

Specifically, I'm asking if there is a way to do one of the following things:

1) Disable the PV Inverters via this 'PV Disable' flag at all times (not just when the system is in inverter mode)?  

[ I am expecting this isn't possible, but I wanted to ask]; 

OR:

2)  Force the system into inverter mode via a MODBUS control?

It would be fine if forcing it into inverter mode means the battery starts to be discharged into the load (i.e. so that the battery energy is used alongside the grid to support the load)...

...that's just fine (and effectively that solves the problem, by discharging some energy from the battery so it isn't at risk of overload shutdown if the grid fails).

I can control that mode shifting from the Redflow ZCell BMS based on SoC - but I just need to know the best MODBUS controls to use to do this, in order to achieve that outcome...

Thanks!

edevries-victron, 2017-01-03:
Solution 2 might work: you can use modbus register 33 (switch position). Value 2 will switch the Multi to inverter only mode, value 3 back to normal mode.

SimonHackett, 2017-01-03:
If I do that (switch position to 2)... does that turn off grid assist (so I'll lose the whole system if the demand rises above the battery output limit)?

SimonHackett, 2017-01-03:
On a related note - can I used modbus to adjust the charge limit (the control on the ESS menu page on the CCGX) - e.g. if I set the charge limit to zero watts, would that (by side effect) turn off the charger... and maybe result in the inverter being on as well? 

(Clutching at straws, maybe)

SimonHackett, 2017-01-04:
Ed,

I can't do it by switching the Multi to inverter-only mode (switch value set to 2), because this disables the use of grid energy (gridassist isn't available)...  so that I'm vulnerable to the system shutting down due to spike AC demands. That's not sensible when the grid -is- present. I proved this to myself (and wound up with a dark house as a result :) ).

The workaround I can use (in addition to using the PV (Inverter) Disable feature, is to have the Redflow BMS act to stop charging the batteries just before they all reach 'full'. If I can do that - if I can command the system to stop charging batteries via MODBUS - then I can this remaining issue in that way.

I am having a bit of a challenge doing that - the Hub4 MODBUS controls aren't quite acting in the way I expect when using ESS (and there are no new ESS specific MODBUS controls).

But - I'll take that up in the ESS Disqus forum at this point, since its not actually to do with "PV Disable" as such. 

If you think of some other way to force the PV Inverters off without losing 'grid assist', though, I'd love to know about it :)

mpvader, 2017-01-04:
@SimonHackett:

There is no way to force the system into Inverter mode (to start frequency shifting) without losing 'grid assist'.

I think the best solution is your last idea: stop charging before the batteries start blocking charge currents. The best (and only!) way to achieve that is to have the batteries send 0A as max charge current, in the canbms protocol.

Ps. please dont start a conversation about hub4 and ess modbustcp controls. We'll first make a new document.

mpvader, 2017-01-04:
@SimonHackett:

answering your earlier message:

> Thank you for putting this feature in. I have just been testing it in my home system after loading the ESS software and it works as expected.
>
> Actually it works better than expected - the PV Disable [...]

No, that is not how it works. Its value is not stored into non-volatile memory anywhere: so power cycling the Multi, or the CCGX, for example for a software update, will reset it. Please keep updating the flag.

> I think its actually useful this way - and it would be nice if it stayed that way - i.e. it would be great if it was persistent in general as a supported aspect of the setting.

Perhaps, but it is not a setting, it is a run time flag.

> I think it would be great, also, to show the setting on the CCGX GUI in the future (to let a user see this value and also to let them manually configure it there if they want to). It could be called something like "Disable PV output when grid is down" (as a boolean setting).
> 
> I can think of cases where an installer might like to have the PV installed on the output side of the Multi but where they might want to have "Disable PV when grid is down" active 'by design' (for instance, where they can not respect the 'factor 1.0' rule in the system design when the grid is down).

That won't work. The response won't be fast enough. The factor 1.0 rule is always in effect. This flag is (unfortunately!) no way to get around it.

> Anyway: I think this is a really nice setting in general, and if we can keep it working like it works now (persistent, including across CCGX restarts) that would be great.

Hearing this, it might be persistent across CCGX restarts, but it will certainly not be persistent across Multi restarts. Also in your system it is part of a operational loop, not a user setting, so therefor I prefer to keep it that way (non-volatile run time flag). Making it a setting means writing to flash, which means thinking of flash wear, etc, which I want to prevent.

SimonHackett, 2017-01-04:
Hi Matthijs,

Yes, I’d love the new document - right now its a bit hard to work out which old ‘hub 4’ parts of this story still work, and which do not.

I’m quite comfortable with proactively stopping battery charge before the last battery hits a ‘hard stop’. That plus the new PV (inverter) Disable will do what I need.

Just confirming what you are saying below  - in particular whether the CANBus protocol is really the only way to do this…

I have found that I can (also) stop the charger by using the old hub4 control point "om.victronenergy.hub4 /MaxChargePercentage” (setting that to 0 to stop charge, setting it to 100 to re-enable charge).  

However, it seems to do something a bit strange when I do it - which I am inferring to be a strange interaction between the Hub 4 control and the CANbus charge current limit (the system starts to oscillate slowly between normal operation and passthru when it is discharging).

Is the old Hub4 control something I should (now) avoid using?

It would be great to have an external (MODBUS) capable mechanism that is fully supported, to do what the old hub 4 controls do, i.e. to allow the charger to be turned off and on as needed.

Meantime - I will get my guys to give me the control I need to send 0A as the max charge current, as you suggest below, and I expect that will work (and it is probably the best way to make it work).

Regards,
 Simon

SimonHackett, 2017-01-04:
> Hearing this, it might be persistent across CCGX restarts, but it will certainly not be persistent across Multi restarts. Also in your system it is part of a operational loop, not a user setting, so [...]

I can confirm by observation that is *is* persistent across CCGX restarts, but I understand what you are saying about the Multi.

Yes, I have now got the Redflow BMS refreshing the flag into the CCGX at regular intervals now - no problem. That works.

What doesn’t work is my original understanding of the fact that this doesn’t solve the entire problem - and that I (also) need to disable battery charging ‘on command’ when needed too (per the other aspect of this thread).

Thank you,
  Simon

mpvader, 2017-01-04:
@SimonHackett 

> I have found that I can (also) stop the charger by using the old hub4 control point "om.victronenergy.hub4 /MaxChargePercentage” (setting that to 0 to stop charge, setting it to 100 to re-enable [...]

Also `/MaxChargePercentage` doesn't work under all conditions.

Please just focus on the maxcharge via the canbusbms protocol. It is the only way to achieve what you want.

Matthijs

SimonHackett, 2017-01-04:
> Please just focus on the maxcharge via the canbusbms protocol. It is the only way to achieve what you want.
> 

Got that now - loud and clear. Will do!

Cheers, 
 Simon

mpvader, 2017-08-31:
@vejpasop a while ago, you have received a test version of the PV Inverter Assistant with that flag implemented. I suppose that besides testing you also need to add support for it in the Vebus service. What is the status?

vejpasop, 2017-09-07:
PV inverter assistant support was added to Venus V2.00 (vecan-mk2 V2.42)  
It adds a  "/PvInverter/Disable" flag to the DBus.

SimonHackett, 2017-09-07:
Unfortunately, while the flag exists in the DBus and while this flag can be set using MODBUS commands, the Disable function does not work if the PV Assistant is loaded into a VEBus inverter/charger. 

With the PV Assistant loaded, there is no frequency shift in response to this flag being set.

[ The function does work if the ESS Assistant is loaded and if ESS is running in island-mode;  but this operating mode is not supported by Victron Energy for a pure off-grid system - so the PV Assistant must be used instead ]

The function is supposed to work when the PV Assistant is present.

We now have customer systems that were installed with the belief that this would work, and their design requires that this flag does exist and that it does work. Right now those systems are regularly overloading and shutting down.

Can we please have this function fully implemented in the PV Assistant?

vejpasop, 2017-09-07:
Something went wrong on our side. 
It turns out that release of the PV inverter assistant with added Disable flag was still pending.
I just tested it and we will release it soon.

mpvader, 2017-09-07:
new PV Inverter Assistant has been released just now, which closes this issue.

closed 2017-09-07 by mpvader